OpenZWave/open-zwave #851: Send "STOP" command via Somfy ZRTSI - Z-Wave to RTS Interface

raypp2, 2016-04-13:
I'm looking for a command that could trigger the STOP signal in Somfy window shade motors. It seems this should be possible by sending the following:

0x26  COMMAND_CLASS_SWITCH_MULTILEVEL
argument:  0x05    (SWITCH_MULTILEVEL_STOP_LEVEL_CHANGE)

I've tried a number of unsuccessful variations on: 
zwave.setValue(shade_node,38,1,00x05);

My UP and DOWN commands have been working correctly. 

**Reports of Working Z-Wave Commands**

> In the UI, if a device supports COMMAND_CLASS_SWITCH_MULTILEVEL, the UP button sends SWITCH_MULTILEVEL_START_LEVEL_CHANGE with the arguments 0x60, 0x0, and the DOWN button uses arguments 0x20, 0x0, while STOP does SWITCH_MULTILEVEL_STOP_LEVEL_CHANGE.
> 
> If the device does not support that command class, but does support COMMAND_CLASS_BASIC_WINDOW_COVERING, the UP button sends BASIC_WINDOW_COVERING_START_LEVEL_CHANGE with the argument 0x0, DOWN sends 64, 
> and STOP sends BASIC_WINDOW_COVERING_STOP_LEVEL_CHANGE.

http://wiki.micasaverde.com/index.php/Window_Covering_Notes

> "i tried using a z-wave send raw command using bytes "0x26, 0x05" as above. i can confirm that it works as STOP, i.e. when the shades were moving and this command was sent, the motors stop in their current position. "

http://forums.indigodomo.com/viewtopic.php?f=5&t=9356&sid=82e75990c85c4d2219c3de22d55bf9c8&start=30

**Z-Wave Device Specifications**
ZRTSI - Z-Wave to RTS Interface
https://www.somfysystems.com/file.cfm/ZRTSI_Spec_Sheet.pdf?contentid=311990

The stop command serves a number of important functions:
1. Stopping the shade while in motion 
2. Sending the shade to a programmed position when triggered in standby (i.e. 50% down)
3. Programming key settings like upper limit, lower limit, device pairing, etc ...

**Device Information**
node6: Somfy, Unknown: type=5a52, id=5402
node6: name="", type="Motor Control Class A", location=""
node6: class 37
node6:   Switch=false

node6: class 38
node6:   Level=0
node6:   Bright=undefined
node6:   Dim=undefined
node6:   Ignore Start Level=true
node6:   Start Level=0

node6: class 134
node6:   Library Version=7
node6:   Protocol Version=3.42
node6:   Application Version=1.08

Z-Wave Interface: Aeotec Z-Stick Gen 5

Fishwaldo, 2016-04-19:
What ValueID's do you have exposed on this device? 

If you have the Increment/Decrement ValueID's, the Stop Command is sent when the Button is Released.

Fishwaldo, 2016-05-17:
Closing. Please reopen if you have updates.

closed 2016-05-17 by Fishwaldo

joelrbrandt, 2019-01-31:
I had the same problem, but eventually got it to work. Here's how to use the Somfy ZRTSI interface:

To make the shades go down, "push" the **Bright** button:
```zwave.setValue(<shade id>, 38,  1,  1, true);```

To stop the shades while they're moving down, "release" the **Bright** button:
```zwave.setValue(<shade id>, 38,  1,  1, false);```

To make the shades go up, "push" the **Dim** button:
```zwave.setValue(<shade id>, 38,  1,  2, true);```

To stop the shades while they're going up, "release" the **Dim** button:
```zwave.setValue(<shade id>, 38,  1,  2, false);```

tomaszzz, 2020-02-04:
Hi @joelrbrandt, what API are you running in your examples? I wonder what the arguments represent (besides the shade id of course). 

I am trying to do the same - make the dimmable part work with my roller shades so that the shade stops at 25, 50 , 75% and I am wondering if  all it takes is adjustment of the values that homseer uses to send to the shade?

Much appreciated.
Tomasz

My setup: 
- Z-Wave to RTS interface: 1811265
- Roller shades controlled by Sonesse 30 RTS motors
- HomeSeer HS3 Standard Edition 3.0.0.548 (Linux)

tomaszzz, 2020-02-04:
Never mind, just realized what page I am on :) .... open-zwave of course
I wonder if I will be able to translate these to homseer params.