OpenZWave/open-zwave #853: qubino-ZMNHCA2 H1S2P1

Warp95, 2016-04-14:
Hello,
Since you changed the qubino-ZMNHJA2.xml file to fully support the features of the S3 version, and after, I can not associate my H1S2P1 qubino-ZMNHCA2 !! Can you do something for the S2 version is not penalized.
Eric

nechry, 2016-04-14:
hello Warp95 if you have to add ctl in association grp 3 is work again?

Warp95, 2016-04-14:
The problem is that after the associate, the module generates a network saturation Zwave (LED Zwave the controller blinks constantly) and the association is not done very well, so it is impossible to act on it.

Fishwaldo, 2016-04-19:
This sounds like a Device Bug. Can you get Log from OZW?

Warp95, 2016-04-20:
Hello,
I have 4 devices and 4 H1S2P1 react the same way, it does not associate with your ZMNHCA2.xml version of January 13, 2016, but they work very well with the version of December 11, 2015.
Can not remain without these devices I have come to associate with the file ZMNHCA2.xml of 11 December 2015.
Today it is very difficult to repeat the test because my setup is clean and I would like disrupting the order of associations, but believe me there is a real problem.

Fishwaldo, 2016-04-25:
Without a log, I can't do anything....

crashcoq, 2016-05-26:
Hello

I have thé same problem with the zmnhca2 qubino
The usb z wave is saturated
Where I can find the old version for testing?
If I exclude and include, the module is unknown a few moment (2 days) ...

Cordially
David

Warp95, 2016-05-26:
Hello crashcoq,

You can find the version of December 11, 2015 here:

https://github.com/OpenZWave/open-zwave/blob/37711dc151d25c202e74c719d0df7c5f3e4b2039/config/qubino/ZMNHCA2.xml
Put this file into "/home/pi/domoticz/Config/qubino" 
We must also think about delete files "/home/pi/domoticz/Config/zwcfg_*.xml" before inclusion.

Eric.

crashcoq, 2016-05-26:
Hello

Thanks, it's working correctly now.

David

nechry, 2016-05-26:
the documentation contains 5 Associations groups and many more parameters. @crashcoq you have a old version of this device? 
This device is now in rev 6: User manual is valid for Shutter module with SW version S6 and above.

crashcoq, 2016-05-26:
I don't know my version, I have sold this in march 2016.
Where I can find this?

Warp95, 2016-05-26:
It is marked on the label on the back of the module:
example:
"ZMNHCA2 H1S2P1" for version 2
"ZMNHCA2 H1S6P1" for version 6

crashcoq, 2016-05-26:
I have 1 module h1s6p1 and 3 modules h1s2p1
I use this config file.
I don't have any problem on node.
I can now use percentage and on off on all shutter but I don't have any configuration on hardware menu.

Warp95, 2016-05-26:
You should have access to the configuration in the hardware menu.
Perhaps :
1.exclusion
2.delete files "/home/pi/domoticz/Config/zwcfg_*.xml"
3.restart domoticz
4.inclusion

crashcoq, 2016-05-26:
It's all ok
Thanks for all 
Cordially
David

nechry, 2016-06-08:
the jeedom config
https://github.com/OpenZWave/open-zwave/pull/901
are now merge
we restore 2 associations groups

closed 2016-06-08 by nechry