OpenZWave/open-zwave #854: No way to enumerate Options

tomq42, 2016-04-14:
I'd like to be able to list through the options in the Options object, but there's no method for doing that, I need to know what option I'm looking for (I want to dump out for debug purposes, and I'd rather do it directly, rather than reading the XML file in some way).

I'm quite happy to add this and send a pull request, if I can get an opinion on the preferred interface.

I notice for example, that to list scenes, the interface is to pass out a uint8* that must be freed, rather than, for example, to pass in a std::vector<uint8> by reference that gets populated. 
The simplest method of enumerating the options is to have a method that returns a std::vector(or set) of std::string, or passes one of those in to be populated, but I was worried that would not be idiomatic for this project. I don't see any particular reason to avoid that style (there are already classes that clients can construct instances of on the stack that contain STL objects, so client and openzwave already have to agree on compiler etc)

Thanks.

tomq42, 2016-04-14:
I notice that GetNodeStatistics has, in effect, got an out parameter of list<CommandClassData>, and some std::strings.

I plan to add something like 
GetOptionNames(list<string>&)

Fishwaldo, 2016-04-19:
I'd prefer to keep it at the simplest layer - ie, a uint8* as that would avoid any issues with Compilers if we go for ABI compatibilities later on Windows.

tomq42, 2016-04-26:
Achieving ABI compatibility for a C++ interface is essentially impractical. You may as well just write a C interface, or a "CLASS" style C interface (http://rfc.zeromq.org/spec:21). You can't pass ownership of memory across the boundary. You can't use any std collections such as std::string as arguments or return values. Objects that are instantiated can't contain std::strings or any other such classes as members. You can't even use the primitive long data type in the interface (different size on MSVC 64 as on GCC 64).

Not sure I understand how an interface would be structured to use a uint8*.

gms1, 2016-06-07:
there is already a private member "map<string,Option*> m_options" so you probably would only need a getter, but then it would be necessary to export the explicit template instantiation on windows
see:
https://support.microsoft.com/en-us/kb/168958
quote: Also note that some STL containers (map, set, queue, list, deque) cannot be exported.

tomq42, 2016-06-07:
I have a bunch of changes for this, I just haven't yet had time to put in a pull request. I will try and collect together my commits and put one in.

Fishwaldo, 2017-05-30:
We print out the options after Locking the options class now (to the Log File). I'll add a Iterator to this soon as well.