OpenZWave/open-zwave #859: Send Raw Packets

Fishwaldo, 2016-04-19:
Create a Manager Method to allow developers to send raw messages to the Controller. This could be useful for example with the Philio Siren

https://groups.google.com/d/msg/openzwave/IuImmA9PJIU/KnKs4r3WBQAJ

RichieRogers, 2016-05-31:
I know this has only been added last month, but do you have any ideas when this extremely useful feature might be available?
Currently running Domoticz and would love this feature to enable me to properly control a DLink DCH-Z510 siren (rebranded Aeotec siren, same a Philio and Zipato)

Richie

Fishwaldo, 2016-06-01:
If it really is a rebranded Aeotec Siren, then it should work fine without any need to send raw packets. I have one here at home, and no issues at all....

RaymondNg2, 2016-06-01:
You are able to play the other sounds besides the default alarm sound?

nechry, 2016-07-02:
@Fishwaldo I have also this siren, If you need any tests let me know.

vmoreau, 2016-07-02:
This feature would also be needed on my side for the Zipato siren. Thanks in advance for your work

fnhieu, 2016-07-04:
Really need this feature in order to get my Zipato siren working.
Thx for your hard work :)

Soldierounet, 2016-07-08:
I need the feature too. Thanks for your work!

nechry, 2016-09-07:
@Fishwaldo  is interesting ?:
https://community.smartthings.com/t/looking-for-help-for-z-wave-alarm-signal-dlink-dch-z510/33039

Fishwaldo, 2016-09-07:
@nechry nods. I figured this out when the doco was released. the Siren is recieving the Notification CC (old Alarm CC). 

Right now, we dont have a way to send Notification CC to devices...

nechry, 2016-09-08:
ok, so we will wait for a next version to get this support, I the right way.

Fishwaldo, 2016-09-08:
Yep. I'll probably revert my changes in the Dev tree to the Alarm Class as well, as after seeing the official specs, it's a lot more complex now than what we have so time to rethink the API there.

nechry, 2016-09-08:
Cool, so you will postpone the v1.6, with better features

ludomin, 2016-09-19:
Thanks you very much for the work! I have a siren Zipato and I would like to use all the features.

mschmit06, 2016-09-29:
Would be fantastic to be able to use all the features of the Zipato siren. Thanks a lot for your hard work.

marynofear, 2016-10-13:
BUMP I'm also in. Thanks for good work....and please...give us this feature.
I'm sure it will come in handy in many many situations..
This siren could be very usefull if we can get it to function with openzwave.
Sending command directly would be fine
Like:
Door Chime
luup.call_action('urn:micasaverde-com:serviceId:ZWaveNetwork1','SendData',{Node='13',Data='0x71 0x5 0x0 0x0 0x0 0x0 0x6 0x16 0x0 0x0'},1)

tobiasgraf, 2017-05-16:
Wouldn't something simple like this be enough? 
```
void Manager::SendRawFrame(uint32 const _homeId, Msg* msg, Driver::MsgQueue const _queue)
{
	if( Driver* driver = GetDriver( _homeId ) )
	{
		msg->Append( driver->GetTransmitOptions() );
		driver->SendMsg(msg, _queue);
	}
}
```
Then you are able to send raw frames using something like this:

```
    Msg* msg = new Msg( "Notification_Report", nodeID , REQUEST, FUNC_ID_ZW_SEND_DATA, true );
    msg->Append( nodeID );
    msg->Append( 10 ); //Length
    msg->Append( COMMAND_CLASS_NOTIFICATION_V5 );
    msg->Append( 0x05 );
    msg->Append( 0x00 );
    msg->Append( 0x00 );
    msg->Append( 0x00 );
    msg->Append( 0x00 );
    msg->Append( 0x06 );
    msg->Append( 0x16 );
    msg->Append( 0x00 );
    msg->Append( 0x00 );
    Manager::Get()->SendRawFrame(m_homeId, msg, OpenZWave::Driver::MsgQueue_Send);
```

gazoscalvertos, 2017-09-14:
Anyone tested this?

3hs4n, 2017-11-30:
Just tested the snippet above posted by @tobiasgraf and it works like a charm.
You can easily generalize the method to accept any command class and any byte array.

gazoscalvertos, 2017-12-01:
It would be great if this could make its way into both OpenZWave and Python-openzwave to be used in Home assistant

gdave321, 2017-12-23:
Also tested, this works great, I've all the sounds from my PSE02 siren now :-)
FYI this is the patch I used if you want to try:

[patch.txt](https://github.com/OpenZWave/open-zwave/files/1584042/patch.txt)

adriantaut, 2018-02-06:
Hello all, where exactly have you guys wrote the code in the snippet of @tobiasgraf  ?

tobiasgraf, 2018-02-06:
This really depends on the application you are using. 
Instead of using e.g. Manager::Get()->SetValue() to turn on a SWITCH_BINARY you can use Manager::Get()->SendRawFrame() to send whatever frame you like like in the snippet above.

Yipyipyip, 2018-03-30:
Do we still have a chance to get this working in OpenZWave? I'd like to control the Z510 siren from hass.io/HomeAssistant (which relies on python-openzwave), so I don't have any chance to modify the OpenZWave code and try the workaround.

tkintscher, 2019-01-22:
Hi everyone,
this functionality is very interesting (as it allows sending external temperature reports to my Eurotronic SpiritZ thermostat), so I have added it to both open-zwave and python-openzwave here:
https://github.com/tkintscher/open-zwave/commit/3fae0f33ce3da8a106c0195bf67fc0a28e6a60a7
https://github.com/tkintscher/python-openzwave/commit/c2ac86e7f841d60fba199cc9a7d9adac44cd79e1
It works very well for me for several months now.
Should I create a pull request?

Fishwaldo, 2019-05-03:
This was done in #1701

closed 2019-05-03 by Fishwaldo