OpenZWave/open-zwave #862: Report status in ZWave

jotakar, 2016-04-19:
I use Domoticz and I see  that double relays modules are reported as 3 relays: 1 common and one more for every real relay
I've observed that when we turn it ON/OFF manually, it sends to controller a signal for the common relay, but this data has no information about which relay has been turned ON or OFF. The bidirectionality is broken. 
In Domoticz there is a workaround to solve this issue: an script that poll the device when it has been manually switched. This get the status of every relay. So we avoid polling for all the switches in the net.
I wonder if it would be posible to add this functionality to ZWave library, so this script will be unnecessary.
I mean: 
when switch module is manually activated, it sends a signal and controller **updates satus of the common relay**. The new funcionality I refer to is one that make a ** poll of the module in this time and return the status of every relay**. So controller knows which relay has been activated by hand.

I don't know too much about how ZWave works, so I don't know if this suggestion nas no sense or if it must be a function of the controller (Domoticz in this case).
Thanks

Fishwaldo, 2016-04-25:
Could you post a OZW log snippet when you manually activate a change. 

You could use something like:
`<TriggerRefreshValue Genre="user" Index="0" Instance="1">
    <RefreshClassValue CommandClass="51" RequestFlags="0" Index="255" Instance="1" />
</TriggerRefreshValue>
`

But I need to look at the log to tell you exactly what to put in the config.

@nechry 
This just gave me a idea on a workaround for the Multi-instance Issues on Qubino Devices. We might be able to do the same..

nechry, 2016-04-25:
I try TriggerRefreshValue with other device and I will also add this behavior on some other device like aeon-labs to force a power refresh immediately after a switch CC, instead of wait for a power report.

nechry, 2016-04-25:
@Fishwaldo for my knowledge, what is the RequestFlags attribute?

jotakar, 2016-04-25:
Ok, I send the OZW debug file. The node with this doublé switch is Node0005 
[OZW_debug_nodo5.txt](https://github.com/OpenZWave/open-zwave/files/234917/OZW_debug_nodo5.txt)
Thanks

jotakar, 2016-04-25:
Sorry I didn't want close the issue!!!!

Fishwaldo, 2016-04-27:
@nechry I dont think we should add this to other devices. I don't like putting "Policy" into OZW and prefer to only keep it as something that fixes bugs etc. 

I think refreshing power values should be triggered by the application/user rather than OZW.

Fishwaldo, 2016-04-27:
@nechry RequestFlags is dependant upon the CommandClass you are triggering. its basically the first variable passed to the RequestValue method on the CommandClass, which most of the time is not used, but some CC's do use it.

nechry, 2016-04-27:
@Fishwaldo Thank you for the explanation and advice. anyway I tested the RefreshTriggerValue with switch and refresh comes too soon, in conclusion, it is unnecessary in this case.

Fishwaldo, 2016-05-04:
Please replace the file config/qubino/ZMNHBDx.xml with the attached version and tell us if this works. 

[ZMNHBDx.xml.zip](https://github.com/OpenZWave/open-zwave/files/248152/ZMNHBDx.xml.zip)

nechry, 2016-05-04:
Hi @Fishwaldo, I will also try this with the ZMNHBA2, I use the polling to refresh the instance 1 with this is more easy, I think.

rofrantz, 2016-05-04:
Hi @nechry is the ZMNHBA2 working for you with the existing config, because I can't control switches if I start with turning on switch 2. Qubino support told that we should see 3 switches: 2 for the real switches + 1 master which should rutn on/off both switches

jotakar, 2016-05-04:
Well I’ve used this config file, but I deleted the last line (CommandClass id="96" mapping="endpoints") to get the three switches (two for relays and one for the virtual relay1+relay2). 

And …it works!!!!!

Now my controller (Domotiz) shows real status of locally switched relays !!!

 

I’ll publish this result in Domoticz forum

 

Thanks

Juan Carlos

Enviado el: miércoles, 04 de mayo de 2016 6:31
Para: OpenZWave/open-zwave
CC: jotakar; State change

nechry, 2016-05-04:
Cool, I did not yet get time for testing but is nice to read.

rofrantz, 2016-05-04:
Ok @nechry I will test that config in a couple of hours.
In case it works @Fishwaldo, do you think we should update ZMNHBA2 config file so that anyone should benefit from this fix ?

nechry, 2016-05-04:
hi all, hi validate with a ZMNHBDx and a old ZMNHBA2 and it's work.
I just send this patch internally for massive tests with jeedom users. but we can also apply in ozw

rofrantz, 2016-05-04:
so @nechry you're basically saying that in order for my ZMNHBA2 double switch to work I should use that config file that was previously attached and:
a) rename the file inside that archive from ZMNHBDx.xml to ZMNHBA2.xml
b) remove that last line (CommandClass id="96" mapping="endpoints")

Right ?

Fishwaldo, 2016-05-05:
@jotaka: thanks for testing. I'll push it to the official repo shortly.

Fishwaldo, 2016-05-05:
@rofrantz: probably not. I'm sure @nechry will post a updated file soon.

nechry, 2016-05-05:
@rofrantz no they don't share entire parameter definition and association grp. I will add only the valuetrigger policy on old file and remove the Cc 96 behavior

jotakar, 2016-05-10:
I'just discovered that this modification on config file
` <TriggerRefreshValue Genre="user" Index="0" Instance="1">
 <RefreshClassValue CommandClass="51" RequestFlags="0" Index="255" Instance="1" />
 </TriggerRefreshValue>`
seems to work also in a Philio PAN04  double relay.
Perhaps it could be tested for others modules of relays to be include in config files.

Fishwaldo, 2016-05-16:
@jotakar Please submit a PullRequest for the devices you have tested with this change and confirmed working. Its really hard to track different devices in issues with other devices.

Fishwaldo, 2016-08-11:
Closing this.

closed 2016-08-11 by Fishwaldo