OpenZWave/open-zwave #863: The Device did not acknowledge the message of type "AlarmCmd_Get". You should investigate if this device really supports the associated CommandClass

vincemic, 2016-04-20:
Schlage BE369 Z-Wave Deadbolt Lock AlarmCmd_Get is not acknowledged even though the device seems to work in most respects with OZW.

Log file.
http://www.openzwave.com/log-analyzer/public/96210215-9bc7-46c5-aa71-30123d904339/

The log scan reports that the zwconfig was present, but it wasn't.

The log doesn't report my Aeotec Gen5 Z-Stick correctly or my Aeotec  MultiSensor 6 even though they seem to work correctly. It also shows many more nodes than the four that I actually have.

vincemic, 2016-04-20:
Sorry "You should investigate" was in the log and I copied it. It wasn't my intent to command you to investigate.

vincemic, 2016-04-20:
PS:  This log is incomplete because of a Seg Fault.  I ran it from MinOZW with the same results.
Last lines before the Seg Fault

 ZW_SEND_DATA Request with callback ID 0x01 received (expected 0x01)
2016-04-20 12:51:05.446 Info, Node004, Request RTT 1493 Average Request RTT 1409
2016-04-20 12:51:05.499 Detail, Node004,   Received: 0x01, 0x1e, 0x00, 0x04, 0x00, 0x04, 0x18, 0x98, 0x81, 0x4c, 0x6a, 0xaf, 0xea, 0xf5, 0xeb, 0xf4, 0x62, 0x0a, 0xa0, 0x52, 0x5d, 0xe4, 0x2f, 0x29, 0x06, 0xc2, 0x19, 0xbb, 0x6a, 0xe2, 0x87, 0x25
2016-04-20 12:51:05.500 Info, Raw: 0x98, 0x81, 0x4c, 0x6a, 0xaf, 0xea, 0xf5, 0xeb, 0xf4, 0x62, 0x0a, 0xa0, 0x52, 0x5d, 0xe4, 0x2f, 0x29, 0x06, 0xc2, 0x19, 0xbb, 0x6a, 0xe2, 0x87, 0x25
2016-04-20 12:51:05.501 Detail, Node004, Decrypted Packet: 0x00, 0x63, 0x03, 0x03, 0x00
2016-04-20 12:51:05.501 Detail, 
2016-04-20 12:51:05.502 Info, Node004, Response RTT 1549 Average Response RTT 1578
2016-04-20 12:51:05.503 Info, Node004, User Code Packet is 0
2016-04-20 12:51:05.504 Detail, Node004, Initial read of value

Fishwaldo, 2016-04-25:
The segfault should have just been fixed. 

As for the AlarmCmd_Get message:
It reports it supports Access Control, Burglar, Power Management and System, but only replies to Access Control and System. The other two time out.

Have you ever seen those two change values?

openzwave-dev-watcher, 2016-04-25:
I did a factory reset of my controller to start fresh so that I can create 
some documentation as I set things up.  Now I can no longer get my lock to 
pair.  I'll get more information when I get past this latest hurdle.

dhylands, 2016-04-25:
> I did a factory reset of my controller to start fresh so that I can create some documentation as I set things up. Now I can no longer get my lock to pair. I'll get more information when I get past this latest hurdle.

Factory reset doesn't reset the pairing information. Try doing an unpair first (you can unpair any device from any network). I had to do this when I factory reset my dongle (which changes the homeID.

Fishwaldo, 2016-04-27:
yeah - you need to reset the lock, as its still thinking its paired with the old HomeID.

openzwave-dev-watcher, 2016-04-28:
> Yup, unpairing it did the trick. Even though the controllers home id had 
> been changed.  After that it paired with new controller just fine.
>

Thank you.
vince

openzwave-dev-watcher, 2016-04-29:
I created a new log that shows the lock not responding to the Gets.

http://openzwave.com/log-analyzer/public/f2e4f302-ff94-445a-85f8-af87a0cc3e8e/

vincemic, 2016-05-03:
The Device did not acknowledge the message of type "DoorLockCmd_Get". You should investigate if this device really supports the associated CommandClass 
The Device did not acknowledge the message of type "AlarmCmd_Get". You should investigate if this device really supports the associated CommandClass.

My guess is that the lock doesn't support these messages, how do eliminate them from the config?  In the master branch latest, it seems that the timeouts for these messages causes OZW to report the node as Dead.  I didn't see that behavior in previous versions.

Thoughts?

vincemic, 2016-05-04:
After reading some older issues with similar locks, I decided to unpair the lock and do a hard reset on it. I then readded to the network and it seems to be working again,

Thank you.

closed 2016-05-04 by vincemic