OpenZWave/open-zwave #878: ZMNHID1

jotakar, 2016-05-04:
Well, a new device a new issue.
This is a thermostate of Qubino and it has a pair of issue:
1. It report an erroneus value for power. My device is controlling a heater of 600 watts and it reports about 98 watts.
2. If I write a decimal value in setpoint, the scale of temperatura changes to Farenheit!!!! How reset to Celsius?

Thanks

nechry, 2016-05-04:
I'm also work on this device:
https://github.com/jeedom/plugin-openzwave/blob/master/ressources/openzwave/config/qubino/ZMNHIDx.xml

I use this override:
```
<CommandClass id="67" override_precision="2"  />
```

but we go also many issues with the device, they report on 2 instances and is not fully reliable.

can you share your ZMNHIDx.xml

jotakar, 2016-05-06:
The first issue is fixed, but the problem wiht random change of scale stays.
The config XML is the attached
[ZMNHID1.txt](https://github.com/OpenZWave/open-zwave/files/251434/ZMNHID1.txt)
I've added the code for override_precision="2"

jotakar, 2016-05-09:
The config modif doesn't fix the issue. So I modified the file ThermostatSetpoint.cpp as I had read in other post: change
`value->SetUnits( scale ? "F" : "C" )`
to
`value->SetUnits(scale == 1 ? "F" : "C")`

Now scales are OK, although I can set decimals.

jotakar, 2016-05-10:
I have had an strange issue with this module (I use Domoticz as controller).
I've observed that it read TWO temperaturas, first the real temperatura and then 0. It send two temperatures, 
it look like it read from two instances (?). The thermostate Works and switch ON/OFF but I can see the actual temperature. I see a fixed value.
Then I think:  it makes two readings... well delete associations from group 9 with controller. And voilá. Now the device sends just a temperature reading and it works OK. 
Ah, I've observed this device is not in manufacturer_specific.xml of last versión, but it was in an older version.
Bye

jotakar, 2016-05-14:
Will this device added in the next update of OpenZwave? As I told it's working fine for me, Although indeed  I've not checked the I2 and I3 inputs, but the device works: it turn ON/Off at setpoint temperature.

nechry, 2016-05-14:
Yes the jeedom merge is in progress I hope next week all updated config will be send to ozw repo

jotakar, 2016-05-16:
I've found a new issue with this device. Last time I connect the thermostat the power measurement gave a bad  value. Until now It was working fine, but now it gives me a value of about 98 watts when I have connected a resistor of 600 watts. 
From log file I get
2016-05-16 03:46:25.730 Detail, Node005,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x05, 0x0a, 0x32, 0x02, 0x21, 0x34, 0x00, 0x00, 0x03, 0xa3, 0x00, 0x00, 0x61
2016-05-16 03:46:25.730 Info, Node005, Response RTT 77 Average Response RTT 75
2016-05-16 03:46:25.730 Detail, Node005, Refreshed Value: old value=false, new value=false, type=bool
2016-05-16 03:46:25.730 Detail, Node005, Changes to this value are not verified
2016-05-16 03:46:25.731 Info, Node005, Received Meter report from node 5: Power=93.1W
2016-05-16 03:46:25.731 Detail, Node005, Refreshed Value: old value=95.6, new value=93.1, type=decimal
2016-05-16 03:46:25.731 Detail, Node005, Changes to this value are not verified
2016-05-16 03:46:25.731 Detail, Node005,   Expected reply and command class was received
2016-05-16 03:46:25.731 Detail, Node005,   Message transaction complete

I don't know why this strange behaviour. It was measuring fine, I didn't touch any param for this device. ???

Fishwaldo, 2016-05-16:
Duplicate as #776 which requires #819 and #820 to fix.

closed 2016-05-16 by Fishwaldo