OpenZWave/open-zwave #880: Again ZMNHBDx

jotakar, 2016-05-06:
I set power report for every relay in 30 seconds to check. 
This module is shown as 3 relays: 1 relay for Q1 + Q2 (say Q3) and two real relay Q1 and Q2. This is OK.
If either relay Q1 or Q2 is switched, then Q3 is activated. Thats fine.
But the relays doesn't send the power report properly. I use Domoticz.
It sends the report to the relay Q3. And it sends the total energy meadurement, and some times it shows the power of one relay, then the power of the other relay and then the total energy.
I don't know if its a problem of firmware or config

The info of the device I can see through Domoticz app is:
ZWave + versión 1
Library vers. 3
Protocol 4.05
Aplication vers. 1.01

Thanks

nechry, 2016-05-06:
If you force refresh on instance 2 and 3 your 3 reports are ok? Me be we must also add the refresh policy on power report

jotakar, 2016-05-07:
I don't think so. The issue is that the module doesn't ignores values set in config. In log of Domoticz I see that there are a read of common relay every few seconds, although both relays are OFF. I've send a report to Qubino and I'm waiting an answer. I think it's a firmware fault.

jotakar, 2016-05-08:
I've made a snif using Python-openzwave and I have see that there is readings from this modules that include a parameter instance with values 1, 2, 3. And it looks that the 1 is the total power, while 2 is the endpoint 1 and 3 is the endpoint 2. So it looks like the mode send the reports, but then why doesn't meters show the right values?
I attach a file wth data from the node of this module.
[debug_snif_nodo5.txt](https://github.com/OpenZWave/open-zwave/files/254085/debug_snif_nodo5.txt)
I hope this data are useful for fixing this issue.
Bye

jotakar, 2016-05-10:
Qubino says hat there is not a bug. :(
But I see that the device sends readings of power of every endpoint but without any info about the source endpoint. So I see three reports in the same endpoint (the instance 0, that shows the total power comsumtion). When I reset the device, that is, force a reading then a value ofpower appears for every endpoint. And if I switch off a relays its power change to 0.0 w.
I don't know too much about ZWave, but I've read something about multiinstance command class an so. Perhaps this is a problem of reading multinstance device? 
Thanks

Fishwaldo, 2016-05-11:
Not a Bug - But not the way every other vendor does it on the market (sending the update in a MultiInstance packet)

Qubino expect you to associate each Group with a different instance on the controller. The issue is that we only have a single instance on the controller..... 

Are you sure if you manually refresh the values, you don't get the correct readings? that should work.

jotakar, 2016-05-11:
Of course, if a manually refresh the device I get the right values. And when a switch is switched off its power reading becomes 0. But this is not a real time value. 
Then the only fix is looking for a "multi instance" controller  (?) or no to use Qubino if you want to have power readings. Isn't it?
Or perhaps is there a posible fix via software?
Thanks

jotakar, 2016-05-12:
Qubino and Philio, I've just discovered that Philio has the same problem. Is it an issue for all double relays? Total power is read, but individual power must be read with an refresh.... 
Then the fix (if any) depends on controller (stick USB) or on host software (in my case Domoticz) or on ZWave libraries?
Thanks.... and well, I'm learning a bit about ZWave .... Every problem is an opportuniy for learning ;-)

openzwave-dev-watcher, 2016-05-13:
I'v been looking at this too. Using Domoticz. And the Qubino support 
results are pretty screwey

Using the latest code from github for both openzwave and domoticz and it's 
reasonably clear to me that OZW always seems the instance as 0 when 
receiving meter reports from the Qubino. A quick bit of editing to print 
the _instance when logging the meter readings and it doesn't matter which 
relay is on, or both the _instance is always reported as 1. e.g.

Relay1 (98W) (Grepping for 'Meter report|*HN*' from the OZW log

2016-05-13 22:13:37.679 Detail, Node003,   **HN** Received: 0x01, 0x07, 
0x00, 0x13, 0x4f, 0x00, 0x00, 0x0f, 0xab

2016-05-13 22:13:37.689 Detail, Node003,   **HN** Received: 0x01, 0x04, 
0x01, 0x13, 0x01, 0xe8

2016-05-13 22:13:37.849 Detail, Node003,   **HN** Received: 0x01, 0x07, 
0x00, 0x13, 0x50, 0x00, 0x00, 0x10, 0xab

2016-05-13 22:13:37.972 Detail, Node003,   **HN** Received: 0x01, 0x0d, 
0x00, 0x04, 0x00, 0x03, 0x07, 0x60, 0x0d, 0x01, 0x01, 0x25, 0x03, 0xff, 0x46

2016-05-13 22:13:37.974 Info, Node003, Received a MultiChannelEncap from 
node 3, endpoint 1 for Command Class COMMAND_CLASS_SWITCH_BINARY

2016-05-13 22:13:39.112 Detail, Node003,   **HN** Received: 0x01, 0x09, 
0x00, 0x04, 0x00, 0x03, 0x03, 0x20, 0x01, 0xff, 0x2c

2016-05-13 22:13:40.115 Detail, Node003,   **HN** Received: 0x01, 0x09, 
0x00, 0x04, 0x00, 0x03, 0x03, 0x25, 0x03, 0xff, 0x

2016-05-13 22:13:41.576 Detail, Node003,   **HN** Received: 0x01, 0x09, 
0x00, 0x04, 0x00, 0x03, 0x03, 0x25, 0x03, 0xff, 0x2b

2016-05-13 22:13:42.607 Detail, Node003,   **HN** Received: 0x01, 0x10, 
0x00, 0x04, 0x00, 0x03, 0x0a, 0x32, 0x02, 0x21, 0x34, 0x00, 0x00, 0x03, 
0xa8, 0x00, 0x00, 0x6c

2016-05-13 22:13:42.608 Info, Node003, Received Meter report from node 3: 
instance 1: Power=93.6W

Then I turn off relay 1 and turn on relay 2... Which is ~8W

016-05-13 22:16:35.184 Info, Node003, Value::Set - 
COMMAND_CLASS_SWITCH_BINARY - Switch - 0 - 2 - True

2016-05-13 22:16:35.194 Detail, Node003,   **HN** Received: 0x01, 0x04, 
0x01, 0x13, 0x01, 0xe8

2016-05-13 22:16:35.342 Detail, Node003,   **HN** Received: 0x01, 0x07, 
0x00, 0x13, 0x53, 0x00, 0x00, 0x0f, 0xb7

2016-05-13 22:16:35.352 Detail, Node003,   **HN** Received: 0x01, 0x04, 
0x01, 0x13, 0x01, 0xe8

2016-05-13 22:16:35.512 Detail, Node003,   **HN** Received: 0x01, 0x07, 
0x00, 0x13, 0x54, 0x00, 0x00, 0x10, 0xaf

2016-05-13 22:16:35.632 Detail, Node003,   **HN** Received: 0x01, 0x0d, 
0x00, 0x04, 0x00, 0x03, 0x07, 0x60, 0x0d, 0x02, 0x01, 0x25, 0x03, 0xff, 0x45

2016-05-13 22:16:35.633 Info, Node003, Received a MultiChannelEncap from 
node 3, endpoint 2 for Command Class COMMAND_CLASS_SWITCH_BINARY

2016-05-13 22:16:36.036 Detail, Node003,   **HN** Received: 0x01, 0x09, 
0x00, 0x04, 0x00, 0x03, 0x03, 0x25, 0x03, 0xff, 0x2b

2016-05-13 22:16:37.066 Detail, Node003,   **HN** Received: 0x01, 0x10, 
0x00, 0x04, 0x00, 0x03, 0x0a, 0x32, 0x02, 0x21, 0x34, 0x00, 0x00, 0x00, 
0x56, 0x00, 0x00, 0x91

2016-05-13 22:16:37.068 Info, Node003, Received Meter report from node 3: 
instance 1: Power=8.6W

2016-05-13 22:16:40.074 Detail, Node003,   **HN** Received: 0x01, 0x10, 
0x00, 0x04, 0x00, 0x03, 0x0a, 0x32, 0x02, 0x21, 0x34, 0x00, 0x00, 0x00, 
0x4a, 0x00, 0x00, 0x8d

2016-05-13 22:16:40.076 Info, Node003, Received Meter report from node 3: 
instance 1: Power=7.4W

I'm not sure how to decode the message itself... So I can't say 100% 
whether the Qubino is actually reporting the instance or not... Some 
guidance there would be good. Reading jotakars post it would appear that 
the Qubino only reports TOTAL power (instance 1). And the relay power is 
reported as instance 2 and instance 3? But only by poll... But that's just 
weird. Because the switches are instance 1 and instance 2... 

Here's the traffic from turning them both on, which results in a figure 
that's about equal to relay1 + rely2... e.g.

016-05-13 22:18:58.618 Info, Node003, Value::Set - 
COMMAND_CLASS_SWITCH_BINARY - Switch - 0 - 1 - True

2016-05-13 22:18:58.628 Detail, Node003,   **HN** Received: 0x01, 0x04, 
0x01, 0x13, 0x01, 0xe8

2016-05-13 22:18:58.776 Detail, Node003,   **HN** Received: 0x01, 0x07, 
0x00, 0x13, 0x57, 0x00, 0x00, 0x0f, 0xb3

2016-05-13 22:18:58.786 Detail, Node003,   **HN** Received: 0x01, 0x04, 
0x01, 0x13, 0x01, 0xe8

2016-05-13 22:18:58.946 Detail, Node003,   **HN** Received: 0x01, 0x07, 
0x00, 0x13, 0x58, 0x00, 0x00, 0x10, 0xa3

2016-05-13 22:18:59.071 Detail, Node003,   **HN** Received: 0x01, 0x0d, 
0x00, 0x04, 0x00, 0x03, 0x07, 0x60, 0x0d, 0x01, 0x01, 0x25, 0x03, 0xff, 0x46

2016-05-13 22:18:59.072 Info, Node003, Received a MultiChannelEncap from 
node 3, endpoint 1 for Command Class COMMAND_CLASS_SWITCH_BINARY

2016-05-13 22:18:59.963 Detail, Node003,   **HN** Received: 0x01, 0x09, 
0x00, 0x04, 0x00, 0x03, 0x03, 0x20, 0x01, 0xff, 0x2c

2016-05-13 22:19:00.956 Detail, Node003,   **HN** Received: 0x01, 0x09, 
0x00, 0x04, 0x00, 0x03, 0x03, 0x25, 0x03, 0xff, 0x2b

2016-05-13 22:19:02.418 Detail, Node003,   **HN** Received: 0x01, 0x09, 
0x00, 0x04, 0x00, 0x03, 0x03, 0x25, 0x03, 0xff, 0x2b

2016-05-13 22:19:03.448 Detail, Node003,   **HN** Received: 0x01, 0x10, 
0x00, 0x04, 0x00, 0x03, 0x0a, 0x32, 0x02, 0x21, 0x34, 0x00, 0x00, 0x03, 
0xef, 0x00, 0x00, 0x2b

2016-05-13 22:19:03.450 Info, Node003, Received Meter report from node 3: 
instance 1: Power=100.7W

H

>

jotakar, 2016-05-13:
Yes, What I see in log for the device is:

2016-05-09 01:30:22.636 Detail, Node005,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x05, 0x0a, 0x32, 0x02, 0x21, 0x34, 0x00, 0x00, 0x01, 0x51, 0x00, 0x00, 0x91
2016-05-09 01:30:22.636 Detail,
2016-05-09 01:30:22.636 Detail, Node005, Refreshed Value: old value=false, new value=false, type=bool
2016-05-09 01:30:22.636 Detail, Node005, Changes to this value are not verified
2016-05-09 01:30:22.636 Info, Node005, Received Meter report from node 5: Power=33.7W
2016-05-09 01:30:22.637 Detail, Node005, Refreshed Value: old value=13.6, new value=33.7, type=decimal
2016-05-09 01:30:22.637 Detail, Node005, Changes to this value are not verified
2016-05-09 01:30:22.637 Detail, Node005, Notification: ValueChanged
2016-05-09 01:30:22.637 Detail, Node005, Notification: ValueChanged

Some times it's the total power consum, some time Q1 and some time Q2. But all is sent to endpoint 1.
I've read that if every endpoint is associated to a diferent instance of controller it would work. Then I break association with controller (node 1) for meter group (4 and 7). Then I add a refresch value in config file:
                <TriggerRefreshValue Genre="user" Instance="1" Index="8">
                    <RefreshClassValue CommandClass="50" RequestFlags="0" Instance="2" Index="8" />
                    <RefreshClassValue CommandClass="50" RequestFlags="0" Instance="3" Index="8" />
                </TriggerRefreshValue>
Then every time endpoint 1 send a report, system get values for the endpoints 2 and 3, and I get instant report measurements. It Works, but I dont like this fix (this increases traffic in net). 
If device sends reports from every endpoint it must be a way to get this values. 
I'm looking for more info Reading about association, multichannel and so... but I can't find anything.

openzwave-dev-watcher, 2016-05-13:
On Friday, 13 May 2016 23:57:02 UTC+1, jotakar wrote:
> Yes, What I see in log for the device is:
>
> 2016-05-09 01:30:22.636 Detail, Node005, Received: 0x01, 0x10, 0x00, 0x04,
> 0x00, 0x05, 0x0a, 0x32, 0x02, 0x21, 0x34, 0x00, 0x00, 0x01, 0x51, 0x00,
> 0x00, [...]

Yep. And it confuses the crap out of domoticz... Because the total comes as 
instance 1 domoticz assumes it's for relay 1. If you enable polling, you do 
at least get the values for relay 1 & 2... But of course  when relay 2 is 
running, you get values for 'relay 1'. because domoticz assumes the total 
is really relay1... sigh...

I've read that if every endpoint is associated to a diferent instance of 
> controller it would work. Then I break association with controller (node 1) 
> for meter group (4 and 7). Then I add a refresch value in config file:
>
> Then every time endpoint 1 send a report, system get values for the 
> endpoints 2 and 3, and I get instant report measurements. It Works, but I 
> dont like this fix (this increases traffic in net). 
> If device sends reports from every endpoint it must be a way to get this 
> values. 
>

Not sure I understand how endpoints work, versus instances (e.g in the 
qubino config mapping endpoints to instances?)... Is that a zwave thing? Or 
OZW thing? 
 

> I'm looking for more info Reading about association, multichannel and 
> so... but I can't find anything.
>
>
>
I've not really found any docs around at all... Trying to read the OZW 
code. but I loath C++. And I loath monolithic apps... So not having a 
lovely time so far...

H

openzwave-dev-watcher, 2016-05-14:
Some progress after reading the config files... And playing with both the 
OZW control panel and domoticz. It would appear that there are 8 groups 
available on the device itself. (I think there's at least 1 post eluding to 
this somewhere floating around). And you set those to enable the various 
reports... Group 4 and Group 7 are the destination of power meter reports 
for the various meters. 

However setting them doesn't make the reported instance sane still... You 
get reports without polling, however OZW still logs the device report as 
instance 1 and the relays as instance 1 & 2. Meaning relay2 report is 
correct, but relay1 alternates between the relay 1 value, and the device 
value. e.g.

2016-05-14 10:10:32.170 Info, Node003, Received Meter report from node 3: 
instance 2: Power=7.5W

2016-05-14 10:10:35.705 Info, Node003, Received Meter report from node 3: 
instance 1: Power=94.6W

2016-05-14 10:10:36.257 Info, Node003, Received Meter report from node 3: 
instance 2: Power=7.7W

2016-05-14 10:10:43.065 Info, Node003, Received Meter report from node 3: 
instance 1: Power=102.7W

2016-05-14 10:10:44.066 Info, Node003, Received Meter report from node 3: 
instance 1: Power=102.0W

2016-05-14 10:10:45.067 Info, Node003, Received Meter report from node 3: 
instance 1: Power=102.5W

2016-05-14 10:10:49.071 Info, Node003, Received Meter report from node 3: 
instance 1: Power=103.8W

2016-05-14 10:10:54.078 Info, Node003, Received Meter report from node 3: 
instance 1: Power=102.1W

2016-05-14 10:10:55.075 Info, Node003, Received Meter report from node 3: 
instance 1: Power=102.4W

This is with 2 loads. 1x ~95W and 1x~8W. You can see that the total is 
reported as ~102W on instance 1 and the relays are 94W and 7W on instance 1 
& 2... I believe the device itself assumes you'll associate each group with 
a separate instance at the controller. But IIUC OZW only implements 1 
instance. And although the traffic comes in, I haven't found a way yet to 
detect the destination instance and log it (And then do something sane with 
it to map the values to something that works correctly). 

H

openzwave-dev-watcher, 2016-05-14:
Deleting the endpoint mapping from the ZML config for the device makes it 
much better though...

2016-05-14 10:29:15.663 Info, Node003, Received Meter report from node 3: 
instance 2: Energy=0.9kWh
2016-05-14 10:29:36.253 Info, Node003, Received Meter report from node 3: 
instance 3: Energy=0.1kWh
2016-05-14 10:29:36.421 Info, Node003, Received Meter report from node 3: 
instance 2: Power=93.4W
2016-05-14 10:29:36.837 Info, Node003, Received Meter report from node 3: 
instance 3: Power=7.6W
2016-05-14 10:29:42.748 Info, Node003, Received Meter report from node 3: 
instance 3: Power=8.1W
2016-05-14 10:29:43.636 Info, Node003, Received Meter report from node 3: 
instance 1: Power=100.0W
2016-05-14 10:30:36.306 Info, Node003, Received Meter report from node 3: 
instance 2: Power=93.9W
2016-05-14 10:30:36.858 Info, Node003, Received Meter report from node 3: 
instance 3: Power=7.5W
2016-05-14 10:30:43.666 Info, Node003, Received Meter report from node 3: 
instance 1: Power=100.5

To the point where I'd have to question why there's any such config in the 
first place... Or is it that the mapping is somehow incorrect?

>

openzwave-dev-watcher, 2016-05-14:
On Saturday, 14 May 2016 11:31:58 UTC+1, Hamish Marson wrote:
>
>
> Deleting the endpoint mapping from the ZML config for the device makes it 
> much better though...
>
> [deleted]
 

>
> To the point where I'd have to question why there's any such config in the 
> first place... Or is it that the mapping is somehow incorrect?
>
>
>
Maybe... I'd have to do some more checking (I've only looked at his WITHOUT 
the mapping enabled), but I THINK that the actual instance readings are 
always sent as  MultiChannelEncap, while the device reading is not... 
Suggesting that when the endpoint mapping is enabled, OZW is ASSUMING that 
un-encapsulated (device) reports are instance 1... And not differentiating 
between the real instance 1 (encapsulated) and the device...

jotakar, 2016-05-14:
After Reading your post, I undo my patches (refresh in config file) and ... well if it wait for different instances of controller let give it difference instances: I make an association in node 6 (Qubino 2D) from group 4 to 1.1 and from group 7 to 1.1 (node 1 is my controller) . And I don't know why but it's working now. If I switch on the relay 1 or 2 or  and 2 and I get 

2016-05-14 20:03:45.926 Detail, Node006,   Received: 0x01, 0x14, 0x00, 0x04, 0x00, 0x06, 0x0e, 0x60, 0x0d, 0x01, 0x01, 0x32, 0x02, 0x21, 0x34, 0x00, 0x00, 0x00, 0x83, 0x00, 0x00, 0x2c
2016-05-14 20:03:45.926 Detail, 
2016-05-14 20:03:45.927 Info, Node006, Received a MultiChannelEncap from node 6, endpoint 1 for Command Class COMMAND_CLASS_METER
2016-05-14 20:03:45.927 Detail, Node006, Refreshed Value: old value=false, new value=false, type=bool
2016-05-14 20:03:45.927 Detail, Node006, Changes to this value are not verified
2016-05-14 20:03:45.927 Info, Node006, Received Meter report from node 6: Power=13.1W
2016-05-14 20:03:45.927 Detail, Node006, Refreshed Value: old value=15.0, new value=13.1, type=decimal
2016-05-14 20:03:45.927 Detail, Node006, Changes to this value are not verified
2016-05-14 20:03:45.928 Detail, Node006, Notification: ValueChanged
2016-05-14 20:03:45.928 Detail, Node006, Notification: ValueChanged
.....
Several total readings 
.....
Node006,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x06, 0x0a, 0x32, 0x02, 0x21, 0x34, 0x00, 0x00, 0x01, 0x4b, 0x00, 0x00, 0x88
2016-05-14 20:04:21.148 Detail, 
2016-05-14 20:04:21.148 Detail, Node006, Refreshed Value: old value=false, new value=false, type=bool
2016-05-14 20:04:21.148 Detail, Node006, Changes to this value are not verified
2016-05-14 20:04:21.148 Info, Node006, Received Meter report from node 6: Power=33.1W
2016-05-14 20:04:21.149 Detail, Node006, Refreshed Value: old value=36.1, new value=33.1, type=decimal
2016-05-14 20:04:21.149 Detail, Node006, Changes to this value are not verified
2016-05-14 20:04:21.150 Detail, Node006, Notification: ValueChanged
2016-05-14 20:04:21.150 Detail, Node006, Notification: ValueChanged
.............
A new Reading now endpoint 2 (my second relay)
2016-05-14 20:04:21.255 Info, Node006, Received a MultiChannelEncap from node 6, endpoint 2 for Command Class COMMAND_CLASS_METER
2016-05-14 20:04:21.255 Detail, Node006, Refreshed Value: old value=false, new value=false, type=bool
2016-05-14 20:04:21.255 Detail, Node006, Changes to this value are not verified
2016-05-14 20:04:21.255 Info, Node006, Received Meter report from node 6: Power=21.9W
2016-05-14 20:04:21.255 Detail, Node006, Refreshed Value: old value=21.8, new value=21.9, type=decimal
2016-05-14 20:04:21.256 Detail, Node006, Changes to this value are not verified
2016-05-14 20:04:21.256 Detail, Node006, Notification: ValueChanged
2016-05-14 20:04:21.256 Detail, Node006, Notification: ValueChanged

Then I have instant report of power meter!!!!
Please confirm if you get the same behaviour.

openzwave-dev-watcher, 2016-05-14:
On Saturday, 14 May 2016 14:01:11 UTC+1, Hamish Marson wrote:
> Maybe... I'd have to do some more checking (I've only looked at his
> WITHOUT the mapping enabled), but I THINK that the actual instance readings
> are always sent as  MultiChannelEncap, while the [...]
Sorry if this gets tedious... But in case I get bored and stop, someone 
else will know what's been investigated...

If endpoint mapping is re-enabled, it looks like something in OZW decides 
that in the multi-channel encapsulated data, the instances are now 1 & 2 
instead of 2 & 3... 

This kind of suggests that OZW is getting it wrong. IIUC the xml files are 
produced by the vendor. And the vendor has inserted the endpoint mapping? 
So having OZW try to interpret that as 2 switches without the master would 
suggest that it needs rectification...

openzwave-dev-watcher, 2016-05-14:
Yes, that's correct. However unless you disable endpoint mapping (Remove it 
from the XML for the device) you get both the device and relay 1 reported 
as instance 1. This confuses Domoticz as it can't tell the difference 
between messages for the device as instance 1, or relay 1 as instance 1.

When endpoint mapping is DISABLED you get the following. Debugging shows 
that the multiinstanceencap comes through as instance1 and 2 (2 in example 
below).. But then is apparently incremented by OZW to instance 2 and 3 (3 
shown below) when passed up the stack from the multichannelencap 
decapsulation to the Meter. (I added the instance into the logs in 
Meter.cpp to see it - without it you just can't tell what's happening, so I 
might do a quick commit for that). e.g.

2016-05-14 20:24:13.192 Info, Node003, Received a MultiChannelEncap from 
node 3, endpoint 2 for Command Class COMMAND_CLASS_METER
2016-05-14 20:24:13.192 Detail, Node003, Refreshed Value: old value=false, 
new value=false, type=bool
2016-05-14 20:24:13.193 Detail, Node003, Changes to this value are not 
verified
2016-05-14 20:24:13.193 Info, Node003, Received Meter report from node 3: 
instance 3: Power=7.5W
2016-05-14 20:24:13.194 Detail, Node003, Refreshed Value: old value=7.7, 
new value=7.5, type=decimal

The  report for meter 2 comes in as a multichannelencap for endpoint 2, but 
is reported as instance 3. Which works for domoticz because it then sees 3 
switches and power reports. Which is what you really want.

H

jotakar, 2016-05-14:
Yes, It seems so. If I see config zwzcfg_0x[homeid]_ I see Q1 (I1, endpoint 1) is instance 2 and Q2 is instance 3
                <Instance index="1" />
                <Instance index="2" endpoint="1" />
                <Instance index="3" endpoint="2" />
But indeed I don't know why the system works after this "association trick" (associate to node 1.1, that doesn't exist).
I'll try it with a Phiiio PAN04 that i have with the same issue, but tomorrow ;)

travellingkiwi, 2016-05-15:
When an unencapsulated (global) report is received from the qubino, does OZW ASSUME instance 1? or is it being specified in the report? I'm suspecting OZW is just assuming that anything unencapsulated is automatically instance 1.

Assuming that OZW is doing the assuming (A double assumption. What could go wrong), would it would be better if unencapsulated traffic was set as instance 0? Thus there would be no mixup required for encapsulated instances. And less chance for error.

jotakar, 2016-05-15:
With associations now my Qubino Works as I wanted:
Shows actual status of switches, even if it are turned on/off by hand in physical switch
Total relay (instance 1, endpoint, 0?) Works fine
Shows actual power consume for ever relay and total
Associations are: every group associated to a 1.1 node (1 controller, .1 instance). The mistery  for me is that I has a controlles in node 1 which has no instances  (ZWCP doesn't show instances for it)

I tried the same trick with a Philio PAN04, but it doesn't accept instances associations, so I can't associate a 1.1. Then all readings go to instance 1 of controller and this cannot distinguish from where are coming readings. 
Why PAN04 doesn't send encapsulated (node+endpoint) readings to the controller? Or why OZWave doesn't read this info?

I'm using Domoticz with an Aeotec Stick GEN5.

rofrantz, 2016-05-16:
Hi @jotakar, any chance to repaird also the older version of Qubino double switch relay, the model ZMNHBA2 ?

jotakar, 2016-05-16:
I have not any ZMNHBA2 so I can't check it. But you can try this modif if you have this module :).
The only thing to add is that the controller I used is a Aeon Stick GEN5.

rofrantz, 2016-05-16:
I also have a Aeon Gen5 stick so that should not be a problem.
What config exactly should I try ?

Fishwaldo, 2016-05-16:
Hi,
A lot of things to answer here:
@travellingkiwi

> When an unencapsulated (global) report is received from the qubino, does OZW ASSUME instance 1? or is it being specified in the report? I'm suspecting OZW is just assuming that anything [...]

The "core" Z-Wave Protocol doesn't support instances... it assumes only one type of command class per device, so on devices like this, instances above 1 (eg, 2 & 3) should always be encapsulated via the MultiInstance Command Class. In the case of the devices here, if we do a SET or GET on the device, it correctly replies with the results encapsulated in a MultiInstance class which is why these work.

But it doesn't encapsulate REPORT messages sent by the device, so there is no way to determine which instance a REPORT message corresponds to. Every other manufacturer that we deal with does REPORTS encapsulated in a multi-instance class (and I've had someone tell me in private that the official specs from SIGMA mandate this type of communication.

I raised the issue with QUBINO previously, and they replied the controller should support MultiInstanceAssociation (a different command class), and each group associated with a different instance on the controller. I pointed out this was different from every other manufacturer, but I suspect it got "lost in translation" - The guy I was chatting with was not a native english speaker..

We only support "Setting" MultiInstanceAssociations (as @jotakar found out with the 1.1 association) on devices that support it, but don't support "receiving" REPORT messages to different "instances" on OZW. It just still processes everything instance 1 anyway.

About the EndPoint Mappings:
this was added to OZW a long time ago, as some devices reported "endpoints" starting at 0 and some started at 1. When mapping to Instances, this screwed things up. Quibino submitted the device config files to us with the EndPoint Mapping option present, so I assumed they had researched this and determined they needed it (I don't have any of their devices so I couldn't confirm it).

About the PAN devices:
If they don't support the MultiInstanceAssociation Commandclass then associating with 1.1 wont work, as it will fall back to the old Association CommandClass which will just transmit 1 to the node.

> Why PAN04 doesn't send encapsulated (node+endpoint) readings to the controller? Or why OZWave doesn't read this info?

If the message is not encapsulated, then there is no Instance value in the transmitted message, so we can't determine what instance its for. This sounds like a real bug in the PAN device, rather than a a different approach like Quibino have taken.

With all the research and troubleshooting you guys have done, do you mind posting a "this is how I got it working" post at http://discourse.db.openzwave.com/t/qubino-zmnhbdx-flush-2-relays/114 so others in the future can make it work.

Finally, 
Probably more feedback to the app developers than users, but you should not assume different instances on different command classes match each other. (eg, Meter 1 -> relay 2). Nothing mandates this as far as I know, and its possible in the future this might not be true for all devices. 

Please also let me know if any final updates are required on the config file before I submit the changes previously. it would be good if you can post the final working version for verification.

openzwave-dev-watcher, 2016-05-16:
On Monday, 16 May 2016 15:50:25 UTC+1, Justin Hammond wrote:
> Hi,
> A lot of things to answer here:
> @travellingkiwi <https://github.com/travellingkiwi>
>
> When an unencapsulated (global) report is received from the qubino, does
> OZW ASSUME instance 1? or is it [...]

Maybe Qubino updated their firmware... but the reports generated when you 
set group 4 & 7 (Doesn't matter if you set it to multi instance association 
via 1.4 & 1.7 or just both 1.1, or just 1) are logged by OZW as being 
multichannelencap.. e.g.

*2016-05-14 20:24:13.192 Info, Node003, Received a MultiChannelEncap from 
node 3, endpoint 2 for Command Class COMMAND_CLASS_METER*
2016-05-14 20:24:13.192 Detail, Node003, Refreshed Value: old value=false, 
new value=false, type=bool
2016-05-14 20:24:13.193 Detail, Node003, Changes to this value are not 
verified
2016-05-14 20:24:13.193 Info, Node003, Received Meter report from node 3: 
instance 3: Power=7.5W
2016-05-14 20:24:13.194 Detail, Node003, Refreshed Value: old value=7.7, 
new value=7.5, type=decimal

Note that there appears to be a bit of confusion over the qubino sending 
reports. They're supported. But not configured by default. You configure 
groups 4 and 7 to enable meter reports for relay1 and relay 2. Note that 
the lack of reporting instance in the logs didn't help the confusion. It 
really didn't start to become clear what was happening until I put that in.
 

> I raised the issue with QUBINO previously, and they replied the controller
> should support MultiInstanceAssociation (a different command class), and
> each group associated with a different instance [...]
 
Not a problem. It works as is for the Qubino.

About the EndPoint Mappings:
> this was added to OZW a long time ago, as some devices reported
> "endpoints" starting at 0 and some started at 1. When mapping to Instances,
> this screwed things up. Quibino submitted the device [...]

Ahh.. Well, it certainly doesn't :) I'll do a  new fork and update the XML 
without the endpoint mapping,.
 

> About the PAN devices:
> If they don't support the MultiInstanceAssociation Commandclass then 
> associating with 1.1 wont work, as it will fall back to the old Association 
> CommandClass which will just transmit 1 to the node.
>

PAN devices? 

H

openzwave-dev-watcher, 2016-05-16:
As mention on the GitHub issue just now, if the original message wasn’t encapsulated with a MultiInstance CommandClass, there is no way to determine what instance its from. 

 

From: OpenZWave <openzwave@googlegroups.com>
Reply-To: OpenZWave <openzwave@googlegroups.com>
Date: Saturday, 14 May 2016 at 6:26 AM
To: OpenZWave <openzwave@googlegroups.com>
Cc: <open-zwave@noreply.github.com>, <reply@reply.github.com>, <notifications@github.com>
Subject: [openzwave] Re: [OpenZWave/open-zwave] Again ZMNHBDx (#880)

 

I'v been looking at this too. Using Domoticz. And the Qubino support results are pretty screwey

 

Using the latest code from github for both openzwave and domoticz and it's reasonably clear to me that OZW always seems the instance as 0 when receiving meter reports from the Qubino. A quick bit of editing to print the _instance when logging the meter readings and it doesn't matter which relay is on, or both the _instance is always reported as 1. e.g.

 

Relay1 (98W) (Grepping for 'Meter report|*HN*' from the OZW log

 

2016-05-13 22:13:37.679 Detail, Node003,   **HN** Received: 0x01, 0x07, 0x00, 0x13, 0x4f, 0x00, 0x00, 0x0f, 0xab

2016-05-13 22:13:37.689 Detail, Node003,   **HN** Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8

2016-05-13 22:13:37.849 Detail, Node003,   **HN** Received: 0x01, 0x07, 0x00, 0x13, 0x50, 0x00, 0x00, 0x10, 0xab

2016-05-13 22:13:37.972 Detail, Node003,   **HN** Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x03, 0x07, 0x60, 0x0d, 0x01, 0x01, 0x25, 0x03, 0xff, 0x46

2016-05-13 22:13:37.974 Info, Node003, Received a MultiChannelEncap from node 3, endpoint 1 for Command Class COMMAND_CLASS_SWITCH_BINARY

2016-05-13 22:13:39.112 Detail, Node003,   **HN** Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x03, 0x03, 0x20, 0x01, 0xff, 0x2c

2016-05-13 22:13:40.115 Detail, Node003,   **HN** Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x03, 0x03, 0x25, 0x03, 0xff, 0x

2016-05-13 22:13:41.576 Detail, Node003,   **HN** Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x03, 0x03, 0x25, 0x03, 0xff, 0x2b

2016-05-13 22:13:42.607 Detail, Node003,   **HN** Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x03, 0x0a, 0x32, 0x02, 0x21, 0x34, 0x00, 0x00, 0x03, 0xa8, 0x00, 0x00, 0x6c

2016-05-13 22:13:42.608 Info, Node003, Received Meter report from node 3: instance 1: Power=93.6W

 

 

Then I turn off relay 1 and turn on relay 2... Which is ~8W

 

016-05-13 22:16:35.184 Info, Node003, Value::Set - COMMAND_CLASS_SWITCH_BINARY - Switch - 0 - 2 - True

2016-05-13 22:16:35.194 Detail, Node003,   **HN** Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8

2016-05-13 22:16:35.342 Detail, Node003,   **HN** Received: 0x01, 0x07, 0x00, 0x13, 0x53, 0x00, 0x00, 0x0f, 0xb7

2016-05-13 22:16:35.352 Detail, Node003,   **HN** Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8

2016-05-13 22:16:35.512 Detail, Node003,   **HN** Received: 0x01, 0x07, 0x00, 0x13, 0x54, 0x00, 0x00, 0x10, 0xaf

2016-05-13 22:16:35.632 Detail, Node003,   **HN** Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x03, 0x07, 0x60, 0x0d, 0x02, 0x01, 0x25, 0x03, 0xff, 0x45

2016-05-13 22:16:35.633 Info, Node003, Received a MultiChannelEncap from node 3, endpoint 2 for Command Class COMMAND_CLASS_SWITCH_BINARY

2016-05-13 22:16:36.036 Detail, Node003,   **HN** Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x03, 0x03, 0x25, 0x03, 0xff, 0x2b

2016-05-13 22:16:37.066 Detail, Node003,   **HN** Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x03, 0x0a, 0x32, 0x02, 0x21, 0x34, 0x00, 0x00, 0x00, 0x56, 0x00, 0x00, 0x91
[... 24 lines cut ...]

2016-05-13 22:18:59.071 Detail, Node003,   **HN** Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x03, 0x07, 0x60, 0x0d, 0x01, 0x01, 0x25, 0x03, 0xff, 0x46

2016-05-13 22:18:59.072 Info, Node003, Received a MultiChannelEncap from node 3, endpoint 1 for Command Class COMMAND_CLASS_SWITCH_BINARY

2016-05-13 22:18:59.963 Detail, Node003,   **HN** Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x03, 0x03, 0x20, 0x01, 0xff, 0x2c

2016-05-13 22:19:00.956 Detail, Node003,   **HN** Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x03, 0x03, 0x25, 0x03, 0xff, 0x2b

2016-05-13 22:19:02.418 Detail, Node003,   **HN** Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x03, 0x03, 0x25, 0x03, 0xff, 0x2b

2016-05-13 22:19:03.448 Detail, Node003,   **HN** Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x03, 0x0a, 0x32, 0x02, 0x21, 0x34, 0x00, 0x00, 0x03, 0xef, 0x00, 0x00, 0x2b

2016-05-13 22:19:03.450 Info, Node003, Received Meter report from node 3: instance 1: Power=100.7W

 

 

H

-- 
You received this message because you are subscribed to the Google Groups "OpenZWave" group.
To unsubscribe from this group and stop receiving emails from it, send an email to openzwave+unsubscribe@googlegroups.com.
For more options, visit https://groups.google.com/d/optout.

Fishwaldo, 2016-05-16:
> Maybe Qubino updated their firmware... but the reports generated when you set group 4 & 7 (Doesn't matter if you set it to multi instance association via 1.4 & 1.7 or just both 1.1, or just 1) are [...]

Are you sure that is from a unsolicited REPORT (and not as a result of a GET) message?
Reason I ask is nothing actually changed, so kinda strange it would send a update if nothing is different.

> Note that there appears to be a bit of confusion over the qubino sending reports. They're supported. But not configured by default. You configure groups 4 and 7 to enable meter reports for relay1 [...]

This is why I'm starting up the discourse.db.openzwave.com forums. This type of knowledge is essential,but not a lot of users have the time or patients to figure it out. :)

Pan Devices: PAN04 device @jotakar mentioned.

Fishwaldo, 2016-05-16:
> Are you sure that is from a unsolicited REPORT (and not as a result of a GET) message?
Reason I ask is nothing actually changed, so kinda strange it would send a update if nothing is different.

Ignore me. I'm blind.

jotakar, 2016-05-16:
Perhaps I'm getting lost. : (
But if I do not associate my Qubino to 1.1 or 1.2, and associate it just to 1, I do not get the endpoint from the measurement was done (unsolicited REPORT). And of course, I am not using the TriggerRefreshValue. In this way I get power meter and relay switched by hand (in Wall switch).
Nevertheless with Philio (PAN04, sorry, I mentioned it without specify the device model)  I need to use this TriggerRefreshValue (a Kind of temporary polling), because it doen't support multiInstance association and sends all from node (not from endpoint). Now I see that this is the way Philio works. 
Of course GET reports always send node and endpoint, GET are made to every endpoint. It's like a manual refresh.

openzwave-dev-watcher, 2016-05-16:
On Monday, 16 May 2016 17:14:10 UTC+1, jotakar wrote:
>
> Perhaps I'm getting lost. : (
>

Nope you're not... It would appear that sometimes the qubino doesn't change 
it's behaviour when changing the groups. Originally I setup the groups as 
1.1, 1.2, 1.3 etc... That's when it started working. Then I changed groups 
4 and 7 to just 1. And it stayed working. 

When you posted this, I went back and removed all the groups... Put just 1 
on channel 4 & 7 and got the same result. Changed it to 1.4 and 1.7 and 
stayed broken... Until I switched things off and back on. It all started 
working again... Strange... I wonder if it's the order I did things 
before...

Sigh...

But if I do not associate my Qubino to 1.1 or 1.2, and associate it just to 
> 1, I do not get the endpoint from the measurement was done (unsolicited 
> REPORT). And of course, I am not using the TriggerRefreshValue. In this way 
> I get power meter and relay switched by hand (in Wall switch).
>

TriggerRefreshValue?

H

>

jotakar, 2016-05-16:
> TriggerRefreshValue?
Yes, in a post here (look for about 6th post) I spoke about use this "option" in config file for get an instant refresh. 
I reached to this because I wanted to get instant status of swtiches, although I turn it by hand (physical switch). I wanted to avoid use an refresh script and found this fix. I looke for info about it, didn't find anything but I got it working. 
I put it for command class 37 (swtich binary) and get status of switches and for command class 50, power meter and get power measurements of every relay when the global relay change. In Qubino Works but it's not necesary, association is better.
Thanks

kjamsek, 2016-05-17:
Hello,
in order to resolve the endpoint reporting confusion please first update your guide about adding devices to OZW ([https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices](url)) since it lacks a section about endpoint mapping entirely (to be honest, Z-Wave has an endpoint discovery mechanism via the Mutli Channel Command Class so no static mapping of endpoints should be needed in any case, especially since they can also be dynamically added/removed).

Qubino implemented multi channel reporting in such a way for maximum compatibility with controllers that do not support Multi Channel Command Class. So, upon inclusion and without setting any Multi Channel Associations to groups on endpoint devices (or a single Multi Channel Association on the root device's group 1) all reports sent to the controller will be unencapsulated. 

This enables the module to report data for all endpoints (as is the root device's job), and since the module cannot know whether a controller supports Multi Channel Command Class or not, they are sent unencapsulated. When the actual Multi Channel Association is set then the module will report encapsulated data with I1 being reported as endpoint 1 and I2 as endpoint 2. A connected temperature sensor to the module will make endpoint 3 be reported as supported by the module via Multi Channel Command Class' commands.

As far as unsolicited reporting goes, the same approach is used. Only this time the unsolicited report intervals or thresholds are set via configuration parameters.

Since i don't know OZW that well, can you give more information how this endpoint mapping is to be done correctly?

Fishwaldo, 2016-05-17:
@kjamsek 
> (to be honest, Z-Wave has an endpoint discovery mechanism via the Mutli Channel Command Class so no static mapping of endpoints should be needed in any case, especially since they can also be dynamically added/removed).

We already do the automatic discovery of the endpoints. The mapping option is the configuration files is due to some devices reporting endpoints that are off by 1 - ie, buggy devices. 

The code that decodes this mapping is here: https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/command_classes/MultiInstance.cpp#L420

We don't support mapping endpoints via config files, as this discovery works most of the time. 

> Qubino implemented multi channel reporting in such a way for maximum compatibility with controllers that do not support Multi Channel Command Class. So, upon inclusion and without setting any Multi [...]

Unfortunately, Qubino is the only one that has taken this approach. Every other device vendor will send unsolicited MultiChannel encap messages without having to specify any MultiChannelAssociation configuration... and honestly, if any controller is not supporting the MultiChannel CC now days, they are not going to work with most devices.... 

From a user/application perspective right now, if they don't set any MultiChannel Associations, all Unsolicited reports are sent to endpoint 0, and we cannot differentiate between the actual instance/endpoint that changed. We can trigger a refresh of the other instances/endpoint via a MultiChannel Encapsulated Get message, but the "users interface" (ie, the apps that use OZW) would have a "bogus" value for endpoint 0 while waiting for the refresh to come in.

its good to know if we set a MultiChannelAssociation then the devices will start unsolicated reporting in a MultiChannelEncap message - The challenge is that most software utilizing OZW don't support MultiChannelAssociation CC (we do support it in OZW, they have not caught up) so its still going to be "broken" for a lot of users.

rofrantz, 2016-05-17:
TLTR (endless logs, quotes on endless logs)...in conclusion, Qubino implementation for multi instance devices seems "different" than all other vendors.

Can something be done about it ? (of course it can) 
Will it be done ? Because I understand it would be a hackish way specially implemented for Qubino that might not even work for all of users :(

Fishwaldo, 2016-05-17:
please see https://github.com/OpenZWave/open-zwave/issues/857#issuecomment-219678976

Fishwaldo, 2016-08-11:
Closing this... If there are still issues, please discuss on the mailing list first.

closed 2016-08-11 by Fishwaldo