OpenZWave/open-zwave #882: OZW log doesn't list Units

travellingkiwi, 2016-05-09:
In the OZW log Node's are listed by ID, but updates don't include the unit. This makes it difficult to view at a glance which unit for devices like a Qubino 2x relay device are being reported/commanded.

Fishwaldo, 2016-05-10:
I don't understand. what is a "Unit?" 
Could you post a example?

travellingkiwi, 2016-05-10:
Possibly I'm interpreting terminology... The Unit would be for example the relay on a Qubino 2 relay switch... 

i.e. when reporting the power consumption on a Qubino 2 relay switch the Qubino (presumably) reports the value separately for each relay. But in the OZW logs it isn't made clear which of the relays (units) the reading is for, you simply get a power reading logged twice. For example at 8W and 98W.

H

Fishwaldo, 2016-05-11:
This is a known issue with Qubino Devices. They don't send the instance ID when sending updates so no way for us to know what is what. 

If you manually refresh, then you should see the instance ID's.
See #880

closed 2016-05-11 by Fishwaldo