OpenZWave/open-zwave #883: _instance is not logged when reporting node messages

travellingkiwi, 2016-05-14:
When logging information about a node (Received traffic, meter updates etc) the _instance is not reported. This makes it extremely difficult to diagnose issues on multi instance devices

travellingkiwi, 2016-05-14:
Two choices... Do we want something quick & dirty (e.g. edit Meter.cpp to add _instance at various places, OR do we want to do something a little more involved?

A little more involved could be an extra parameter to Log::Write to add the instance. This would then be passed through all the way to LogImpl::GetNodeString and be formatted as "Nodexxx:xxx" in the log files.

I favor Nodexxx:xxx but not sure what this may break along the way (e.g. The logAnalyzer on the website). Happy to do the code... In fact I'll probably start on option 2 now in the hopes someone else agrees with me.

Fishwaldo, 2016-05-16:
Closing this to discuss in the PR

closed 2016-05-16 by Fishwaldo