OpenZWave/open-zwave #885: Bogus Vision ZD2102-5 Alarm Reports mark open contact as closed

NotTheEvilOne, 2016-05-15:
Hi,

I have some door/window contacts of the type ZD2102-5 EU and found a rather odd issue: It looks like the device sends bogus "AlarmCmd_Get" replies if being asked while wake-up from the wake-up interval setting.

`2016-05-15 00:42:53.235 Info, Node020, Received Wakeup Notification from node 20
2016-05-15 00:42:53.235 Info, Node020,   Node 20 has been marked as awake
2016-05-15 00:42:53.235 Detail, Node020, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_ALARM
2016-05-15 00:42:53.235 Detail, Node020, Queuing (WakeUp) AlarmCmd_Get (Node=20): 0x01, 0x0c, 0x00, 0x13, 0x14, 0x05, 0x71, 0x04, 0x00, 0x06, 0x01, 0x25, 0x7a, 0xdc
2016-05-15 00:42:53.235 Detail, Node020, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_ALARM
2016-05-15 00:42:53.235 Detail, Node020, Queuing (WakeUp) AlarmCmd_Get (Node=20): 0x01, 0x0c, 0x00, 0x13, 0x14, 0x05, 0x71, 0x04, 0x00, 0x07, 0x01, 0x25, 0x7b, 0xdc
2016-05-15 00:42:53.235 Detail, Node020, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_ALARM
2016-05-15 00:42:53.235 Detail, Node020, Queuing (WakeUp) AlarmCmd_Get (Node=20): 0x01, 0x0c, 0x00, 0x13, 0x14, 0x05, 0x71, 0x04, 0x00, 0x08, 0x01, 0x25, 0x7c, 0xd4
2016-05-15 00:42:53.236 Detail, Node020, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_ALARM
2016-05-15 00:42:53.236 Detail, Node020, Queuing (WakeUp) AlarmCmd_Get (Node=20): 0x01, 0x0c, 0x00, 0x13, 0x14, 0x05, 0x71, 0x04, 0x00, 0x0b, 0x01, 0x25, 0x7d, 0xd6
2016-05-15 00:42:53.236 Detail, Node020, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_ALARM
2016-05-15 00:42:53.236 Detail, Node020, Queuing (WakeUp) AlarmCmd_Get (Node=20): 0x01, 0x0c, 0x00, 0x13, 0x14, 0x05, 0x71, 0x04, 0x00, 0x0d, 0x01, 0x25, 0x7e, 0xd3
2016-05-15 00:42:53.236 Detail, Node020, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_BATTERY
2016-05-15 00:42:53.236 Detail, Node020, Queuing (WakeUp) BatteryCmd_Get (Node=20): 0x01, 0x09, 0x00, 0x13, 0x14, 0x02, 0x80, 0x02, 0x25, 0x7f, 0x2b
2016-05-15 00:42:53.236 Detail, Node020, Queuing (Query) Query Stage Complete (Dynamic)
2016-05-15 00:42:53.236 Detail, Node020, Notification: Notification - Node Awake
[...]
2016-05-15 00:42:53.838 Info, Node020, Received Alarm report: type=6, level=0, sensorSrcID=0, type:Access Control event:254, status=255
2016-05-15 00:42:53.838 Detail, Node020, Initial read of value
2016-05-15 00:42:53.838 Detail, Node020, Initial read of value
2016-05-15 00:42:53.838 Detail, Node020, Initial read of value
2016-05-15 00:42:53.838 Detail, Node020, Initial read of value
2016-05-15 00:42:53.838 Detail, Node020,   Expected reply and command class was received
2016-05-15 00:42:53.839 Detail, Node020,   Message transaction complete
2016-05-15 00:42:53.839 Detail, 
2016-05-15 00:42:53.839 Detail, Node020, Removing current message
2016-05-15 00:42:53.839 Detail, Node020, Notification: ValueChanged
2016-05-15 00:42:53.842 Detail, Node020, Notification: ValueChanged
2016-05-15 00:42:53.845 Detail, Node020, Notification: ValueChanged
2016-05-15 00:42:53.845 Detail, Node020, Notification: ValueChanged`

The correct Alarm report is sent when being triggered by the integrated sensors:

`2016-05-15 09:37:36.412 Info, Node020, Received Alarm report: type=6, level=0, sensorSrcID=0, type:Access Control event:23, status=255
2016-05-15 09:37:36.412 Detail, Node020, Initial read of value
2016-05-15 09:37:36.412 Detail, Node020, Initial read of value
2016-05-15 09:37:36.413 Detail, Node020, Initial read of value
2016-05-15 09:37:36.413 Detail, Node020, Initial read of value
2016-05-15 09:37:36.413 Detail, Node020, Notification: ValueChanged
2016-05-15 09:37:36.417 Detail, Node020, Notification: ValueChanged
2016-05-15 09:37:36.418 Detail, Node020, Notification: ValueChanged
2016-05-15 09:37:36.419 Detail, Node020, Notification: ValueChanged`

Is there a possibility (configuration switch?) to disable the Alarm Report being processed after receiving the wake-up notification?

Fishwaldo, 2016-05-16:
Could you post the entire log file please?

NotTheEvilOne, 2016-05-16:
Hi,

Please find attached the grep'd log for Node020 [OZW_Node020_Log.txt](https://github.com/OpenZWave/open-zwave/files/266391/OZW_Node020_Log.txt). Let me know if you need any additional information.

Fishwaldo, 2016-05-16:
Nothing wrong here.
The Alarm_Get messages are from OZW starting up, and querying for the initial values of the Alarm CC:
`2016-05-14 12:42:16.141 Detail, Node020, Queuing (Send) SensorBinaryCmd_Get (Node=20): 0x01, 0x09, 0x00, 0x13, 0x14, 0x02, 0x30, 0x02, 0x25, 0x79, 0x9d
2016-05-14 12:42:16.141 Detail, Node020, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_ALARM
2016-05-14 12:42:16.141 Detail, Node020, Queuing (Send) AlarmCmd_Get (Node=20): 0x01, 0x0c, 0x00, 0x13, 0x14, 0x05, 0x71, 0x04, 0x00, 0x06, 0x01, 0x25, 0x7a, 0xdc
2016-05-14 12:42:16.141 Detail, Node020, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_ALARM
2016-05-14 12:42:16.141 Detail, Node020, Queuing (Send) AlarmCmd_Get (Node=20): 0x01, 0x0c, 0x00, 0x13, 0x14, 0x05, 0x71, 0x04, 0x00, 0x07, 0x01, 0x25, 0x7b, 0xdc
2016-05-14 12:42:16.142 Detail, Node020, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_ALARM
2016-05-14 12:42:16.142 Detail, Node020, Queuing (Send) AlarmCmd_Get (Node=20): 0x01, 0x0c, 0x00, 0x13, 0x14, 0x05, 0x71, 0x04, 0x00, 0x08, 0x01, 0x25, 0x7c, 0xd4
2016-05-14 12:42:16.142 Detail, Node020, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_ALARM
2016-05-14 12:42:16.142 Detail, Node020, Queuing (Send) AlarmCmd_Get (Node=20): 0x01, 0x0c, 0x00, 0x13, 0x14, 0x05, 0x71, 0x04, 0x00, 0x0b, 0x01, 0x25, 0x7d, 0xd6
2016-05-14 12:42:16.142 Detail, Node020, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_ALARM
2016-05-14 12:42:16.142 Detail, Node020, Queuing (Send) AlarmCmd_Get (Node=20): 0x01, 0x0c, 0x00, 0x13, 0x14, 0x05, 0x71, 0x04, 0x00, 0x0d, 0x01, 0x25, 0x7e, 0xd3
`
But the device goes to sleep before processing these messages, so they are moved to the WakeUp Queue:

`2016-05-14 12:42:32.420 Info, Node020, Node not responding - moving message to Wake-Up queue: AlarmCmd_Get (Node=20): 0x01, 0x0c, 0x00, 0x13, 0x14, 0x05, 0x71, 0x04, 0x00, 0x06, 0x01, 0x25, 0x7a, 0xdc
2016-05-14 12:42:32.420 Info, Node020, Node not responding - moving message to Wake-Up queue: AlarmCmd_Get (Node=20): 0x01, 0x0c, 0x00, 0x13, 0x14, 0x05, 0x71, 0x04, 0x00, 0x07, 0x01, 0x25, 0x7b, 0xdc
2016-05-14 12:42:32.420 Info, Node020, Node not responding - moving message to Wake-Up queue: AlarmCmd_Get (Node=20): 0x01, 0x0c, 0x00, 0x13, 0x14, 0x05, 0x71, 0x04, 0x00, 0x08, 0x01, 0x25, 0x7c, 0xd4
2016-05-14 12:42:32.420 Info, Node020, Node not responding - moving message to Wake-Up queue: AlarmCmd_Get (Node=20): 0x01, 0x0c, 0x00, 0x13, 0x14, 0x05, 0x71, 0x04, 0x00, 0x0b, 0x01, 0x25, 0x7d, 0xd6
2016-05-14 12:42:32.420 Info, Node020, Node not responding - moving message to Wake-Up queue: AlarmCmd_Get (Node=20): 0x01, 0x0c, 0x00, 0x13, 0x14, 0x05, 0x71, 0x04, 0x00, 0x0d, 0x01, 0x25, 0x7e, 0xd3
2016-05-14 12:42:32.420 Info, Node020, Node not responding - moving message to Wake-Up queue: BatteryCmd_Get (Node=20): 0x01, 0x09, 0x00, 0x13, 0x14, 0x02, 0x80, 0x02, 0x25, 0x7f, 0x2b
2016-05-14 12:42:32.420 Info, Node020, Node not responding - moving QueryStageComplete command to Wake-Up queue
`

And the device sleeps a few more times before these messages finally get through to the device.

Also, a Tip. I wouldn't include this device "securely" - it doesn't support beaming, and uses the Security Command Class which means extra traffic everytime it wants to send a alert - and honestly, i've ssen lots of issues around sleeping devices using security without supporting beaming (how many acronyms can I get in a sentence?)

Id just include it non-securely (meaning no encryption only). You will get better battery life and improved response times.... :)

Fishwaldo, 2016-05-16:
(damn github formatting... )

closed 2016-05-16 by Fishwaldo

NotTheEvilOne, 2016-05-16:
Hope it stays closed but you receive my question. The issue I'm facing is that it still has the "opened" state (from last opening the door) when I restarted the controller for debugging but in the night while being active based on the wake-up interval it all the sudden reports "closed".

From what you see, is this a firmware issue? And if it is the case, does it make sense to work around it?

Fishwaldo, 2016-05-17:
Thats because when you restart - the device goes to sleep before its able to refresh the Alarm values... hence why its reporting open. I guess in the middle of the night, it wakes up, we finally get to refresh those values, and it now reports the correct value.

This is just a limitation of Sleeping Devices. We can't talk to them when they are asleep.

Fishwaldo, 2016-05-17:
you can't trust any values from a Node till it reaches a QueryStage_Complete

NotTheEvilOne, 2016-05-18:
Hi,

Thanks for the explanation about QueryStage_Complete. This time I have added a debug log
[debug log](https://github.com/OpenZWave/open-zwave/files/270037/OZW_Log.Node021.txt)
 session and made sure that I close and open the contact (20:38:03.280) after "QueryStage_Complete" (20:37:09.352).

Less than 3 hours later it felt back to the closed state once again (23:04:54.101).

Based on the [vendor documentation](http://products.z-wavealliance.org/MarketCertification/File?folder=&amp;filename=MarketCertificationFiles/988/ZD%202102_04162014%20-%20500%E7%B3%BB%E5%88%97.pdf) it should also use Notification V4 Type "COMMAND_CLASS_NOTIFICATION_V4 (Mapping COMMAND_CLASS_BASIC)" but nothing regarding that appears in the log.

I wonder if the removal of unused Alarm types like 0x08, 0x09 and 0x0b could help in chasing down the issue or circumvent the device to fall asleep? At least it does not answer each time type 0x08 is queried (or OZW is waiting for its answer).

I'll leave the issue as closed for you to decide but still think there is a bug somewhere here.

NotTheEvilOne, 2016-05-18:
An additional node: Based on the vendor documentation Alarm Report event 0x16 is open, 0x17 is close. At 23:04:54.101 it receives a Report event 0xfe which is even not documented and might be misinterpreted as closed (as value is 0x00) but may mean "nothing changed, state unknown".

Is there a way to ignore this event type for this device based on the configuration? This might be an additional way to fix the issue (beside removing the requests for 0x08 and higher).

nechry, 2016-05-18:
the config file contain
`
<CommandClass id="32" setasreport="true"/>`

I'm suppose you receive Sensor report on CC 0x30 ?

NotTheEvilOne, 2016-05-18:
I guess I should have had a closer look in the log file. For the correct reports (close and open changes) there is:

Detail, Node021, Decrypted Packet: 0x00, 0x20, 0x01, 0x00
Info, Node021, Received Basic set from node 21: level=0. Treating it as a Basic report.
Detail, Node021, Refreshed Value: old value=255, new value=0, type=byte
Detail, Node021, Changes to this value are not verified
Detail, Node021, Notification: ValueChanged

I guess using <CommandClass id="113" action="remove" /> would somehow resolve the issue as only the Basic set will be used for open/close changes afterwards but this seems to be a bad workaround. I wonder if "getsupported" to "false" for command class 113 would result in correct behaviour.

NotTheEvilOne, 2016-05-26:
Hi developers,

I can now verify that the undocumented Report event 0xfe is misinterpreted by OpenZWave (because of value 0x00) as contact being closed. Please find attached a [patch](https://github.com/OpenZWave/open-zwave/files/283526/000-disable-zd2102-alarmcmd_get.patch.txt) to disable the AlarmCmd_Get for this device type. Furthermore I think this issue is no longer a question, even thought Fishwaldo's answer was very detailed and helpful in interpreting the logs :)