OpenZWave/open-zwave #886: ZMNHID1: set node association

jotakar, 2016-05-16:
Some days ago I got that this thermostat of Qubino Works fine. I coul set the setpoint, read power, temperature. All It was fine.
But.... happyness is not forever :-(
I deleted the association between node1 (controller) and group1 of the device and now I can't set it again. I tried using my host software (Domoticz) and the Zensys tools of Aeotec: the command_class is sent ok, but the assotiation is not set. I always get that the node of the module is not associated.
The device Works: turn ON/OFF, send temeperature reports, but send an erroneus value of power (I deleted the association trying to find the problem of this energy reading).
Then I wonder if there is any way to reset the device without exclude/include. I've seen a comand class (0x5A, COMMAND_CLASS_DEVICE_RESET_LOCALLY) but I can't find any informationa about wht it does and its params. Sometimes basic things are not easy to find.
Because I can't believe that this is a malfunction of the device, devices ar binaries Works or not works  :)  ... or perhaps it is?
Thanks

kjamsek, 2016-05-17:
Hello.
According to the Z-Wave protocol the association group 1 should always be set because it represents the main reporting group to the controller for Z-Wave devices. In case you cannot set it manually, i would suggest a reinclusion of the module since that association group is always set in the inclusion process.
As far as i know Qubino modules also support split reporting via association groups/multi channel association groups, so instead of deleting the lifeline group 1 on the root device try enabling the other ones. Check your device's manual for the actual group numbers and values.
The command class device reset locally is used when the user locally resets the module (and as such the module also clears it's node and home Ids), but according to my knowledge no controller actually uses this mechanism as it was intended (auto deletion of the module upon packet received since it becomes excluded on the module's side).

jotakar, 2016-05-17:
Yes. I had to exclude/include... for 5 times. Every I reinclude the device maintain the last config and the association... Finally I got a reset... Now the problema is that the power meter is not the real power. Some times it report 600 watts (real power) and another about 100 watts. It's controlling a heater (a resistor) so there must be something that is not working fine.

jotakar, 2016-05-19:
Bad meter power: its an issue about installation of the device, If live and neutral wires are interchanged the power readings are erroneus. It read 80 without any device pluged in Q1. So this issue is fixed.

closed 2016-05-19 by jotakar