OpenZWave/open-zwave #887: Devices Unknown

echo7alpha, 2016-05-17:
Hello, 

Recently I was able to get help in Domoticz forum to get my devices recognized. They were all U.S. versions: Aeotec Siren Gen5, Fibaro Motion Sensor  V3.2, Zipato RFID. Would it be possible to add these three lines into the file so it won't affect the next update?

https://www.domoticz.com/forum/viewtopic.php?f=6&t=11965

nechry, 2016-05-17:
we normally merge jeedom devices config. I added there definitions:
https://github.com/jeedom/plugin-openzwave/commit/66f148ff6ac843cb57127a5cb65916c688be721e

Fishwaldo, 2016-06-01:
@nechry when is the Jeedom Merge happening?

nechry, 2016-06-01:
@sarakha63 will normally send the PRs really soon.

Fishwaldo, 2016-06-08:
@nechry can we close this?

nechry, 2016-06-08:
https://github.com/OpenZWave/open-zwave/commit/d22da1fa4437e19f4884b41d11e31f14e1a6104d

closed 2016-06-08 by nechry