OpenZWave/open-zwave #892: Qubino ZMNHBA2 Flush 2 Relays

jano42, 2016-05-24:
I've got a problem with Qubino ZMNHBA2 Flush 2 Relays.

I've cabled it with all possible options:
 * Output1 => Led1
 * Output2 => Led2
 * Input1 <= mono stable switch (push button)
 * Input2 <= bi stable switch (classic switch)

Initial state : the two leds are OFF
When I switch ON the Input2, the led2 is ON => OK
But my sowftware using openz-wave receive to notification
 * The first indicating that the Output2 has changed to True  (ok perfect)
 * A second one, providing the Output1 value to True (**falsy Output1 value** )

This only apply when I switch input2. In case of input1 switch all is ok.

**Here my log file**

09:06:06 :  : [Debug] : Input2 => True
09:06:06 :  : [Debug] : OpenZWave notification : ValueChanged
09:06:06 :  : [Debug] : ===================================
09:06:06 :  : [Debug] : OpenZWave notification [Type_ValueChanged] : instance=2
09:06:06 :  : [Debug] : OpenZWave notification [Type_ValueChanged] : node=2
09:06:06 :  : [Debug] : OpenZWave notification [Type_ValueChanged] : value=1
09:06:07 :  : [Debug] : OpenZWave notification : ValueChanged
09:06:07 :  : [Debug] : ===================================
09:06:07 :  : [Debug] : OpenZWave notification [Type_ValueChanged] : instance=1
09:06:07 :  : [Debug] : OpenZWave notification [Type_ValueChanged] : node=2
09:06:07 :  : [Debug] : OpenZWave notification [Type_ValueChanged] : value=1  **bad value**
09:06:09 :  : [Debug] : OpenZWave notification : ValueChanged

09:06:06 :  : [Debug] : Input2 => False
09:06:09 :  : [Debug] : ===================================
09:06:09 :  : [Debug] : OpenZWave notification [Type_ValueChanged] : instance=2
09:06:09 :  : [Debug] : OpenZWave notification [Type_ValueChanged] : node=2
09:06:09 :  : [Debug] : OpenZWave notification [Type_ValueChanged] : value=0
09:06:09 :  : [Debug] : OpenZWave notification : ValueChanged
09:06:09 :  : [Debug] : ===================================
09:06:09 :  : [Debug] : OpenZWave notification [Type_ValueChanged] : instance=1
09:06:09 :  : [Debug] : OpenZWave notification [Type_ValueChanged] : node=2
09:06:09 :  : [Debug] : OpenZWave notification [Type_ValueChanged] : value=0

jotakar, 2016-05-24:
How many relays are detected? I think t must be 3: 1 root and two childs. Now your instance 1 ias acting as root relay.

jano42, 2016-05-24:
I have only two relays (id 55 and 56 in the following image)

[image]

Here my zwave configuration file

``````xml
<?xml version="1.0" encoding="utf-8" ?>
<Driver xmlns="http://code.google.com/p/open-zwave/" version="3" home_id="0xf9b47db1" node_id="1" api_capabilities="0" controller_capabilities="40" poll_interval="500" poll_interval_between="false">
	<Node id="1" name="" location="" basic="2" generic="2" specific="1" type="Static PC Controller" listening="true" frequentListening="false" beaming="true" routing="false" max_baud_rate="40000" version="4" query_stage="Complete">
		<Manufacturer id="115" name="Z-Wave.Me">
			<Product type="400" id="1" name="ZME_UZB1 USB Stick" />
		</Manufacturer>
		<CommandClasses>
			<CommandClass id="32" name="COMMAND_CLASS_BASIC" version="1" after_mark="true">
				<Instance index="1" />
				<Value type="byte" genre="basic" instance="1" index="0" label="Basic" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0" />
			</CommandClass>
		</CommandClasses>
	</Node>
	<Node id="2" name="" location="" basic="4" generic="16" specific="1" type="Routing Slave" listening="true" frequentListening="false" beaming="true" routing="true" max_baud_rate="40000" version="4" query_stage="Complete">
		<Manufacturer id="159" name="Qubino">
			<Product type="2" id="1" name="ZMNHBA2 Flush 2 Relays" />
		</Manufacturer>
		<CommandClasses>
			<CommandClass id="32" name="COMMAND_CLASS_BASIC" version="1">
				<Instance index="1" endpoint="1" />
				<Instance index="2" endpoint="2" />
			</CommandClass>
			<CommandClass id="37" name="COMMAND_CLASS_SWITCH_BINARY" version="1" request_flags="5" innif="true">
				<Instance index="1" endpoint="1" />
				<Instance index="2" endpoint="2" />
				<Value type="bool" genre="user" instance="1" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
				<Value type="bool" genre="user" instance="2" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
			</CommandClass>
			<CommandClass id="39" name="COMMAND_CLASS_SWITCH_ALL" version="1" request_flags="5" innif="true">
				<Instance index="1" />
				<Value type="list" genre="system" instance="1" index="0" label="Switch All" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="0" size="1">
					<Item label="Disabled" value="0" />
					<Item label="Off Enabled" value="1" />
					<Item label="On Enabled" value="2" />
					<Item label="On and Off Enabled" value="255" />
				</Value>
			</CommandClass>
			<CommandClass id="49" name="COMMAND_CLASS_SENSOR_MULTILEVEL" version="3" request_flags="1" innif="true">
				<Instance index="1" />
				<Value type="decimal" genre="user" instance="1" index="1" label="Temperature" units="C" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="24.9" />
			</CommandClass>
			<CommandClass id="50" name="COMMAND_CLASS_METER" version="3" request_flags="3" innif="true">
				<Instance index="1" endpoint="1" />
				<Instance index="2" endpoint="2" />
				<Value type="decimal" genre="user" instance="1" index="0" label="Energy" units="kWh" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0.0" />
				<Value type="decimal" genre="user" instance="1" index="8" label="Power" units="W" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0.0" />
				<Value type="bool" genre="user" instance="1" index="32" label="Exporting" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
				<Value type="button" genre="system" instance="1" index="33" label="Reset" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
				<Value type="decimal" genre="user" instance="2" index="0" label="Energy" units="kWh" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0.0" />
				<Value type="decimal" genre="user" instance="2" index="8" label="Power" units="W" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0.0" />
				<Value type="bool" genre="user" instance="2" index="32" label="Exporting" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
				<Value type="button" genre="system" instance="2" index="33" label="Reset" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
			</CommandClass>
			<CommandClass id="96" name="COMMAND_CLASS_MULTI_INSTANCE/CHANNEL" version="3" request_flags="1" innif="true" mapping="endpoints">
				<Instance index="1" />
			</CommandClass>
			<CommandClass id="112" name="COMMAND_CLASS_CONFIGURATION" version="1" request_flags="5" innif="true">
				<Instance index="1" />
				<Value type="list" genre="config" instance="1" index="1" label="Input 1 switch type" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="1" vindex="0" size="1">
					<Item label="Mono-stable switch type (push button)" value="0" />
					<Item label="Bi-stable switch type" value="1" />
				</Value>
[... cut ...]
				<Value type="string" genre="system" instance="1" index="0" label="Library Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="6" />
				<Value type="string" genre="system" instance="1" index="1" label="Protocol Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="3.67" />
				<Value type="string" genre="system" instance="1" index="2" label="Application Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="1.12" />
			</CommandClass>
		</CommandClasses>
	</Node>
</Driver>

``````

jano42, 2016-05-24:
Here the log of what happens (include ozw log)

``````
2016-05-24 12:41:05.428 Detail, Node002,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x02, 0x06, 0x60, 0x06, 0x02, 0x25, 0x03, 0xff, 0x4e
2016-05-24 12:41:05.433 Detail,
2016-05-24 12:41:05.454 Info, Node002, Received a MultiInstanceEncap from node 2, instance 2, for Command Class COMMAND_CLASS_SWITCH_BINARY
2016-05-24 12:41:05.461 Info, Node002, Received SwitchBinary report from node 2: level=On
2016-05-24 12:41:05.475 Detail, Node002, Refreshed Value: old value=false, new value=true, type=bool
2016-05-24 12:41:05.496 Detail, Node002, Changes to this value are not verified
2016-05-24 12:41:05.515 Detail, Node002, Notification: ValueChanged
12:41:05 :  : [Debug] : OpenZWave notification : ValueChanged
12:41:05 :  : [Debug] : ===================================
12:41:05 :  : [Debug] : OpenZWave notification [Type_ValueChanged] : instance=2
12:41:05 :  : [Debug] : OpenZWave notification [Type_ValueChanged] : node=2
12:41:05 :  : [Debug] : OpenZWave notification [Type_ValueChanged] : valeur=1
2016-05-24 12:41:05.841 Detail, Node002,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x02, 0x03, 0x25, 0x03, 0xff, 0x2a
2016-05-24 12:41:05.855 Detail,
2016-05-24 12:41:05.864 Info, Node002, Received SwitchBinary report from node 2: level=On
2016-05-24 12:41:05.886 Detail, Node002, Refreshed Value: old value=false, new value=true, type=bool
2016-05-24 12:41:05.899 Detail, Node002, Changes to this value are not verified
2016-05-24 12:41:05.912 Detail, Node002, Notification: ValueChanged
12:41:05 :  : [Debug] : OpenZWave notification : ValueChanged
12:41:05 :  : [Debug] : ===================================
12:41:05 :  : [Debug] : OpenZWave notification [Type_ValueChanged] : instance=1
12:41:05 :  : [Debug] : OpenZWave notification [Type_ValueChanged] : node=2
12:41:05 :  : [Debug] : OpenZWave notification [Type_ValueChanged] : valeur=1

``````

I d'ont know/understand what happen.

jotakar, 2016-05-24:
Have you this ine in the XML config file of your ZMHBA2? 
`<CommandClass id="96" mapping="endpoints" /> `

jano42, 2016-05-24:
Yes I've got this line.

I use the release 1.4.164; but I've tested with yesterday master branch; and I've the same behavior.

``````xml
<?xml version="1.0" encoding="utf-8"?>
<Product xmlns='http://code.google.com/p/open-zwave/'>

<!-- Qubino: ZMNHBA2 Flush 2 Relay -->

	<!-- Configuration  -->
	<CommandClass id="112">
		<Value type="list" genre="config" instance="1" index="1" label="Input 1 switch type" size="1" min="0" max="1" value="1">
			<Help/>
			<Item label="Mono-stable switch type (push button)" value="0" />
			<Item label="Bi-stable switch type" value="1" />
		  </Value>
	
		<Value type="list" genre="config" instance="1" index="2" label="Input 2 switch type" size="1" min="0" max="1" value="1">
			<Help/>
			<Item label="Mono-stable switch type (push button)" value="0" />
			<Item label="Bi-stable switch type" value="1" />
		</Value>

		<Value type="list" genre="config" instance="1" index="10" label="Activate / deactivate functions ALL ON/ALL OFF" size="1" min="0" max="255" value="255">
			<Help>Flush 2 relays module responds to commands ALL ON / ALL OFF that may be sent by the main controller or by other controller belonging to the system.</Help>
			<Item label="ALL ON active, ALL OFF active" value="255" />
			<Item label="ALL ON is not active ALL OFF is not active" value="0" />
			<Item label="ALL ON is not active ALL OFF active" value="1" />
			<Item label="ALL ON active ALL OFF is not active" value="2" />
		</Value> 
	
		<Value type="short" genre="config" instance="1" index="11" label="Automatic turning Off relay Q1 after set time" min="0" max="65535" value="0">
			<Help>0 - Auto OFF disabled. 1 - 65535 => 0,01 second - 655,35 seconds. Auto OFF enabled with define time, step is 10mseconds. Default value 0</Help>
		</Value>

		<Value type="short" genre="config" instance="1" index="12" label="Automatic turning Off relay Q2 after set time" min="0" max="65535" value="0">
			<Help>0 - Auto OFF disabled. 1 - 65535 => 0,01 second - 655,35 seconds. Auto OFF enabled with define time, step is 10 mseconds. Default value 0</Help>
		</Value>

		<Value type="list" genre="config" instance="1" index="30" label="Saving the state of the 2 relays after a power failure" size="1" min="0" max="1" value="0">
			<Help/>
	 		<Item label="Flush 2 relays module saves its state before power failure (it returns to the last position saved before a power failure)" value="0" />
			<Item label="Flush 2 relays module does not save the state after a power failure, it returns to off position." value="1" />
		</Value>
	
	 	<Value type="byte" genre="config" instance="1" index="40" label="Power reporting in Watts on power change for Q1" size="1" min="0" max="100" value="1">
			<Help>Set value from 0 - 100 (0%- 100%). 0 - Reporting Disabled. 1 - 100 = 1% - 100% Reporting enabled. Power report is send (push) only when actual power in Watts in real time change for more than set percentage comparing to previous actual power in Watts, step is 1%. Default value 1</Help>
		</Value>

		<Value type="byte" genre="config" instance="1" index="41" label="Power reporting in Watts on power change for Q2" size="1" min="0" max="100" value="1">
		<Help>Set value from 0 - 100 (0%- 100%). 0 - Reporting Disabled. 1 - 100 = 1% - 100% Reporting enabled. Power report is send (push) only when actual power in Watts in real time change for more than set percentage comparing to previous actual power in Watts, step is 1%. Default value 1</Help>
		</Value>

		<Value type="short" genre="config" instance="1" index="42" label="Power reporting in Watts by time interval for Q1" min="0" max="65535" value="300">
			<Help>Set value means time interval (0 1 65535) in seconds, when power report is send. 0 - Reporting Disabled. 1 - 65535 = 1 second - 65535 seconds. Reporting enabled. Power report is send with time interval set by entered value. Default value 300 (power report in Watts is send each 300s)</Help>
		</Value>

		<Value type="short" genre="config" instance="1" index="43" label="Power reporting in Watts by time interval for Q2" min="0" max="65535" value="300">
			<Help> Set value means time interval (0 1 65535) in seconds, when power report is send. 0 - Reporting Disabled. 1 - 65535 = 1 second - 65535 seconds. Reporting enabled. Power report is send with time interval set by entered value. Default value 300 (power report in Watts is send each 300s)</Help>
		</Value>

	</CommandClass>

	<!-- Association Groups -->
	<CommandClass id="133">
		<Associations num_groups="3">
			<Group index="1" max_associations="5" label="Output Q1 state" />
			<Group index="2" max_associations="5" label="Output Q2 state" auto="true" />
			<Group index="3" max_associations="1" label="Default reporting" auto="true"/>
		</Associations>
	</CommandClass>

	<!-- Remove COMMAND_CLASS_BASIC -->
	<CommandClass id="32" action="remove" />
	
	<!-- Map endpoints to instances -->
	<CommandClass id="96" mapping="endpoints" />

</Product>
``````

jotakar, 2016-05-24:
I'm using 1.4-244-ge9e3fce-dirty. My Qubino ZMNHBD2 (versión ZWave plus of ZMNHBA2) had this last line with mapping, after deleting it the controller recognized three relays and all works fine.
Try to delete this line (that with mapping="endpoints") in your ZWMNHBA2.xml and let ZWave organize the mapping.
After deleting this line you must modify your zwave config file (zwcfg_0x[homeid].XML): stop controller, delete the whole node 2 (the node about this module) and restart controller. The program will créate zwave config agin with the new config. Or, if this is too complicated, you can exclude/include the module after modify config file ZWMNHBA2.xml

Egregius, 2016-05-28:
With "<CommandClass id="96" mapping="endpoints" />" in the XML you have 2 switches, without it 3. 
With the endpoints line it works fine from within Domoticz but not when I1 or I2 are switched. 
Without the endpoints line there's a 'general' switch that is active if either Q1 or Q2 is activated. 
The status isn't updated automatically in domoticz, but I have a php script that triggers the refresh function of OZWCP. After that refresh all statusses are correct.

jotakar, 2016-05-29:
Try association to node 1.1, for me it Works and I don't need the script. With this setting device send reports with info about the source
Attached is the config of my Qubino ZMNHBDx of my net. Now I get status of switches and power without script. Power is reported in real time.
[configqubinozmnhbdx.xml.txt](https://github.com/OpenZWave/open-zwave/files/288279/configqubinozmnhbdx.xml.txt)

Fishwaldo, 2016-06-01:
@jotakar : Have you been able to test out the new config option for Force us to associate with instances? (so you don't have to specify 1.1 as a node)

If not, can you try:
1) Please add <CommandClass id="142" ForceInstances="true" /> to a config file affected by this and let me know how it goes. 
2) Remove this device from your network (unpair it) and best, factory reset
3) re-include the device to your network. It should not automatically add a 1.1 node to the automatic association groups. 

If this works, we can add to all Qubino devices operating like this, and make it work transparent to users.

jotakar, 2016-06-02:
@Fishwaldo, can I add the new option  
`<CommandClass id="142" ForceInstances="true" /> `
without updating OZwave and recompile my Domoticz? or I have to recompile all my platform?
My OpenZwave is Version: 1.4-244-ge9e3fce-dirty
I have to include a new Qubino Flush 2 Relay to my net and this is a good chance to test the new option.
Thanks.

Fishwaldo, 2016-06-08:
@jotakar you need to get the latest version and add that entry to the config file to test.

jotakar, 2016-06-16:
@Fishwaldo 
Hi, I included a ZMNHBD2 (detected as ZMNHBDx) using the option in XML file
` <CommandClass id="142" ForceInstances="true" /> `
And deleting
`<CommandClass id="96" mapping="endpoints" />`

And when I look my Domoticz controller nodes I saw that the Group 1 is linked to a.1.1 node (controller)
Then I linked **Groups 2 - 7 to node 1**, and the soft set **every node linked to 1.1**.  So now **every group of ZMNHBDx (qubino 2 Relays, s4) is connected to node 1.1**
Manual switching is detected by controller and meters of consume and energy are OK.
I hoe this info be useful for improve Open-Zwave.

Bye

nechry, 2016-06-17:
Hi, I try same thing with a S2 version, but the group1 never take 1.1 associations. so we still need the refresh policy to get direct input feedback

Fishwaldo, 2016-06-22:
@jotakar : thanks for testing.
@nechry : Is the S2 version a different Product Type/ID?

nechry, 2016-06-22:
@Fishwaldo no unfortunately. I suppose Application Version is different. I also try others models like the ZMNHADx (single relay, with 2 binary inputs) also now wait to set 1.1 for any association group.

jotakar, 2016-07-06:
Then this option is not to be included in config for ZMNHBDx?

nechry, 2016-07-06:
No they break old devices in S2, until we can apply a config file for a specified version I think is not possible.

Fishwaldo, 2016-07-21:
Depends upon #644

jotakar, 2016-07-21:
HI
And i've read that Qubino have just launched a new version of firmware for this device, this is the S5. Too many versions in a little period of time. And this is no bad if we could update the devices as ZWave Plus says, but it's not posible.

nechry, 2016-07-22:
As I ear the version S5 is a buggy version. We need to polling the switches to get state. 
Like the S4 the ForceInstances to true is also required

jotakar, 2016-07-22:
Yes, you confirm what I read in other forum. 
Qubino works with a quality hardware, but soft.... 
With this history in its software they should give users the tools for updating software, don't you think so?. 
We must ask for it to Qubino.