OpenZWave/open-zwave #896: Missed WakeUpCmd_NoMoreInformation

bkggkt3d, 2016-05-28:
After a WakeUp and handling all queries, the WakeUpCmd_NoMoreInformation do not send the device to sleep as soon as possible. This leads to long wake-up times (usually until the next wake-up interval triggers) and therefore high battery consumption.

Steps to reproduce:
1. Include an EUROtronic Stella or Comet (I don't know whether other devices also show this problem)
2. Setup WakeUp Interval = 240s
3. Setup Polling of Temperature = 120s (to be sure having at least one poll on each WakeUp)

Erroneous behavior:
When looking to the WakeUp, ValueChanged and Sleeping Notifications you see the device being awake much longer as needed, sometimes even for the full span of 240s until the next WakeUp comes.

Expected behavior:
Sleeping Notification shall come as fast as possible behind the last Temperature ValueChanged Notifications to being awake for only 1s 

Reason:
The WakeUpCmd_NoMoreInformation is not send to the device after the message queue is finished

Workaround:
In command_classes/WakeUp.cpp on line 458, comment out the line "sendToSleep = false;"
I don't understand the sense of this line, but in the above scenario it suppresses sending the WakeUpCmd_NoMoreInformation message and therefore prevents sending the device to sleep.

Open questions:
Is the problem specific for Stella & Comet, or also reproducible on other battery powered devices with WakeUp?
Why does it looks like there are there still outstanding queries preventing to send to sleep, even if all are handled?

Fishwaldo, 2016-06-01:
Something is wrong there. The reason it doesn't send NoMoreInformation to the device is its querystage is not complete yet. 

Can you provide a full OZW_Log.txt please?

Fishwaldo, 2016-06-22:
Ping. Closing if no response.

bkggkt3d, 2016-06-22:
No, please don't close. I need to reconfigure my network to be able to
create the log... takes time, sorry!

openzwave-dev-watcher, 2016-07-02:
> On Wed, Jun 22, 2016 at 6:53 AM, Justin Hammond <notifi...@github.com
> <javascript:>>

Finally the logs to make any analyses are here.
I've captured 2 logfiles in the attachment:

   - NOK.log used the original OpenZWave, including the problem
   at 2016-07-02 17:26:13.091 it is not going to sleep, the Info is missed
   at 2016-07-02 17:27:15.519 (which is the next wakeup cycle) it is going 
   to sleep
   - OK.log used the patched WakeUp.cpp file, where I simply commented out 
   the line 458: "sendToSleep = false;"
   at 2016-07-02 17:45:35.293 it is going to sleep

Each logfile contains 3 groups:

   1. requesting some values (by pressing the Refresh button in ozwcp)
   2. the problematic wakeup cycle, showing that it goes not to sleep
   3. the next wakeup cycle, which is always fine (configured here to come 
   very fast within 1 minute, normally it comes 10 minutes or 1h later, so the 
   device stays awake for such long time)

post: key=fun data=racp size=4
post: key=node data=3 size=1
2016-07-02 17:44:57.370 Detail, Node003, AdvanceQueries queryPending=0 queryRetries=0 queryStage=Dynamic live=1
2016-07-02 17:44:57.370 Detail, Node003, QueryStage_Dynamic
2016-07-02 17:44:57.378 Detail,
2016-07-02 17:44:57.379 Detail, Node003, Queuing (WakeUp) SwitchMultilevelCmd_Get (Node=3): 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x26, 0x02, 0x25, 0x1b, 0xfe
2016-07-02 17:44:57.379 Detail,
2016-07-02 17:44:57.380 Detail, Node003, Queuing (WakeUp) SensorMultilevelCmd_Get (Node=3): 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x31, 0x04, 0x25, 0x1c, 0xe8
2016-07-02 17:44:57.391 Detail,
2016-07-02 17:44:57.392 Detail, Node003, Queuing (WakeUp) ThermostatModeCmd_Get (Node=3): 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x40, 0x02, 0x25, 0x1d, 0x9e
2016-07-02 17:44:57.393 Detail,
2016-07-02 17:44:57.393 Detail, Node003, Queuing (WakeUp) BatteryCmd_Get (Node=3): 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x80, 0x02, 0x25, 0x1e, 0x5d
2016-07-02 17:44:57.394 Info,
2016-07-02 17:44:57.395 Detail, Node003, Queuing (WakeUp) Query Stage Complete (Dynamic)

2016-07-02 17:45:34.933 Detail, Node003,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x03, 0x02, 0x84, 0x07, 0x71
2016-07-02 17:45:34.934 Detail,
2016-07-02 17:45:34.935 Info, Node003, Received Wakeup Notification from node 3
2016-07-02 17:45:34.936 Info, Node003,   Node 3 has been marked as awake
2016-07-02 17:45:34.936 Detail, Node003, Queuing (WakeUp) SwitchMultilevelCmd_Get (Node=3): 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x26, 0x02, 0x25, 0x1b, 0xfe
2016-07-02 17:45:34.938 Detail, Node003, Queuing (WakeUp) SensorMultilevelCmd_Get (Node=3): 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x31, 0x04, 0x25, 0x1c, 0xe8
2016-07-02 17:45:34.939 Detail, Node003, Queuing (WakeUp) ThermostatModeCmd_Get (Node=3): 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x40, 0x02, 0x25, 0x1d, 0x9e
2016-07-02 17:45:34.939 Detail, Node003, Queuing (WakeUp) BatteryCmd_Get (Node=3): 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x80, 0x02, 0x25, 0x1e, 0x5d
2016-07-02 17:45:34.940 Detail, Node003, Queuing (Query) Query Stage Complete (Dynamic)
2016-07-02 17:45:34.943 Detail, Node003, Queuing (WakeUp) WakeUpCmd_NoMoreInformation (Node=3): 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x84, 0x08, 0x25, 0x1f, 0x52
2016-07-02 17:45:34.945 Detail, Node003, Notification: Notification - Node Awake
2016-07-02 17:45:34.945 Info, Notification: Notification home d47dfc81 node 3 Awake
2016-07-02 17:45:34.947 Detail,
2016-07-02 17:45:34.949 Info, Node003, Sending (WakeUp) message (Callback ID=0x1b, Expected Reply=0x04) - SwitchMultilevelCmd_Get (Node=3): 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x26, 0x02, 0x25, 0x1b, 0xfe
node = 3 id = 6 value = 25.0 precision = 1
node = 3 id = 8 value = 22.0 precision = 1
node = 3 id = 9 value =  precision = 0
node = 3 id = 10 value = 18.0 precision = 1
2016-07-02 17:45:34.965 Detail, Node003,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2016-07-02 17:45:34.966 Detail, Node003,   ZW_SEND_DATA delivered to Z-Wave stack
2016-07-02 17:45:34.983 Detail, Node003,   Received: 0x01, 0x07, 0x00, 0x13, 0x1b, 0x00, 0x00, 0x02, 0xf2
2016-07-02 17:45:34.984 Detail, Node003,   ZW_SEND_DATA Request with callback ID 0x1b received (expected 0x1b)
2016-07-02 17:45:34.985 Info, Node003, Request RTT 34 Average Request RTT 33
2016-07-02 17:45:34.985 Detail,   Expected callbackId was received
2016-07-02 17:45:35.003 Detail, Node003,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x03, 0x03, 0x26, 0x03, 0x00, 0xd7
2016-07-02 17:45:35.005 Detail,
2016-07-02 17:45:35.005 Info, Node003, Response RTT 56 Average Response RTT 51
2016-07-02 17:45:35.006 Info, Node003, Received SwitchMultiLevel report: level=0
2016-07-02 17:45:35.006 Detail, Node003, Refreshed Value: old value=0, new value=0, type=byte
2016-07-02 17:45:35.007 Detail, Node003, Changes to this value are not verified
2016-07-02 17:45:35.008 Detail, Node003,   Expected reply and command class was received
2016-07-02 17:45:35.008 Detail, Node003,   Message transaction complete
2016-07-02 17:45:35.009 Detail,
2016-07-02 17:45:35.009 Detail, Node003, Removing current message
2016-07-02 17:45:35.011 Detail, Node003, Notification: ValueChanged
2016-07-02 17:45:35.013 Info, Notification: Value Changed Home 0xd47dfc81 Node 3 Genre user Class SWITCH MULTILEVEL Instance 1 Index 0 Type byte
[... 261 lines cut ...]
node = 3 id = 10 value = 18.0 precision = 1

2016-07-02 17:27:15.473 Detail, Node003,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x03, 0x02, 0x84, 0x07, 0x71
2016-07-02 17:27:15.475 Detail,
2016-07-02 17:27:15.475 Info, Node003, Received Wakeup Notification from node 3
[... cut ...]
2016-07-02 17:27:15.517 Detail,   Expected reply was received
2016-07-02 17:27:15.517 Detail,   Message transaction complete
2016-07-02 17:27:15.518 Detail,
2016-07-02 17:27:15.518 Detail, Node003, Removing current message
2016-07-02 17:27:15.519 Detail, Node003, Notification: Notification - Node Asleep
2016-07-02 17:27:15.520 Info, Notification: Notification home d47dfc81 node 3 Sleep
node = 3 id = 6 value = 25.0 precision = 1
node = 3 id = 8 value = 22.0 precision = 1
node = 3 id = 9 value =  precision = 0
node = 3 id = 10 value = 18.0 precision = 1

Fishwaldo, 2016-07-07:
ok. this is a duplicate of #441

Fishwaldo, 2016-07-07:
Closing. Please track in the other issue.

closed 2016-07-07 by Fishwaldo