OpenZWave/open-zwave #902: compilation error - no such file or directory

eztarget, 2016-06-04:
sorry, linux idiot here just following some instructions to install open zwave
when i run this, i get the failure below.  help....

git clone https://github.com/OpenZWave/open-zwave.git
ln -s open-zwave open-zwave-read-only
cd open-zwave
**make**
cd ..

Building tinyxmlparser.o
Building hid.o
/home/sjs/open-zwave/cpp/hidapi/linux/hid.c:46:21: fatal error: libudev.h: No such file or directory
compilation terminated.
/home/sjs/open-zwave/cpp/build/support.mk:141: recipe for target '/home/sjs/open-zwave/.lib/hid.o' failed
make[1]: *** [/home/sjs/open-zwave/.lib/hid.o] Error 1
make[1]: Leaving directory '/home/sjs/open-zwave/cpp/build'
Makefile:20: recipe for target 'all' failed
make: *** [all] Error 2

rbaindourov, 2016-06-06:
Hi,

I'm testing z-stick and wall plug for a major hardware manufacturer, working for the IoT department.

I'm having the same problem using Raspberry PI 3:

pi@raspberrypi:~ $ cd open-zwave/
pi@raspberrypi:~/open-zwave $ make
CPPFLAGS= make -C /home/pi/open-zwave/cpp/build/ - 
make[1]: Entering directory '/home/pi/open-zwave/cpp/build'
Building OpenZWave Version 1.4-260-ge12ba7f-dirty
Building hid.o
/home/pi/open-zwave/cpp/hidapi/linux/hid.c:46:21: fatal error: libudev.h: No such file or directory
 #include <libudev.h>
                     ^
compilation terminated.
/home/pi/open-zwave/cpp/build/support.mk:141: recipe for target '/home/pi/open-zwave/.lib/hid.o' failed
make[1]: *** [/home/pi/open-zwave/.lib/hid.o] Error 1
make[1]: Leaving directory '/home/pi/open-zwave/cpp/build'
Makefile:20: recipe for target 'all' failed
make: *** [all] Error 2
pi@raspberrypi:~/open-zwave $ 
 

Can you help please? Thank You!

ualex73, 2016-06-06:
Following should fix your issue:

sudo apt-get install libudev-dev

rbaindourov, 2016-06-06:
had the same issue on ubuntu desktop seems solved by doing a 
sudo apt-get install libudev-dev

eztarget, 2016-06-06:
thanks, i went back and started over and it worked this time.  i may have missed the lib update the first time.

closed 2016-06-06 by eztarget

cerealkill, 2017-09-04:
Worked perfectly. `sudo apt-get install libudev-dev` is a good line to add to the project README.md

JoeStrout, 2017-11-02:
I just ran into the same thing.  I agree with @cerealkill, this requirement should be added to the README or INSTALL file (neither of which makes any mention of it in the latest master).