OpenZWave/open-zwave #903: Compilation without libstdc++.so.6

and7ey, 2016-06-06:
Is there any way to disable libstdc++.so.6 usage by open-zwave?
I am trying to compile open-zwave for OpenELEC 6.0 running on Raspberry Pi and libstdc++.so.6 is missing there.

gms1, 2016-06-07:
I just stumbled on this question:
OpenZwave does not contain a reference to a particular version of the libstdc++ shared library, so there must be something wrong in your build environment, but it's hard to guess without seeing the output.

'libstdc++-v3' is more or less the package name for libstdc++ provided by gcc 3.0 and higher.
'libstdc++.so.6.0.0' is the version included in gcc 3.4 and
gcc 6.1 should come with 'libstdc++.so.6.0.22'

and7ey, 2016-06-07:
I haven't faced with any issues during build process. But when I run the result on Raspberry Pi, I get the following error:

```
Pi3:~/.kodi/addons/service.multimedia.open-zwave # LD_LIBRARY_PATH="/storage/.kodi/addons/service.multimedia.open-zwave/" ./MinOZW 
./MinOZW: /usr/lib/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by ./MinOZW)
./MinOZW: /usr/lib/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by /storage/.kodi/addons/service.multimedia.open-zwave/libopenzwave.so.1.4)
```

gms1, 2016-06-07:
so this '/usr/lib/libstdc++.so.6' version is too old. 
The missing symbol version 'CXXABI_1.3.9' requires at least libstdc++.so.6.0.21 (provided by gcc 5.1 and higher) but has been found during the link-step

Fishwaldo, 2016-06-07:
openzwave is a C++ application. You need the standard c++ library! (its essential)

And whoever is complaining about what version of libstdc++ library its using doesn't know what they are talking about. We use the library version that is shipped with the compiler. we don't need to specify the version numbers as it should be setup automatically by the distribution (libstdc++.so should point to the latest version on your system).

If you are getting missing symbol errors etc, I'm assuming your cross compiling, and in that case, your cross compilers is using different library versions from what is installed on the target system. You need to fix that, which is a distro/cross compiler issue, not a OZW issue.

closed 2016-06-07 by Fishwaldo