OpenZWave/open-zwave #912: RESTful API over OpenZ-Wave

rbaindourov, 2016-06-12:
Hi Guys,

I had posted a question about modifiying the open-zwave-control-panel into an API wrapper over the driver codebase.

https://github.com/OpenZWave/open-zwave-control-panel

I started doing the reversing, putting my instrumentation into the C++ and front end.
There is a lot of code; making this task time consuming.
The cp.js file is pretty big, and I'm on a super tight deadline, so I just wanted to ask this question one more time.

Is there an existing way to probe network nodes via RESTful API?
Just any help or documentation would be appreciated.
Like what endpoints / commands the webserver exposes that I can lean on to use the existing code as much as possible and leverage into an API.

Thanks!

Love your work.

rbaindourov, 2016-06-15:
Hi,

I have successfully added a poll.json end point to the web server handler:
https://github.com/rbaindourov/open-zwave-control-panel

Fishwaldo, 2016-06-22:
Hi,
This is a issue tracker and not a support forum. 

Please use our Forums for Questions regarding OpenZWave: 
https://groups.google.com/forum/#!forum/openzwave

Thanks

closed 2016-06-22 by Fishwaldo

rbaindourov, 2016-06-22:
Isn't my post here cross posted there?
https://groups.google.com/forum/#!topic/openzwave/gsUhWqHUOZo

I'm looking for a little more helpful advice here, 
if you want me to use a system I am less familiar with, fine, I will do so, but I have posted questions there as well.

So far my first interaction with anyone on your team has been this.

Thanks for your hard work.

rbaindourov, 2016-06-22:
Here you go, 

https://groups.google.com/forum/#!topic/openzwave/W9HGRazzru0

Thanks Again.

Fishwaldo, 2016-06-23:
Yes - But this is a issue tracker not a "discussion forum" 

I use this to track actual bugs or enhancements etc that need to be made to code.

rbaindourov, 2016-06-23:
Well I worked in a pretty good JSON API end point.
And that was the enhancement I was contacting you about.

So I am not sure why you think this is an empty discussion, since the title of the issue is a feature request.
This is a serious project. We are going to deploy this tech in a government building in Australia to monitor energy usage with the smart plugs, this week.
http://imgur.com/rBkk1tQ

And I really could have used some help like a week ago, in prepping an API to read updates from.

Anyway, I was able to learn your C++ code on my own and delivered the software! :)
https://github.com/rbaindourov/open-zwave-control-panel 

It's interesting how we have to turn polling on with a call to `pollpost.html`.

The new `/poll.json` endpoint that I created is kind of what I needed. I also need to be able to pass in a filter ID so I don't get every node. But I didn't have time so I just did the filtering on the consumer's end.

Thanks for getting back to me.
I really do admire your hard work for reverse engineering radio communications.

By The Way: There is a node-openzwave library, and once its updated, that would make writing your next generation control panel much easier. C++ is really unforgiving compared to Node.

Thanks again!

openzwave-dev-watcher, 2017-07-31:
Hi rbaindourov,

I Tried to compile your code and run on beaglebone black using ./ozwcp -d 
-p . After executing this number i get an error saying cannot open serial 
port and later one more error is failed to open serial port.

2017-07-31 13:32:57.610 Always, OpenZwave Version 1.4.2692 Starting Up
2017-07-31 13:32:57.619 Info, Setting Up Provided Network Key for Secure 
Communications
2017-07-31 13:32:57.622 Warning, Failed - Network Key Not Set
2017-07-31 13:32:57.625 Info, mgr, Added driver for controller ��CF
N�F
M�FD��V�}Dv���
��=
2017-07-31 13:32:57.630 Info, Opening controller ��CF
N�F
M�FD��V�}Dv���
��=
2017-07-31 13:32:57.634 Info, Trying to open serial port ��CF
N�F
M�FD��V�}Dv���
��= (attempt 1)
2017-07-31 13:32:57.638 Error, ERROR: Cannot open serial port ��CF
N�F
M�FD��V�}Dv���
��=. Error code 2
2017-07-31 13:32:57.640 Error, ERROR: Failed to open serial port ��CF
N�F
M�FD��V�}Dv���
��=
2017-07-31 13:32:57.642 Warning, WARNING: Failed to init the controller 
(attempt 0)
2017-07-31 13:33:02.644 Info, Opening controller ��CF
N�F
M�FD��V�}Dv���
��=
2017-07-31 13:33:03.275 Info, Trying to open serial port ��CF
N�F
M�FD��V�}Dv���
��= (attempt 1)
2017-07-31 13:33:03.276 Error, ERROR: Cannot open serial port ��CF
N�F
M�FD��V�}Dv���
��=. Error code 2
2017-07-31 13:33:03.276 Error, ERROR: Failed to open serial port ��CF
N�F
M�FD��V�}Dv���
��=
2017-07-31 13:33:03.276 Warning, WARNING: Failed to init the controller 
(attempt 1)
2017-07-31 13:33:08.277 Info, Opening controller ��CF
N�F
M�FD��V�}Dv���
��=
2017-07-31 13:33:08.445 Info, Trying to open serial port ��CF
N�F
M�FD��V�}Dv���
��= (attempt 1)
2017-07-31 13:33:08.446 Error, ERROR: Cannot open serial port ��CF
N�F
M�FD��V�}Dv���
��=. Error code 2
2017-07-31 13:33:08.446 Error, ERROR: Failed to open serial port ��CF
N�F
M�FD��V�}Dv���
��=
2017-07-31 13:33:08.446 Warning, WARNING: Failed to init the controller 
(attempt 2)
2017-07-31 13:33:13.447 Info, Opening controller ��CF
N�F
M�FD��V�}Dv���
��=
2017-07-31 13:33:13.448 Info, Trying to open serial port ��CF
N�F
M�FD��V�}Dv���
��= (attempt 1)
2017-07-31 13:33:13.448 Error, ERROR: Cannot open serial port ��CF
N�F
M�FD��V�}Dv���
��=. Error code 2
2017-07-31 13:33:13.451 Error, ERROR: Failed to open serial port ��CF
N�F
M�FD��V�}Dv���
��=
2017-07-31 13:33:13.469 Warning, WARNING: Failed to init the controller 
(attempt 3)
2017-07-31 13:33:18.472 Info, Opening controller ��CF
N�F
M�FD��V�}Dv���
��=
2017-07-31 13:33:18.477 Info, Trying to open serial port ��CF
N�F
M�FD��V�}Dv���
��= (attempt 1)
2017-07-31 13:33:18.483 Error, ERROR: Cannot open serial port ��CF
N�F
M�FD��V�}Dv���
��=. Error code 2
2017-07-31 13:33:18.485 Error, ERROR: Failed to open serial port ��CF
N�F
M�F~D��V�}Dv���
��=
2017-07-31 13:33:18.487 Warning, WARNING: Failed to init the controller 
(attempt 4)
2017-07-31 13:33:18.489 Detail, contrlr, Notification: DriverFailed
2017-07-31 13:33:18.491 Info, Notification: type DriverFailed:19 home 
00000000 node 255 genre 0 class 0 instance 0 index 0 type 0 m_byte 0
PLease help me with this.