OpenZWave/open-zwave #919: MinOZW crash

fschwebel, 2016-06-29:
I am trying to debug an issue with domoticz that seems to crash with a SIGSEGV linked with OpenZWave. I am following the standard process to use MinOZW to generate logs, but the binary crashes with:
```
[...]
2016-06-29 20:27:36.962 Detail, Node003, Removing current message
*** Error in `.lib/MinOZW': free(): invalid next size (fast): 0x00007ff58000ade0 ***
```
I guess that comes from the `delete nodeInfo;` on line 179 in `cpp/examples/MinOZW/Main.cpp`, but no luck so far trying to solve the issue by recompiling MinOZW.
Version: head of the main branch (tested with a379e3a)

Fishwaldo, 2016-07-07:
I can't reproduce this, so you will have to fire up debuggers etc to try to figure it out.

Fishwaldo, 2016-08-11:
Closing as no response

closed 2016-08-11 by Fishwaldo