OpenZWave/open-zwave #931: SIGSEGV in OpenZWave::Msg::setNonce

kravietz, 2016-07-09:
I'm using open-zwave through Python home-assistant and after a few hours of running it's crashing consistently with SIGSEGV and dumping the core. The stack trace suggests that at some point the setNonce method is being called with null this:

```
#0  __memcpy_sse2_unaligned () at ../sysdeps/x86_64/multiarch/memcpy-sse2-unaligned.S:158
#1  0x00007f8ff7551e49 in OpenZWave::Msg::setNonce (this=0x0, nonce=0x7f8fef7fd999 "\344\210\066$\302\262")
    at /home/kravietz/Z-Wave/python-openzwave/openzwave/cpp/src/Msg.h:158
#2  0x00007f8ff75428b0 in OpenZWave::Driver::ProcessMsg (this=0x7f9004001700, _data=0x7f8fef7fd992 "")
    at /home/kravietz/Z-Wave/python-openzwave/openzwave/cpp/src/Driver.cpp:1816
#3  0x00007f8ff75422f8 in OpenZWave::Driver::ReadMsg (this=0x7f9004001700)
    at /home/kravietz/Z-Wave/python-openzwave/openzwave/cpp/src/Driver.cpp:1719
#4  0x00007f8ff753e32e in OpenZWave::Driver::DriverThreadProc (this=0x7f9004001700, _exitEvent=0x7f9004001620)
    at /home/kravietz/Z-Wave/python-openzwave/openzwave/cpp/src/Driver.cpp:480
#5  0x00007f8ff753df6c in OpenZWave::Driver::DriverThreadEntryPoint (_exitEvent=0x7f9004001620, _context=0x7f9004001700)
    at /home/kravietz/Z-Wave/python-openzwave/openzwave/cpp/src/Driver.cpp:383
#6  0x00007f8ff7570e24 in OpenZWave::ThreadImpl::Run (this=0x7f90040021e0)
    at /home/kravietz/Z-Wave/python-openzwave/openzwave/cpp/src/platform/unix/ThreadImpl.cpp:176
#7  0x00007f8ff7570de8 in OpenZWave::ThreadImpl::ThreadProc (_pArg=0x7f90040021e0)
    at /home/kravietz/Z-Wave/python-openzwave/openzwave/cpp/src/platform/unix/ThreadImpl.cpp:161
```

Fishwaldo, 2016-07-11:
hrm... Could you also post a OZW Log File from this? From the backtrace, we are getting a NONCE from the device (so we can encrypt a message to it) which would have been in response to a NONCE Get message, but we don't have a actual message in the queue to encrypt.

it would be really helpfully to see a log, to see exactly why we are getting this.

kravietz, 2016-07-11:
I'm afraid the original OZW log was overwritten since but I'll wait for it to crash again. Also I have the core file if that would help.

Fishwaldo, 2016-07-11:
The LogFile would be really beneficial to understand why we got that Nonce Message (is it a OZW bug, is it a device bug)

I know how to fix the actual crash, but that might not be the actual problem (a symptom, rather than the root cause)

No need for the core file, as your backtrace gave me enough hits to pinpoint whats going on..

kravietz, 2016-07-12:
Ok, now waiting for the crash to happen again.

kravietz, 2016-07-12:
And it did... here's the log file 
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/360018/OZW_Log.txt)

Fishwaldo, 2016-07-21:
ok. We are getting a duplicate NONCE Report message for this device. If its happening all the time, must be some buggy firmware on the device. 

Which, brings me to the next question - What device is this? 

Its "encrypting" the WakeUP CC, which I've never seen any other device do before, and honestly sounds like it could lead to further problems. (sleeping devices, Wakeup Messages, and the extra messages required to send encrypted messages.

I'll push a fix soon, but as a work around, you could re-include this device non-secure and it shouldn't have this issue.

Fishwaldo, 2016-07-25:
Fix Pushed to master branch. Thanks for the help diagnosing this.

closed 2016-07-25 by Fishwaldo