OpenZWave/open-zwave #933: Refreshing node -> ERROR: Dropping command, expected response not received

Ryszard-Trojnacki, 2016-07-15:
I have a problem with refreshing a node.

I have created my own pooling code (thread that controls all polling) which calls `Manager::Get()->RefreshValue(vid)`.
Everything works fine but after a time communication is not working and all attempts to refresh value ends with:
```
Error, Node007, ERROR: Dropping command, expected response not received after 1 attempt(s)
```
And after this nothing more is working (communcation).

I think that it is not the problem with the device (node), because restart is helping. By restart I mean: `Manager::Get()->SoftReset(homeId);` and restarting application.

I'm using almost the latest version of OpenZWave library (downloaded from github master branch at 12.07.2016, but this problem also existed in previous version - from 26.04.2016).

Polling interval is set to 4 seconds for two values from Node 7.
The device is: Philio Technology Corporation type=0001, id=0013 (more in [node7.txt](https://github.com/OpenZWave/open-zwave/files/365354/node7.txt))

I have attached [ozw.txt](https://github.com/OpenZWave/open-zwave/files/365358/ozw.txt). It is a little big, but it is from application start till the problem that is 2016-07-14 10:08:29.358.

Fishwaldo, 2016-07-21:
is this on a rPI? There seems to be issues with the serial driver on that platform. 

But looking at the message, the stick stops responding to any commands. Can you try a different stick?

Ryszard-Trojnacki, 2016-07-26:
It is something else than rPI (I don't know what I will get more informations about that).
I use ZMERAZ Z-Wave controller connected with UART TTL.

I will try diffrent stick, conntected with USB, but it will take some time.

Is there something I can do to check if it is serial driver problem? (dmesg or something else)

iamkrillin, 2016-08-04:
I've seen this error on both a raspberry pi and on a full pc running ubuntu.  Seems to happen anytime you send commands to a node "fast".  I am using domoticz and I could cause these errors by simply clicking the on/off button rapidly.  When the errors occur all zwave comm goes down, it does seem to recover after "a while".  I tested my zwave usb stick (aeon labs) with some other software http://www.indigodomo.com/ and this does not happen

iamkrillin, 2016-08-04:
Update, I updated the device database files that I have in domoticz to the files in the dev branch, the problem went away

Fishwaldo, 2016-08-11:
@Ryszard-Trojnacki : I've never been able to reproduce the issue on rPi, but its been reported often enough. No obvious errors are emitted, so I imagine it would take some "hardcore" debugging to figure out whats going on.....

Please post to our mailing list if you discover something.... 

@iamkrillin Aeotec had issues with older versions of their Z-Stick in that it would completely lock up after a while as well. Newer Z-Sticks don't seem to suffer from this problem.

closed 2016-08-11 by Fishwaldo

roncapat, 2017-04-28:
Same problem, with Aeotec ZW090 Z-Stick Gen5 that can't communicate with A Smart switch 6. Same error. Anyone got a fix?

rwa, 2017-04-30:
I also have this problem very frequently.  Aeotec Z-stick Gen5.  I've ordered the Homeseer z-wave stick to see if this is a hardware problem, because I've just about given up trying to find the cause of this.

nitrag, 2017-06-12:
@rwa How does the Homeseer stick work?

carlostico, 2017-08-31:
Any news about this ?

Ryszard-Trojnacki, 2017-09-01:
It seems that it was a problem with UART TTL connection with Z-Wave controller.
After changing to USB never happen again.

baddar, 2018-01-05:
@Ryszard-Trojnacki what do you mean? I have a Aeotec Z-stick gen5 and wondering what to do with my Debian server to make this working? I keep getting ERROR: Dropping command, expected response not received after 1 attempt(s) for 2 of my devices (Fibaro Radiator thermostats both of them).

rwa, 2018-01-05:
@nitrag My overall performance did improve with the Homeseer stick.  But I've also added more z-wave devices to my network, so that may have helped network reliability as well.

Ryszard-Trojnacki, 2018-01-06:
@baddar We had a hardware problem. It was nothing to do with OpenZWave library. Therefore I can't help.

kmwoley, 2018-02-14:
FWIW, I have this error consistently in my logs. I've got a Aeotec Z-stick gen5 and the error is consistently produced when communicating with my Schlage Connect lock. There is a door sensor on the same door as the lock and I'm continually getting dropped messages from the door sensor when the door is locked/unlocked at nearly the same time as the door being opened/shut. 

I'm running HomeAssistant (0.63.1) and OpenZwave (1.4.2926). 

I'd be happy to open a new issue, produce whatever logs would be useful to track down this issue.

basschipper, 2018-02-27:
Hi @kmwoley ,

I'm experiencing the same issue running HomeAssistant 0.63.3 on a Intel NUC (Docker Environment).

I already tried unpairing all devices, resetting the Aeotec and repairing all devices. But unfortunately this did not solve the problem. Also, I bought a new Aeotec, still no luck...

I don't know for sure yet but I have the feeling that the problem is increasing when adding the Fibaro Motion Sensors (FGMS-001). Maybe there is a cause with the battery operated devices?

psgcooldog, 2018-03-09:
I'm also having this same problem.  HomeAssistant 0.64.0, OpenZwave Version 1.4.2926 , on Ubuntu Server 17.10.  I've got 70+ switches and dimmers, mostly GE/Jasco, a couple of garage door tilt sensors, and some relay switches.  A mix of Z-Wave and Z-Wave Plus.

It works for a while, stalls out with these errors, then sometimes starts working again, and then fails again.
This is driving me crazy.

basschipper, 2018-03-09:
I'm so dispirited that I have replaced all my zwave devices with zwave plus devices, but unfortunately this didn't solve my issues...

Now testing the approach below:
https://community.home-assistant.io/t/rpi-as-z-wave-zigbee-over-ip-server-for-hass/23006

psgcooldog, 2018-03-09:
I'm 100% certain that this issue is not due to bad RF network connectivity, at least in my case. I'm curious if swapping out the Aeotec for that Linear HUSBZ-1, or some other Linux and HomeAssistant compatible interface would work.  I think this is an Aeotec issue.

basschipper, 2018-03-09:
I have already replaced the Aeotec, but it has not solved it either. Maybe I should try a different brand controller stick? Strange because everyone seems to use the Aeotec?

@psgcooldog Are you also seeing these unknown devices in your logs? 

2018-01-12 18:13:57.262 Detail, Node014,   Expected reply was received
2018-01-12 18:13:57.262 Detail, Node014,   Message transaction complete

Node 14, 15, 17, 25, 27, 32, 45, 75, 172, 247 are not added by me...

http://www.openzwave.com/log-analyzer/public/3b33072e-6aa2-4ce3-a97b-1f45d8deb9e0/

iamkrillin, 2018-03-09:
I tried everything to fix his problem, my zwave network has ~40 nodes in it (so I know its not a reachability issue)

- Tried multiple platform (pc, razberry)
- Tried different interface types 
   - Had the aeontec usb, then moved to the razberry (http://z-wave.me/products/razberry/)
- Tried different controller softwares
- Tried different timeouts and retry settings

I was able to fix it eventually, but it involved moving to a different software stack all together and not using openzwave anymore.  Since then, my zwave network has worked flawlessly (seems more responsive too)

iamkrillin, 2018-03-09:
On a side note, it looks like the approach that openzwave takes is "I think this node is dead, stop sending commands to it"  This seems in direct contrast to how other commercial zwave controllers work, I had a zwave node stop responding to commands from the controller, but the controller would try to send commands everytime I asked it to, this particular node had been dead for several days.

psgcooldog, 2018-03-09:
iamkrillen - What software did you change over to?  I'm wondering if putting the Z-Wave stuff behind a Vera hub would work better.

iamkrillin, 2018-03-09:
I used vera for a while and I was pretty happy with it overall.  I moved away from it due to something they started doing with their software that I disagreed with (dont remember what it was now) and I ended up using a raspberry pi + razberry board + zway.  Been using my current setup for well over a year now with no problems at all.

basschipper, 2018-03-09:
Hi @iamkrillin,

Sounds interesting and rises some questions:
1. What I understand from reading is you either use the Z-Way server OR OpenZWave?
2. Are you using the RaZberry with Home Assistant? 
3. Is there a component to communicate with the Z-Way server?
4. My HA is running on a Intel NUC (needs the performance), is this going to work?

psgcooldog, 2018-03-09:
Bas - another alternative would be using a VeraEdge hub for the Z-Wave
interface; HA (which I am using too) can talk to it and discover devices.
That would sidestep OpenZwave too.

iamkrillin, 2018-03-09:
I'm using the zway sever on mine, and it works great on the raspberry pi. The razberry board itself hooks up to the GPIO pins on the raspberry pi, but it does have an HTTP API, so you _could_ do things on your NUC, but it might involve some coding and such on your part. Course, I guess the same might be said for using vera.