OpenZWave/open-zwave #935: Qubino zmnhvd1 / Zmnhvd* not supported.

machielw, 2016-07-17:
Can somebody help with a config-file for Qubino zmnhvd1 / Zmnhvd*. 0-10v dimmer
It is working, but not well recognised.

nechry, 2016-07-18:
Cold you provide the product id and type for ZMNHOD1 ? I will wrote the config file

cdjackson, 2016-07-18:
If you're interested, there is an online database here -:
http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/363

and an export of the OZW file here -:
http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/363?layout=openzwave

Note that it's probably never been tested, so if the format is wrong it can be modified...

machielw, 2016-07-18:
Thanks for your respons, it is the Qubino 0-10v dimmer. I use it for the
house ventilation.
Type 0001, id 0053

thanks!

012 (0x0c) Qubino 0-10 Fan Qubino Unknown: type=0001, id=0053 Qubino 0x0053
0x0001 2016-07-18 20:05:01

machielw, 2016-07-18:
I tried this allready, but I am not good enought in adjusting the script.
"out of the box"its not working
Thanks for your respons!

nechry, 2016-07-18:
I will do tomorrow

cdjackson, 2016-07-18:
I’m happy to adjust the script so that the exporter correctly exports in the OZW format if someone can tell me what the issues are. I don’t use OZW but the database is used for a couple of other zwave software (openHAB and Symcon).

nechry, 2016-07-19:
@cdjackson hi, I try your export but the generated file is not well formatted and there is some information missing. whenever is possible I use list too define a parameter but your source dose not contains all list values.

nechry, 2016-07-19:
close by https://github.com/OpenZWave/open-zwave/commit/860b2351ce9958bb5e91a40259355d3de17c1f03

closed 2016-07-19 by nechry

sumo-89, 2016-11-27:
Hi!
I have a question when I use a 0-10V input on this one. From the manual (it's not really clear):

**Endpoint 2:**
Device Class:
ZWAVEPLUS_INFO_REPORT_ROLE_TYPE_SLAVE_ALWAYS_0N
GENERIC_TYPE_SENSOR_MULTILEVEL
SPECIFIC_TYPE_ROUTING_SENSOR_MULTILEVEL
Command Classes:
COMMAND_CLASS_ZWAVEPLUS_INFO_V2
COMMAND_CLASS_VERSION_V2
COMMAND_CLASS_ASSOCIATION_V2
COMMAND_CLASS_MULTI_CHANNEL_ASSOCIATION_V3
COMMAND_CLASS_ASSOCIATION_GRP_INFO_V2
COMMAND_CLASS_SENSOR_MULTILEVEL_V7

**Endpoint 3:**
Device Class:
ZWAVEPLUS_INFO_REPORT_ROLE_TYPE_SLAVE_ALWAYS_0N
GENERIC_TYPE_SENSOR_MULTILEVEL
SPECIFIC_TYPE_ROUTING_SENSOR_MULTILEVEL
Command Classes:
COMMAND_CLASS_ZWAVEPLUS_INFO_V2
COMMAND_CLASS_VERSION_V2
COMMAND_CLASS_ASSOCIATION_V2
COMMAND_CLASS_MULTI_CHANNEL_ASSOCIATION_V3
COMMAND_CLASS_ASSOCIATION_GRP_INFO_V2
COMMAND_CLASS_SENSOR_MULTILEVEL_V7

NOTE: The above list is valid for the product with a
temperature sensor connected to TS terminal. In case the
sensor is not connected then following command class
isn’t supported:
COMMAND_CLASS_SENSOR_MULTILEVEL_V7 

I don't have a digital temp sensor on the TS terminal, only analogue PT100 -> 0-10V converter -> analogue input. I tried Domoticz now, but shouldn't I get two devices - one "Level" and one "Temp"? 
I get "Level" and that works just fine! :)