OpenZWave/open-zwave #937: implement COMMAND_CLASS_FIRMWARE_UPDATE_MD_V2 0x7a / 122

Dodobibi, 2016-07-21:
It is very usefull if openzwave implement the COMMAND_CLASS_FIRMWARE_UPDATE_MD_V2.

Fibaro begin the release of firmwares for  it's modules.

Thanks.

Fishwaldo, 2016-07-21:
I have the code, but I wont release, as there is a chance that devices can be bricked if the update proceess is interupted etc, and I've had one of the vendors that supports us request us to not release if possible, as they forsee issues around RMA with bricked devices.

Seeing as the vendor is pretty generous with us, I'm going to abide by their wishes...

closed 2016-07-21 by Fishwaldo

Dodobibi, 2016-07-21:
OK , I respect !

Too bad because I really need this feature . Some of my modules must be updated to function properly , such as "multi-instance association".

Optionally , could you provide me the implementation of this CommandClass ?

I'll understand if you refuse ...

Regards,

nechry, 2016-07-22:
@Dodobibi as I ear Fibaro don't share the devices firmware but with a HomeCenter2 you can update your devices. correct me if I'm wrong.

Dodobibi, 2016-07-22:
Yes, but i don't have a HomeCenter2. I use my own solution developped with Openzwave and Python-openzwave.

Fibaro send me directly the firmware for my module because there is an issue with my modules.

nechry, 2016-07-22:
@Dodobibi you can use homeseer with 30 days eval to update your devices.
I don't have a HomeCenter2, I'm one of the developers of Plugin Zwave for jeedom, also base on Openzwave and Python-openzwave.