OpenZWave/open-zwave #948: Qubino ZMNHDD1 Flush Dimmer: Input 2 not working

ThinkPadNL, 2016-07-29:
Hi,

I have a ZMNHDD1 Flush Dimmer, but since i have this device it is a real pain in the ass to get working.
The dimmer level and input1 works fine, but i also have a pulse switch connected to the input2 to use in rules.

However, with Domoticz it was difficult, with my Zipato Zipabox it was difficult and i am now using Home Assistant and again it isn't registering anything on the i2 input. I am beginning to get quite fed up of this device....

For Domoticz i read that the parameter 100 needs to be changed to 'Binary sensor'. I did that through OZWCP in Domoticz, but the setting doesn't seem to stick. I also added the controller to group 5 of this module (source: https://www.domoticz.com/forum/viewtopic.php?t=9598#p67094)

But still, it isn't working. Does anyone have some good tricks to get this working flawlessly?

nechry, 2016-07-29:
Check the device documentation, after set param 100 your need to exclude device without reset and include again

ThinkPadNL, 2016-07-29:
Forgot to mention that, but i already did that.

Change parameter 100 to binary sensor
Exclude device
Include again
--> Still no i2 input

I think i am going to exchange this Qubino module with a different one i have lying around.

sjoerdmulder, 2016-07-29:
@ThinkPadNL I have no idea if this works but looking at the [PDF](http://www.vesternet.com/downloads/dl/file/id/637/product/1559/z_wave_qubino_flush_dimmer_plus_manual.pdf) it seems the I2 and I3 inputs only support command classes **COMMAND_CLASS_SENSOR_BINARY** and **COMMAND_CLASS_BASIC**

The COMMAND_CLASS_BASIC is explicitly removed here https://github.com/OpenZWave/open-zwave/blob/master/config/qubino/ZMNHDDx.xml#L208. What you could try is to comment out the line too see if the inputs are added then? 

I noticed a 'faster' toggle after disabling it on I1 (issue #950) but i don't know why that class is excluded  (must have a reason?)

ThinkPadNL, 2016-07-30:
Today i was able to try a ZMNHDA2 that i borrowed from a friend, and this worked flawlessly right after including, excluding + resetting (to be sure the default parameters are loaded) and including it again.

How do i change that file on my system? On my system it seems to be at `/srv/hass/src/python-openzwave/openzwave/config/qubino/ZMNHDDx.xml`. I think it needs to be edited and then recompile OpenZWave again? I installed it by following this tutorial: https://home-assistant.io/getting-started/installation-virtualenv/ (bottom of the page). I am not very experienced with compiling etc. so please tell me how to do it :)

sjoerdmulder, 2016-07-30:
I edited the file in the Domoticz directory `/src/domoticz/Config/qubino` and then just restarted Domoticz. Should not be needed to recompile since it's a configuration file.

nechry, 2016-07-30:
Just edit the xml file save and remove your zwcfgxxx.xml file before start your network. You don't need to do compilation with xml file.

ThinkPadNL, 2016-07-30:
Ok, i don't have the module included right now, what are the steps then?

Include
Shutdown HASS so Z-Wave is also stopped
Edit `ZMNHDDx.xml` and remove line 208
Remove `zwcfgxxx.xml` from folder where it resides (`/home/hass/.homeassistant`)
Start HASS again and see if module works?

The `zwcfgxxx.xml` is then automatically built again or so?

ThinkPadNL, 2016-07-30:
Removing the [line 208](https://github.com/OpenZWave/open-zwave/blob/master/config/qubino/ZMNHDDx.xml#L208) didn't work. The input still isn't responding.

nechry, 2016-07-30:
Yes your zwcfg is re created if not present.
But if not yet include justedot the zmnhddx.xml and start your network.

ThinkPadNL, 2016-07-30:
I followed the steps above, but it didn't work. The input 2 still isn't responding (neither did input 3).

nechry, 2016-07-30:
are you sure, you exclude without reset the device after set input parameter?

ThinkPadNL, 2016-08-02:
I think so.

I have exchanged the ZMNHDD1 for a ZMNHDA2 that i got from a friend. That one worked directly after including. I am now going to sell the ZMNHDD1. Have had enough of it...

closed 2016-08-02 by ThinkPadNL

s-hal, 2016-08-22:
I too have this problem and asked Qubino support:
"ZMNHDA2 is programmed accordingly to Z-Wave 300 series chip and the protocol specifications are very different from the Z-Wave 500 series. For this reason the 300 series was reporting multichannel encapsulated messages when the module was included into the network. On the 500 series the COMMAND_CLASS_MULTI_CHANNEL_ASSOCIATION.MULTI_CHANNEL_ASSOCIATION_SET must be used to »enable« multichannel encapsulated messages."

mversteegh, 2018-01-29:
Was this issue solved? I'm facing the samething as @ThinkPadNL. I'm using Domoticz v3.8805.

ThinkPadNL, 2018-01-29:
@mversteegh Nope. Like i said in august 2016:

> I have exchanged the ZMNHDD1 for a ZMNHDA2 that i got from a friend. That one worked directly after including. I am now going to sell the ZMNHDD1. Have had enough of it...

mversteegh, 2018-01-29:
Damn, Just bought andere placed three dimmers

harigeharrie, 2018-11-21:
I've been playing with this for a few days and I think I've found a solution but it would be great to get some more confirmation.... 
It requires a few steps, and it's easiest to remove and re-add your device. 
- First make sure you have parameter 100 and 101 setup correctly (should be 9: binary sensor)
- Remove the device from the network without resetting the settings. See the manual on how to do that BUT... for me, power off, power on, wait 10s, press I1 rapidly 10 times works. Totally different from manual...
- Change the following in `qubino/ZMNHDDx.xml`
add `<CommandClass id="142" ForceInstances="true" />` just before `</Product>` at the end of the file. This forces OZW to add a target instance 1 to every association request and makes the dimmer actually report which instance it's reporting about. (thanks @s-hal for putting me on the right track here)
- Re-add the device as usual. 
- Add the controller to association group 7 and 10, to subscribe to the updates of the sensor
- Check the logs for errors, should all be good
- Trigger I2 or I3 by pressing a button or something more fancy. You should see something like `Received a MultiChannelEncap from node 15, endpoint 3 for Command Class COMMAND_CLASS_SENSOR_BINARY` in the OZW logs. if it includes the endpoint, you're good and the value gets passed through correctly. There will be Alarm reports in the logs as well, those you get for free ;)

The update comes in a bit slow still, but at least it is coming in through the right channel now. I'm pretty close to a fix for that too (and it does not require re-adding the device)...

guymcswain, 2019-03-04:
@harigeharrie , I found your instructions very helpful and I now have my Qubino Flush 1 Relay working sensing the scene switch connected to I2.  Thank you.
However, I'm also experiencing quite a delay in the response/update - 2 to 3 seconds it seems.  Have you made any progress on that issue?

Expaso, 2019-12-06:
Or... I wrote again to Qubino Support.. and they told me this:

> Apologise I was sure i did reply you the results, the device actually with the settings you specified don't send the ON status as it should, we're already working on the fix and will be probably be ready towards end of January - beginning of February because of all the tests we do before we release any FW. The devices can be reprogrammed from us to the newer FW at that time.

>I will set the ticket On Hold like the ticket won't close and you won't get the automatic notifications from our system.