OpenZWave/open-zwave #952: Duco CO2 sensor

ronvangestel, 2016-07-31:
i'm trying to add my CO2 sensor from Duco. but i can't get it to work. could someone please send me in the right direction.
i made the xml file. i only get a ligth dimmer instead of a c02 reading and a temp that never changes.

[http://www.pepper-one.de/zwavedb/device/139](url)

`<?xml version="1.0" encoding="utf-8"?>

<Product xmlns='http://code.google.com/p/open-zwave/'>

  
  <CommandClass id="112">
    <Value genre="config" max="255" units="Sec" min="1" type="byte" index="1" label="Measure level" size="1" value="60">
      <Help>How often the value is measured</Help>
    </Value>
	<Value genre="config" max="32767" units="ppm" min="1" type="short" index="2" label="CO2 Setpoint" size="2" value="1000">
      <Help>CO2 Setpoint</Help>
    </Value>
	<Value genre="config" max="100" units="procent" min="0" type="byte" index="3" label="Humidity Setpoint" size="1" value="70">
      <Help>Humidity Setpoint</Help>
    </Value>
    <Value genre="config" max="100" units="procent" min="0" type="byte" index="4" label="Button 1" size="1" value="0">
      <Help>Button 1 level</Help>
    </Value>
    <Value genre="config" max="100" units="procent" min="0" type="byte" index="5" label="Button 2" size="1" value="50">
      <Help>Button 2 level</Help>
    </Value>
    <Value genre="config" max="100" units="procent" min="0" type="byte" index="6" label="Button 3" size="1" value="100">
      <Help>Button 3 level</Help>
    </Value>
    <Value genre="config" max="100" units="procent" min="0" type="byte" index="7" label="Auto min." size="1" value="15">
      <Help>min % bij auto modus</Help>
    </Value>
    <Value genre="config" max="32767" units="C" min="1" type="short" index="8" label="temp high" size="2" value="0">
      <Help>Temperature high setpoint</Help>
    </Value>
    <Value genre="config" max="32767" units="C" min="1" type="short" index="9" label="temp low" size="2" value="0">
      <Help>Temperature low setpoint</Help>
    </Value>
	<Value genre="config" max="32767" units="C" min="1" type="short" index="10" label="delta temp" size="2" value="2">
      <Help>Temperature delta setpoint</Help>
    </Value>
	<Value genre="config" max="32767" units="%" min="1" type="short" index="11" label="RH high" size="2" value="0">
      <Help>RH high setpoint</Help>
    </Value>
	<Value genre="config" max="32767" units="%" min="1" type="short" index="12" label="RH low" size="2" value="0">
      <Help>RH low setpoint</Help>
    </Value>
    <Value genre="config" max="32767" units="%" min="1" type="short" index="13" label="RH delta" size="2" value="5">
      <Help>RH delta setpoint</Help>
    </Value>
    <Value genre="config" max="32767" units="ppm" min="1" type="short" index="14" label="CO2 high" size="2" value="0">
      <Help>CO2 high setpoint</Help>
    </Value>
    <Value genre="config" max="32767" units="ppm" min="1" type="short" index="15" label="CO2 low" size="2" value="0">
      <Help>CO2 low setpoint</Help>
    </Value>
    <Value genre="config" max="32767" units="ppm" min="1" type="short" index="16" label="CO2 delta" size="2" value="500">
      <Help>CO2 delta setpoint</Help>
    </Value>
  </CommandClass>
  
  <CommandClass id="133">
    <Associations num_groups="7">
      <Group index="1" max_associations="20" label="Control" auto="true"/>
	  <Group index="2" max_associations="20" label="Temperature ON" auto="true"/>
	  <Group index="3" max_associations="20" label="Temperature OFF" auto="true"/>
	  <Group index="4" max_associations="20" label="Temperature Delta" auto="true"/>
	  <Group index="5" max_associations="20" label="CO2 ON" auto="true"/>
	  <Group index="6" max_associations="20" label="CO2 OFF" auto="true"/>
	  <Group index="7" max_associations="20" label="CO2 Delta" auto="true"/>
    </Associations>
  </CommandClass>

</Product>`

Fishwaldo, 2016-08-11:
Could you please send a OZW_Log.txt file of this device?

Please shutdown your app, remove zwcfg_*.xml and restart the app, and let it run for a while.

Fishwaldo, 2016-08-18:
No Feedback. Closing.

closed 2016-08-18 by Fishwaldo