OpenZWave/open-zwave #956: MinOZW is hanging when I run it

terimyaki, 2016-08-08:
Hi,

I am currently trying to run MinOZW just to test my connection to my **AeroTec Gen5 Z-Stick** on my Mac computer. But when I try to run it on the command line I just see it hang. This is first time playing with Z-Wave, so maybe I missed something along the way. The `options.xml` is the default one. Below is the log.

```
> MinOZW /dev/tty.usbmodem1431
Starting MinOZW with OpenZWave Version 1.4.340
2016-08-08 16:28:39.001 Error, Cannot find a path to the configuration files at ../../../config/, Using /usr/local/etc/openzwave/ instead...
2016-08-08 16:28:39.001 Warning, Failed to Parse /usr/local/etc/openzwave/options.xml: Failed to open file
2016-08-08 16:28:39.001 Warning, Failed to Parse options.xml: Failed to open file
2016-08-08 16:28:39.001 Always, OpenZwave Version 1.4.340 Starting Up
2016-08-08 16:28:39.001 Info, Setting Up Provided Network Key for Secure Communications
2016-08-08 16:28:39.001 Warning, Failed - Network Key Not Set
2016-08-08 16:28:39.001 Info, mgr,     Added driver for controller /dev/tty.usbmodem1431
2016-08-08 16:28:39.001 Info,   Opening controller /dev/tty.usbmodem1431
2016-08-08 16:28:39.001 Info, Trying to open serial port /dev/tty.usbmodem1431 (attempt 1)
```

Fishwaldo, 2016-08-11:
Wrong Serial Port. Please use hte correct device name in the dev tree.

closed 2016-08-11 by Fishwaldo

terimyaki, 2016-08-23:
ah thanks that worked!