OpenZWave/open-zwave #960: Z-Uno support. Use new Z-Wave Plus rules about MCA/A with LL

PoltoS, 2016-08-12:
According to the new Z-Wave Plus rules, if a device supports MultiChannel, and you need to get unsolicited reports from CCs on channels, you must set MultiChannelAssociation ctrlNodeId:0 to LL group instead of simple Association ctrlNodeId.

In old devices simple Associations can work, but in newly certified it will not and device will not send any unsolicited reports from channels.

All this is needed to support new devices including Z-Wave.me Z-Uno http://z-uno.z-wave.me

Fishwaldo, 2016-08-13:
I highly doubt its a new "Rule":
1) as it would break backwards compatibility for Associations with all pre-zwave+ devices
2) almost every Z-Wave+ device I have does not act this way, with the exception being Qubino

Regardless, please add 
`<CommandClass id="142" ForceInstances="true" />` to the config file of devices affected by this.

lastly:
This is a issue tracker and not a support forum. 

Please use our Forums for Questions regarding OpenZWave: 
https://groups.google.com/forum/#!forum/openzwave

Thanks

closed 2016-08-13 by Fishwaldo

PoltoS, 2016-08-13:
Yes, it's an issue tracker,
yes, it's a new rule in Z-Wave and
yes, this is an issue in OZW.

To be more specific, for devices supporting MCA v3 and being Z-Wave Plus OZW have to follow this rule. Please RTFM.

Hope this will make OZW better.