OpenZWave/open-zwave #961: uClibc's old stable pthread implementation makes resetcontroller to deadlock.

nihilus, 2016-08-12:
Using uClibc 0.9.32 and gcc/libstdc++ 4.6.3 makes factory reset to stall on removedriver in manager.cpp.

I believe this is a heisenbug which appear more frequently if built with uclibc and uclibc++ instead of libstdc++.

Pretty annoying for us embedded developers.

Fishwaldo, 2016-08-13:
patches to fix this issue would be needed, as I don't directly support uclibc. 

if its annoying enough, maybe you can find the root cause and send us the fix. :)

nihilus, 2016-08-13:
I will do more debugging on monday. Might be a bug in uclibc.

nihilus, 2016-08-14:
@Fishwaldo beats me, why don't OZW use the boost library for things that needs to be cross-platform?

Fishwaldo, 2016-08-15:
because Boost introduces a really complex dependency/building requirement to OZW. A lot of users are not programmers/developers so we try to keep it as self contained as possible. 

(I'm a fan of Boost, but as we don't really control the build enviroment here (after all, OZW is only a library), its pretty heavyweight.

nihilus, 2016-08-15:
@Fishwaldo ok... fwiw I can say this was a toolchain issue with libgcc and pthread_cancel(). Enabling NPTL in uClibc resolved the issue as far as I am concerned.

closed 2016-08-15 by nihilus