OpenZWave/open-zwave #963: COMMAND_CLASS_COLOR present in zwcfg causes exception

elcosmelco, 2016-08-14:
Hi,

I've posted a message a while ago detailing my current problems getting a current build of OZW running.
Maybe this is too broad an issue so I've decided to post my problems one at a time with the latest build.

Starting MinOZW with the latest library (1.4.360) with a zwcfg present (generated from the same build) causes an exception while loading the zwcfg.

The node is a Fibaro "FGRGBWM441 RGBW Controller" and it is crashing on parsing(?) the XML for Command Class 51 (the color command class) resulting in the following exception:
Unhandled exception thrown: read access violation. 
_Pnext was 0x1000004.

CallStack:
	MinOZW.exe!std::_Container_base12::_Orphan_all() Line 222	C++
 	MinOZW.exe!std::_Vector_alloc<std::_Vec_base_types<OpenZWave::CommandClass::RefreshValue *,std::allocator<OpenZWave::CommandClass::RefreshValue *> > >::_Orphan_all() Line 614	C++
 	MinOZW.exe!std::vector<OpenZWave::CommandClass::RefreshValue *,std::allocator<OpenZWave::CommandClass::RefreshValue *> >::_Reallocate(unsigned int _Count) Line 1620	C++
 	MinOZW.exe!std::vector<OpenZWave::CommandClass::RefreshValue *,std::allocator<OpenZWave::CommandClass::RefreshValue *> >::_Reserve(unsigned int _Count) Line 1633	C++
 	MinOZW.exe!std::vector<OpenZWave::CommandClass::RefreshValue *,std::allocator<OpenZWave::CommandClass::RefreshValue *> >::push_back(OpenZWave::CommandClass::RefreshValue * const & _Val) Line 1291	C++
 	MinOZW.exe!OpenZWave::CommandClass::ReadValueRefreshXML(const TiXmlElement * _ccElement) Line 366	C++
 	MinOZW.exe!OpenZWave::CommandClass::ReadXML(const TiXmlElement * _ccElement) Line 313	C++
 	MinOZW.exe!OpenZWave::Node::ReadCommandClassesXML(const TiXmlElement * _ccsElement) Line 1214	C++
 	MinOZW.exe!OpenZWave::Node::ReadXML(const TiXmlElement * _node) Line 1056	C++
 	MinOZW.exe!OpenZWave::Driver::ReadConfig() Line 751	C++
 	MinOZW.exe!OpenZWave::Driver::HandleSerialAPIGetInitDataResponse(unsigned char * _data) Line 2620	C++
 	MinOZW.exe!OpenZWave::Driver::ProcessMsg(unsigned char * _data) Line 1916	C++
 	MinOZW.exe!OpenZWave::Driver::ReadMsg() Line 1721	C++
 	MinOZW.exe!OpenZWave::Driver::DriverThreadProc(OpenZWave::Event * _exitEvent) Line 481	C++
 	MinOZW.exe!OpenZWave::Driver::DriverThreadEntryPoint(OpenZWave::Event * _exitEvent, void * _context) Line 385	C++
 	MinOZW.exe!OpenZWave::ThreadImpl::Run() Line 152	C++
 	MinOZW.exe!OpenZWave::ThreadImpl::ThreadProc(void * _pArg) Line 140	C++

This is what the XML of that CC looks like taken from the zwcfg:
....
```
<CommandClass id="51" name="COMMAND_CLASS_COLOR" version="1" request_flags="3" innif="true" colorchannels="29" coloridxbug="true">
	<Instance index="1" />
	<Value type="string" genre="user" instance="1" index="0" label="Color" units="#RRGGBBWW" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="#00000000" />
	<Value type="list" genre="user" instance="1" index="1" label="Color Index" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="0" size="18">
		<Item label="Off" value="0" />
		<Item label="Cool White" value="1" />
		<Item label="Warm White" value="2" />
		<Item label="Red" value="3" />
		<Item label="Lime" value="4" />
		<Item label="Blue" value="5" />
		<Item label="Yellow" value="6" />
		<Item label="Cyan" value="7" />
		<Item label="Magenta" value="8" />
		<Item label="Silver" value="9" />
		<Item label="Gray" value="10" />
		<Item label="Maroon" value="11" />
		<Item label="Olive" value="12" />
		<Item label="Green" value="13" />
		<Item label="Purple" value="14" />
		<Item label="Teal" value="15" />
		<Item label="Navy" value="16" />
		<Item label="Custom" value="17" />
	</Value>
	<Value type="int" genre="system" instance="1" index="2" label="Color Channels" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="-2147483648" max="2147483647" value="29" />
</CommandClass>
```
....

I am building and running this in Debug mode, using VS2015 Community Edition on Win 10.
I have no experience in debugging C++ so please let me know how I can be of more help.

Kind regards,
Elco

Fishwaldo, 2016-08-15:
Its not failing on that zwcfg_*.xml as far as I can see. Its failing in the ReadValueRefreshXML which is related to TriggerRefreshValue Entity. 

If you remove your zwcfg_*.xml and start, does it still crash?

elcosmelco, 2016-08-15:
Hi Justin,

No it does not - it only fails if the zwcfg_*.xml is present and it does so
pretty much immediately.

Cheers,
Elco

Fishwaldo, 2016-08-15:
Could you please post the full contents of the zwcfg_*.xml then please?

elcosmelco, 2016-08-15:
I've attached the file.
('ve renamed it as github doesn't allow attaching xml files
[zwcfg_0x______.txt](https://github.com/OpenZWave/open-zwave/files/418740/zwcfg_0x______.txt)

Cheers,
Elco

Fishwaldo, 2016-08-18:
ok.. firstly,
Did you edit zwcfg_*.xml at all? Its formatted weirdly around CC 51 (the Color CC). 

Could you post a full logfile as well? Maybe that has some hints. I just tried now, and it works fine.... 

You also have a Horstmann Thermostate that has this config, so it might be dying there... only way to tell is with a full log.

elcosmelco, 2016-08-18:
Hi Justin,

It looks like I accidentally saved it after I removing the whitespace so I could copy it into this bug report but that is it.

I've attached the initialization log and the one where it failed.

[OZW_Log_Fail.txt](https://github.com/OpenZWave/open-zwave/files/425509/OZW_Log_Fail.txt)
[OZW_Log_Init.txt](https://github.com/OpenZWave/open-zwave/files/425511/OZW_Log_Init.txt)

Thanks for taking the time to have a look btw!

Cheers,
Elco

elcosmelco, 2016-08-18:
By the way - my last physical device is node 66 but the log is full of non existent node IDs where usually 102 and 225 end up in the config, but many more make an appearance in the Log (247, 156, 140 etc)

Fishwaldo, 2018-10-26:
Closing old issue

closed 2018-10-26 by Fishwaldo