OpenZWave/open-zwave #964: missing support for linking with uclibc++

nihilus, 2016-08-14:
GetColor() method should be rewritten to support linkage with uclibc++ forembedded devices. I will attach my patch tomorrow.

Fishwaldo, 2016-08-15:
why? Whats wrong with the current implementation?

nihilus, 2016-08-15:
@Fishwaldo it is uclibc++ that doesn't handle sstreams.

nihilus, 2016-08-15:
``
Index: cpp/src/command_classes/Color.cpp
===================================================================
--- cpp/src/command_classes/Color.cpp	(revision 16819)
+++ cpp/src/command_classes/Color.cpp	(working copy)
@@ -309,7 +309,13 @@
 	string result = rgbstring.substr(((position - 1) * 2) +1, 2);
 	std::stringstream ss(result);
 	uint16 rawresult;
+
+	/* Markus - Breaks uClibc++ so use strtoul */
+#if 1
+	rawresult = static_cast<uint16>(strtoul(result.c_str(), NULL, 16));
+#else
 	ss >> std::hex >> rawresult;
+#endif
 	return rawresult;
 }
 
``

Fishwaldo, 2016-08-15:
I think there is going to be more "broken" cases here.... (on Dev branch as well)

might have to make our own "strtoul" type function in Utils.cpp/h and use that when we need to convert etc. 

what about the reverse, would this be broken as well:
`string OpenZWave::intToString( int x ) {
        return static_cast< std::ostringstream & >( ( std::ostringstream() << std::dec << x ) ).str();
}
`

nihilus, 2016-08-15:
Actually that does compile. Seems to be related to stringstreams.

nihilus, 2016-08-17:
This seems to have been fixed in uClibc++ 0.2.4.

Fishwaldo, 2016-08-18:
so is the fix required?

nihilus, 2016-08-18:
No

closed 2016-08-18 by nihilus