OpenZWave/open-zwave #966: Please add Z-Uno support

michapr, 2016-08-17:
Hi,

Z-Uno is a Z-Wave Plus certified device, using Arduino you can create different sensors/actors.
XML was already send here
https://github.com/OpenZWave/open-zwave/pull/962

Using this XML the device is working for some simple Z-Wave devices.

Using the Z-Wave Plus *certified* 10-channel sketch the sensors will not report the values to gateway.
Sensors are Multi-Channel-Sensors - I see in OZW log only one value (luminance), but not the temperature value.
Values will be send - I see it in Arduino debug print.
( http://z-uno.z-wave.me/examples/Certified10Channels/ )

Maybe it is like described in #960 ?
I have added the suggested line to XML - it didn't help (logs were created with XML including this line - attached)

If I poll the device - I can see the values in log. So only the notification with new values seems not to work (polling can not be the solution...).

Same device (certified 10-channel sketch) is working in Z-Way (yes,... it's same manufacturer ;) ...) and in Fibaro HomeCenter without problems - all devices will be recognized and values will be changed without polling.

Have you any idea how it can work? Many users (technical users / developers) are working with OpenZwave solutions - so would be great to find a solution here.

for reference: I use OZW 1.4-340-g71af464-dirty

[OZW_Log_z-uno.txt](https://github.com/OpenZWave/open-zwave/files/422259/OZW_Log_z-uno.txt)
[OZW_Log_z-uno_polling.txt](https://github.com/OpenZWave/open-zwave/files/422260/OZW_Log_z-uno_polling.txt)
[zuno.xml.txt](https://github.com/OpenZWave/open-zwave/files/422262/zuno.xml.txt)

Thanks!
Michael

nechry, 2016-08-17:
@michapr  if you can redo the PR for simple usage. the old PR is broken

michapr, 2016-08-17:
Hi,
sorry, what do you mean with PR ?

michapr, 2016-08-17:
Ahh... guess PR = polling request?

Problem is not the polling. Values should be reported **without** polling (like in Fibaro HC is working).
Polling was only a second try (that's why in second log) - to get the values.

nechry, 2016-08-17:
Pull Request to in first step add the device as supported devices, event is not fully supported yet.

michapr, 2016-08-17:
ok, made it, got some temperature values using the polling option.

Switched back (normal mode withut polling) - same as before. Only values I get is Luminance.

I think it is something with the MultiChannel Association but the
`<CommandClass id="142" ForceInstances="true" />`
doesn't help here.

Thanks!

Fishwaldo, 2016-08-17:
in the Log File, ForceInstances wasn't set correctly.
The ``<CommandClass id="142"....`` should be outside the other CommandClass variable. You can see examples in other config files..

michapr, 2016-08-17:
Sorry, my mistake while uploading file... I have used another copy of the config XML.
Just have rechecked - the right file is online:
<?xml version="1.0" encoding="utf-8"?>
<Product xmlns='http://code.google.com/p/open-zwave/'>
    
    
    <CommandClass id="112">
        <Value type="list" index="1" genre="config" label="Debug mode" units="" min="0" max="1" value="0" size="1">
            <Help>Debug mode to apply channels, associations and power mode changes on the fly without exclusion-inclusion sequence. Use for debugging only!</Help>
            <Item label="Disabled (default)" value="0"/>
            <Item label="Enabled" value="1"/>
        </Value>
        <Value type="list" index="2" genre="config" label="Blink Service LED" units="" min="0" max="1" value="1" size="1">
            <Help>Blink with service LED during operation</Help>
            <Item label="Disabled" value="0"/>
            <Item label="Enabled (default)" value="1"/>
        </Value>
    </CommandClass>
    <CommandClass id="142" ForceInstances="true" />
</Product>

Issue is still the same, sorry for confusion whlie uploading wrong config file.

Michael

nechry, 2016-08-17:
and please give us the product type and id

michapr, 2016-08-17:
sorry, I thought it is in OZW.log inside...
Product Type ID: 	0x0110
Product ID: 	0x0001
http://www.pepper1.net/zwavedb/device/985

Fishwaldo, 2016-08-18:
Based on the logs, its not picking up the ForceInstances. Did you remove the zwcfg_*.xml and restart after making the change?

nechry, 2016-08-18:
I put a preliminary device configuration file.

michapr, 2016-08-18:
@Fishwaldo - yes, of course I did. Multiple times before reporting ;)
@nechry - will just test it.

michapr, 2016-08-18:
No changes - only the Luminance will be reported.
There no any additional "Received" lines in log.
Do you want to see a new log - can make a clear start log with only this device inside again.

Fishwaldo, 2016-08-18:
a new log, with no filtering is best....

michapr, 2016-08-18:
[OZW_Log_zuno2.txt](https://github.com/OpenZWave/open-zwave/files/424537/OZW_Log_zuno2.txt)

its a full starting log - only with this device.
You see the initial value of temperature, but no more reports.

Used the XML from nechry above.

Thanks!

Fishwaldo, 2016-08-18:
> 2016-08-18 11:21:42.536 Detail, Node001, QueryStage_CacheLoad

zwcfg_*.xml was present when you ran the last log. can you remove and redo the log?

michapr, 2016-08-18:
sorry,... attached
[OZW_Log_zuno3.txt](https://github.com/OpenZWave/open-zwave/files/424610/OZW_Log_zuno3.txt)

Fishwaldo, 2016-08-18:
> 2016-08-18 12:00:00.701 Detail, Node001, QueryStage_CacheLoad

still there.......

michapr, 2016-08-18:
confused... but you are right.  But now it isn't here more,.. sorry.
[OZW_Log_zuno4.txt](https://github.com/OpenZWave/open-zwave/files/424637/OZW_Log_zuno4.txt)

Fishwaldo, 2016-08-18:
and the saga continues..... :+1: 

can you remove the node, restart OZW, remove any zwcfg_*.xml file and re-add it?

The MultiInstanceAssociation is only set when you include a new node. it wont alter the group memberships of existing nodes if Node 1 is already present.

michapr, 2016-08-18:
ok, done.
And I see in log now Temperature values - great! And values are changing even in Domoticz.
Strange that Domoticz will not add the Luminance sensor - But this maybe anything there (I hope so ;) ...)
But I see in log:
`2016-08-18 13:27:52.219 Error, Node040, Cannot find endpoint map to instance for Command Class COMMAND_CLASS_SENSOR_MULTILEVEL endpoint 0`

[OZW_Log_zuno5.txt](https://github.com/OpenZWave/open-zwave/files/424772/OZW_Log_zuno5.txt)

Fishwaldo, 2016-08-18:
endpoint 0? Thats a new one. I always thought endpoint 0 was only used if a device didn't support the MultiChannel CC. 

But, we only have implemented version 2 of the MultiChannel CC, and I see this reporting as version 4, so maybe endpoint 0 has a new meaning in later versions. 

Looking at the logs, it seems to be always set with Endpoint 8 messages. Is there anything "missing"?

Might want to ask on the zwave.me guys what endpoint 0 is? (and let us know the reply)

PoltoS, 2016-08-20:
Yes, you should re-read latest Command Classes specification (SDS12657-6) section 4.75.3. It describes the new feature.

Before channel 0 was not allowed, not it means report to root device.

Not going deep in the spec details the rule is pretty simple (see #960 - you closed it personally):
- if device is Z-Wave Plus and MCA v3 is supported, set in LL MCA 1:0
- else if device is Z-Wave Plus and MCA is not supported or version < 3, set in LL A 1
- else do like you did before for non-Z-Wave Plus

First line is the most important, second and third you are free to do like you did before

Fishwaldo, 2016-08-23:
@PoltoS I'm not sure you realise, but OZW does not have access to the Sigma SDK or any of the documentation, so I can't "re-read" any specifications, as we do our development by reverse engineering and testing with "real world" devices. 

You might want to be mindful of what information you share in this case (as you are under NDA) as I don't want to see you guys get introuble. (But I do appreciate the help your providing)

But for this particular issue, can you confirm that the report sent to instance 0, is a duplicate of the prior message:

>2016-08-18 13:26:59.445 Detail, Node040,   Received: 0x01, 0x0f, 0x00, 0x04, 0x00, 0x28, 0x09, 0x60, 0x0d, 0x08, 0x01, 0x31, 0x05, 0x03, 0x01, 0x24, 0xa3
2016-08-18 13:26:59.452 Detail, 
2016-08-18 13:26:59.454 Info, Node040, Received a MultiChannelEncap from node 40, endpoint 8 for Command Class COMMAND_CLASS_SENSOR_MULTILEVEL
2016-08-18 13:26:59.455 Info, Node040, Received SensorMultiLevel report from node 40, instance 2, Luminance: value=36%
2016-08-18 13:26:59.456 Detail, Node040, Refreshed Value: old value=30, new value=36, type=decimal
2016-08-18 13:26:59.458 Detail, Node040, Changes to this value are not verified
2016-08-18 13:26:59.460 Detail, Node040, Notification: ValueChanged
2016-08-18 13:26:59.471 Detail, Node040,   Received: 0x01, 0x0f, 0x00, 0x04, 0x00, 0x28, 0x09, 0x60, 0x0d, 0x00, 0x01, 0x31, 0x05, 0x03, 0x01, 0x24, 0xab
2016-08-18 13:26:59.473 Detail, 
2016-08-18 13:26:59.474 Error, Node040, Cannot find endpoint map to instance for Command Class COMMAND_CLASS_SENSOR_MULTILEVEL endpoint 0

Right now, our MCA implementation is only at version 2, but the workaround is provided in the other issue by creating a ForceInstances option for the MCA CommandClass in each devices config files. (that should address the first point in your email above)

Hopefully very soon, we will be able to catch up on our CommandClass versions as I have a long list of things todo around here.

PoltoS, 2016-08-24:
@michapr is this ForceInstances enough to use the device? Are all reports handled by OZW?

@Fishwaldo I knew that previously OZW was made without access to specs, but was not aware how you work now. With new S2 you will have problems without proper documents.

michapr, 2016-08-24:
> @michapr is this ForceInstances enough to use the device? Are all reports handled by OZW?

I do not know in this time about "all reports" - have used for reference the certified sketch.
If we can have this issue for open for a while - I will test more devices (code at Z-Uno) and report it here.
If no more issues, I will close it.
@Fishwaldo - ok?

jotakar, 2016-09-16:
is this issue closed?

michapr, 2016-09-16:
can you give me some additional days?
I will comment it - did not forgot about this. Had some other tasks but will do it as soon as possible ;)

Thanks!

michapr, 2017-01-23:
No more problems found in last time, can be closed for now ;)

closed 2017-01-23 by michapr