OpenZWave/open-zwave #967: HeatIT thermostat: supported thermostat setpoints

Mirarkitty, 2016-08-17:
Either the thermostat or openzwave is doing something strange.

Packet received:
# 2016-08-17 17:43:58.932 Detail, Node011,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x0b, 0x03, 0x43, 0x05, 0x82, 0x3e

gives supported setpoints 2 and 8 (bit 1 and 7 in 0x82) ("cooling 1" and "dry air").

The actual supported setpoints are 1 (heating) and 11 (heating econ). Creating them manually (patching openzwave) allows me to change those and it works fine. 67.2 (cooling) also works - it keeps the value - but it doesn't seem to be supported by the device.

Is this interesting enough to investigate?

Does the support bits work well with other thermostats?

Fishwaldo, 2016-08-18:
The device is messed up :)

OZW is parsing the packet correctly based on what is being sent.... 

it works fine with most "working" devices. Suggest you report this to the manufacturer and see what they say.

Mirarkitty, 2016-08-19:
Thanks for the reply. Feel free to close the ticket.

closed 2016-08-19 by Fishwaldo