OpenZWave/open-zwave #968: "Unhandled command class 0x8f" until device is woken up

Mirarkitty, 2016-08-18:
I have a few Philio Slim door sensors (gen 5). They will send events with MULTI_CMD command class when anything happens, but they rarely wake up for openzwave to get command class information.

Typical output:
```
# 2016-08-18 08:16:07.418 Detail, Node013,   Received: 0x01, 0x24, 0x00, 0x04, 0x00, 0x0d, 0x1e, 0x8f, 0x01, 0x04, 0x03, 0x80, 0x03, 0x64, 0x09, 0x71, 0x05, 0x00, 0x00, 0x00, 0xff, 0x07, 0x08, 0x00, 0x05, 0x31, 0x05, 0x03, 0x01, 0x2f, 0x06, 0x31, 0x05, 0x01, 0x22, 0x01, 0x0e, 0x2d
# 2016-08-18 08:16:07.418 Detail, 
# 2016-08-18 08:16:07.418 Info, Node013, ApplicationCommandHandler - Unhandled Command Class 0x8f

```

If I push the tamper switch it negotiates this with openzwave, but I can't do that every time I restart openzwave. This is what should happen:

```
# 2016-08-18 08:31:29.791 Detail, Node013,   Received: 0x01, 0x24, 0x00, 0x04, 0x00, 0x0d, 0x1e, 0x8f, 0x01, 0x04, 0x03, 0x80, 0x03, 0x64, 0x09, 0x71, 0x05, 0x00, 0x00, 0x00, 0xff, 0x07, 0x08, 0x00, 0x05, 0x31, 0x05, 0x03, 0x01, 0x2e, 0x06, 0x31, 0x05, 0x01, 0x22, 0x01, 0x0e, 0x2c
# 2016-08-18 08:31:29.791 Detail, 
# 2016-08-18 08:31:29.791 Info, Node013, Received encapsulated multi-command from node 13
# 2016-08-18 08:31:29.791 Info, Node013, Received Battery report from node 13: level=100
# 2016-08-18 08:31:29.791 Detail, Node013, Initial read of value
# 2016-08-18 08:31:29.791 Info, Node013, Received Alarm report: type=0, level=0, sensorSrcID=0, type:Burglar event:8, status=255
# 2016-08-18 08:31:29.791 Detail, Node013, Refreshed Value: old value=0, new value=0, type=byte
# 2016-08-18 08:31:29.791 Detail, Node013, Changes to this value are not verified
# 2016-08-18 08:31:29.792 Detail, Node013, Refreshed Value: old value=0, new value=0, type=byte
# 2016-08-18 08:31:29.792 Detail, Node013, Changes to this value are not verified
# 2016-08-18 08:31:29.792 Detail, Node013, Refreshed Value: old value=0, new value=0, type=byte
# 2016-08-18 08:31:29.792 Detail, Node013, Changes to this value are not verified
# 2016-08-18 08:31:29.792 Detail, Node013, Initial read of value
# 2016-08-18 08:31:29.792 Info, Node013, Received SensorMultiLevel report from node 13, instance 1, Luminance: value=46%
# 2016-08-18 08:31:29.792 Detail, Node013, Refreshed Value: old value=53, new value=46, type=decimal
# 2016-08-18 08:31:29.792 Detail, Node013, Changes to this value are not verified
# 2016-08-18 08:31:29.792 Info, Node013, Received SensorMultiLevel report from node 13, instance 1, Temperature: value=27.0C
# 2016-08-18 08:31:29.792 Detail, Node013, Refreshed Value: old value=26.5, new value=27.0, type=decimal
# 2016-08-18 08:31:29.792 Detail, Node013, Changes to this value are not verified
# 2016-08-18 08:31:29.792 Info, Node013, End of encapsulated multi-command from node 13
# 2016-08-18 08:31:29.792 Detail, Node013, Notification: ValueChanged
# 2016-08-18 08:31:29.792 Detail, Node013, Notification: ValueChanged
# 2016-08-18 08:31:29.792 Detail, Node013, Notification: ValueChanged
# 2016-08-18 08:31:29.792 Detail, Node013, Notification: ValueChanged
# 2016-08-18 08:31:29.792 Detail, Node013, Notification: ValueChanged
# 2016-08-18 08:31:29.792 Detail, Node013, Notification: ValueChanged
# 2016-08-18 08:31:29.792 Detail, Node013, Notification: ValueChanged
```

Is there any way openzwave can accept that it's getting information even though it doesn't know that the device supports MULTI_CMD yet? 

(Or what is going on here?)

Mirarkitty, 2016-08-18:
Digging deeper, it seems like openzwave will ignore the device until it can get the supported command classes, even if I force it to create the MultiCmd one. 

This is very annoying with devices that sleep... any workaround?

nechry, 2016-08-18:
We have the same behavior with the fibaro fgk101 (door/window) sensor. Until the first wakeup no binarysensor report.

Fishwaldo, 2016-08-18:
Thats the way OZW works. After restart we need to complete the QueryStages... until then, Sleeping Devices will not be "ready".

No work around exists...

closed 2016-08-18 by Fishwaldo

tridiumcontrols, 2017-10-01:
@Fishwaldo Any update to this? I have a few smoke sensors that are battery powered, and I get the ApplicationCommandHandler - Unhandled Command Class error in my logs,