OpenZWave/open-zwave #973: Ozwcp running error

Jeongki, 2016-08-20:
Hi, I am a developer in Korea.

I have a problem and it is really hard for me to solve it..

First of all, the logs appear are below..

1970-01-01 00:09:42.074 Info, Serial port /dev/ttyUSB0 opened (attempt 1)
1970-01-01 00:09:42.082 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
1970-01-01 00:09:42.082 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
1970-01-01 00:09:42.082 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
1970-01-01 00:09:42.083 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
1970-01-01 00:09:42.083 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
1970-01-01 00:09:42.083 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_RF_POWER_LEVEL_GET: 0x01, 0x04, 0x00, 0xba, 0x0a, 0x4b
1970-01-01 00:09:42.084 Detail,
1970-01-01 00:09:42.084 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x15) - FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
1970-01-01 00:09:43.096 Error, contrlr, ERROR: Dropping command, expected response not received after 1 attempt(s)
1970-01-01 00:09:43.096 Detail, contrlr, Removing current message
1970-01-01 00:09:43.096 Detail, contrlr, Notification: Notification - TimeOut
1970-01-01 00:09:43.099 Info, Notification: Notification home 00000000 node 255 Timeout
1970-01-01 00:09:43.100 Detail,
1970-01-01 00:09:43.100 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x20) - FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc

So, I checked where the error log is from.. and it is from the WriteMsg function in open-zwave/driver.cpp
and It looks like.. compile is fine, calling function is fine, but now my controller can not find any nodeId..

Does it mean that currently the controller's USB connection is not well?

When I connect the Z-Wave USB to the board, I can see the logs about ttyUSB0.. so, I put it into the Device name field. But after I initiate the controller, there is nothing happened and I can see only the logs like upper.. 

I need your help..
Thanks for reading and have a nice weekend.

Fishwaldo, 2016-08-23:
This sounds like a either a incorrect USB port/device, or a dead controller. Have you verified with any other software?

Jeongki, 2016-08-24:
Yes, I did. First of all, I checked the Z-Wave module on PC-Controller, and it works well.
I am not pretty sure that USB works correctly, but when I plugged Z-Wave USB module into the board's USB port, I can see some logs and find the USB port name. I heard that it means connection is fine..

Currently, I use SerialController and I checked it's process was working well, just what I found was there is nothing to read from the port, so every time it notified Time Out whenever I press initial button.

There are some configuration files formed .xml or .xsd.. I guess I should check these files' path in open-zwave code.

Fishwaldo, 2016-09-05:
if your setting it to the correct port, but not getting anything back, its hardware/OS issue. bear in mind the devicename is different for different types of devices. I would check "dmesg" from the startup of the kernel to confirm the serial port name. If your 100% sure you have the correct serial port, then its something else.

closed 2016-09-05 by Fishwaldo

Jeongki, 2016-09-27:
Really Appreciate it for your help and your point is correct. If I saw your
mail I could fix it much quicker.

I have been quite busy for it and have other project, so it was late to
send you the reply. Really Sorry..

I found where this problem is from. It was the USB port problem. After it
was updated newly, problem was solved.

Really Thank you very much for your help.