OpenZWave/open-zwave #976: About Qubino multiclass association

jotakar, 2016-08-23:
This is not an issue but I think it could be interesting. In this conversation
[https://github.com/OpenZWave/open-zwave/issues/857]
I read how Qubino manage multiclass association, speaking about ZMNHBDx, double relay module. And I asked to Qubino about the new firmware S5 and I received this answer

>  As you’ve noticed there is a difference between the way the S4 and S5 versions send unsolicited (Reports sent without preceding Get commands) reports to controllers. On S4 the module reports the status of the switches to the controller on every change, regardless of the way the single channel or multi channel lifeline associations were set. We implemented this in a way similar to other manufacturers, but later on had a discussion with Sigma Designs, where they explained to us the recommended way of sending unsolicited reports (they require strict separation between single channel and multi channel contexts on z-wave plus products), which we implemented in the S5 version in such a way:
> -	If a single channel lifeline association (group 1 associated to the controller) is set then the module shouldn’t report unsolicited Multi channel Encapsulated Reports, which means only the root device will report it’s states without a Get request, not the endpoints (these need to be polled in order to receive reports). From the module’s standpoint this means the controller either doesn’t support multi channel context or the module was integrated in a non-multi channel context.
> -	On the other hand, if a multi channel lifeline association (group 1 on each endpoint is associated to the controller’s endpoint 0 or 1, depending on integration) was set the module will report unsolicited reports from all the device’s endpoints, but not the root device. This will result in the endpoint’s statuses being reported via unsolicited reports also, not just Get commands via polling. 
> 
> Additionally there are other association groups that can be set on the module (via single channel or multi channel associations) in order to filter the amount of report traffic desired.

So, in firmware S5 module doesn't report root device in multiclass context!!. Does OpenZwave consider this fact and activate root device when I1 or I2 is activated?

Bye.

kjamsek, 2016-08-23:
> So, in firmware S5 module doesn't report root device in multiclass context!!. Does OpenZwave consider this fact and activate root device when I1 or I2 is activated?

To clarify the case of unsolicited report sending, in case a single channel lifeline group association was set, the root device reports data on behalf of it's endpoints where possible, BUT not Multi Channel Encapsulated (This causes a problem in regards to switch state and data reporting for 2 outputs since the root device can only report one state).

In case a multi channel lifeline group association is set then each endpoint reports it's own states and data, resulting in each output's data being sent to the controller with it's endpoint number.

When data is polled by the controller then it is irrelevant which association context was used, the module will reply according to the format of Get requests received (encapsulated or not).

jotakar, 2016-08-23:
The, for Qubino double realy, the difference between  firmware S4 and the last S5 is that in **S5 root device doesn't report status in multichannel context**, it just reports status for endpoints. Root needs polling or, better, refresh variable. Indeed root switch is not essential.
S4 did report status from root and from every endpoint in multichannel context. So root sends ON if at least one of endpoints is ON.

Fishwaldo, 2016-08-23:
in MCA (with a ForceInstances config option) the Root device becomes irrelevent, as it doesn't represent any actual "device" (by my understanding), but rather its a "something changed, you should check it". 

If MCA is setup with a endpoint of 0, then we get endpoint 1 or 2 reporting, and hence, we don't really care about the root device anymore.... 

And for the very last time, please use the mailing list. These discussions are best suited there. Next time, I will simply delete these "discussion" issues...

closed 2016-08-23 by Fishwaldo