OpenZWave/open-zwave #979: AssumeAwake option to false did not perform correct device interview

nechry, 2016-08-25:
in relation with the closed issues https://github.com/OpenZWave/open-zwave/issues/666
If the option AssumeAwake is set to false this issue still present.
Set to true everything work as expected.

But this option is also great for a fast network startup. For example a network with 40 devices, half a battery operated the network reach the Topology Loaded in 50 sec, if the AssumeAwake is set to false. With true theTopology Loaded occurs after only 10 minutes. So about 12 time longer.

For AssumeAwake = false, openzwave won't attempt to communicate with sleeping devices, so they will never send the NoMoreInfo message to them till the query stages starts on them.

Fishwaldo, 2016-09-05:
Ok. This would be a difficult one to handle.
AssumeAwake  = True (default) means that sleeping devices will timeout during startup, hence the extra time it takes to query. For those that are "awake", the NoMoreInfo will get sent at the end of the query process, hence issue #441 and #666 are addressed.

If AssumeAwake = false, then we never try to talk to devices marked as Sleeping. If one of them is awake during startup, then a NoMoreInfo would ever be queued till the QueryStage starts...

When you say it doesn't perform the correct interview, are you refering to the fact that a NoMoreInfo is never sent or something else?

nechry, 2016-09-05:
I do more tests around this weird behavior. I thing is not general like the issue https://github.com/OpenZWave/open-zwave/issues/666. I thing now, is more a specific device issue for a Vision Secure ZG8108.
 
For this device, if I set AssumeAwake = True, the device will completed the interview after the first wakeup.
But with AssumeAwake = False, he will never restart the interview. I will receive notification about the sensor changing but nothing else. If I manually wakeup the device the interview will completed as expected. 

I will confirm with others Vision Secure devices, with AssumeAwake = False, if everything work as expected. if yes, I will close this issue and try to find a way to figure how to deal with this device.

nechry, 2016-09-09:
I did more tests with the device and now with AssumeAwake set to False or True the interview is always completed successfully. So this not a ozw issue

closed 2016-09-09 by nechry