OpenZWave/open-zwave #983: new false endpoints added every time I cycle a switch.

ThomasADavis, 2016-08-29:
I am using the Vision ZL7432 In-Wall Switch, Dual Relay in my house, and have tried this with both Domoticz and Homeassistant.

Every time the hardwired switch is used, a new instance/endpoint is created in the application/zwave config.

This does not occur when I have toggled the switch via zwave.  I also have to set polling in either domoticz or homeassistant to get the status of either switch/relay (this is an expected behavior)

How do I help debug this?  What else can I do to get this fixed?  I have a git pulled version of openzwave and ozwcp working (I needed to submit a previous pull request for the GE duplex receptacle.)

In the following snippet from the ozw_cfg file, any instance after 3 is a bad/fake instance.  I normally end up editing them out, and then restarting the application to make it work.  The count stops around 128, which is probably the max instance count.

        <Node id="23" name="" location="" basic="4" generic="16" specific="1" type="Binary Power Switch" listening="true" frequentListening="false" beaming="true" routing="true" max_baud_rate="40000" version="4" query_stage="Complete">
                <Manufacturer id="109" name="Vision">
                        <Product type="2017" id="1717" name="ZL7432 In-Wall Switch, Dual Relay" />
                </Manufacturer>
                <CommandClasses>
                        <CommandClass id="32" name="COMMAND_CLASS_BASIC" version="1" request_flags="5" mapping="37">
                                <Instance index="1" />
                                <Instance index="2" endpoint="1" />
                                <Instance index="3" endpoint="2" />
                                <Instance index="4" endpoint="2" />
                                <Instance index="5" endpoint="2" />
                                <Instance index="6" endpoint="2" />
                                <Instance index="7" endpoint="1" />
                                <Instance index="8" endpoint="2" />
                                <Instance index="9" endpoint="1" />
                                <Instance index="10" endpoint="2" />
                        </CommandClass>
                        <CommandClass id="37" name="COMMAND_CLASS_SWITCH_BINARY" version="1" request_flags="5" innif="true">
                                <Instance index="1" />
                                <Instance index="2" endpoint="1" />
                                <Instance index="3" endpoint="2" />
                                <Instance index="4" endpoint="2" />
                                <Instance index="5" endpoint="2" />
                                <Instance index="6" endpoint="2" />
                                <Instance index="7" endpoint="1" />
                                <Instance index="8" endpoint="2" />
                                <Instance index="9" endpoint="1" />
                                <Instance index="10" endpoint="2" />
                                <Value type="bool" genre="user" instance="1" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
                                <Value type="bool" genre="user" instance="2" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
                                <Value type="bool" genre="user" instance="3" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
                                <Value type="bool" genre="user" instance="4" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
                                <Value type="bool" genre="user" instance="5" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
                                <Value type="bool" genre="user" instance="6" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
                                <Value type="bool" genre="user" instance="7" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
                                <Value type="bool" genre="user" instance="8" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
                                <Value type="bool" genre="user" instance="9" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
                                <Value type="bool" genre="user" instance="10" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
                        </CommandClass>
                        <CommandClass id="39" name="COMMAND_CLASS_SWITCH_ALL" version="1" request_flags="5" innif="true">
                                <Instance index="1" />
                                <Value type="list" genre="system" instance="1" index="0" label="Switch All" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="3" size="1">
                                        <Item label="Disabled" value="0" />
                                        <Item label="Off Enabled" value="1" />
                                        <Item label="On Enabled" value="2" />
                                        <Item label="On and Off Enabled" value="255" />
                                </Value>
                        </CommandClass>
                        <CommandClass id="96" name="COMMAND_CLASS_MULTI_INSTANCE/CHANNEL" version="3" request_flags="1" innif="true">
                                <Instance index="1" />
                        </CommandClass>
                        <CommandClass id="114" name="COMMAND_CLASS_MANUFACTURER_SPECIFIC" version="1" request_flags="5" innif="true">
                                <Instance index="1" />
                        </CommandClass>
                        <CommandClass id="133" name="COMMAND_CLASS_ASSOCIATION" version="1" request_flags="5" innif="true">
                                <Instance index="1" />
                                <Associations num_groups="1">
                                        <Group index="1" max_associations="5" label="Group 1" auto="true">
                                                <Node id="1" />
                                        </Group>
                                </Associations>
                        </CommandClass>
                        <CommandClass id="134" name="COMMAND_CLASS_VERSION" version="1" request_flags="5" innif="true">
                                <Instance index="1" />
                                <Value type="string" genre="system" instance="1" index="0" label="Library Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="6" />
                                <Value type="string" genre="system" instance="1" index="1" label="Protocol Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="3.67" />
                                <Value type="string" genre="system" instance="1" index="2" label="Application Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="13.07" />
                        </CommandClass>
                </CommandClasses>
        </Node>

Fishwaldo, 2016-08-29:
Sounds like a endpoint mapping issue we sometimes hit with some devices. 

Try something like the mappings="endpoint" in the config file. See config/homeseer/hsm100.xml for a example. 

(Posting from iPhone so I can't quote it right now)

ThomasADavis, 2016-08-29:
Ah, this might work.  I'll create a config file for if and see what happens.

from the hsm100.xml
>
    
	<CommandClass id="32" action="add" />
	
	<CommandClass id="96" mapping="endpoints" />
	
	<CommandClass id="133">
		<Associations num_groups="1">
			<Group index="1" max_associations="4" label="Motion"/>
		</Associations>
	</CommandClass>

ThomasADavis, 2016-08-30:
I created a new config file in config/vision/zl7432.xml, modded the manufacturer_specific.xml to use it, ran homeassist, switched on/off one switch, and on the same device switched off that side.  See attached logfile/

zl7432.xml:

>
[root@router config]# more vision/zl7432.xml 
<?xml version="1.0" encoding="utf-8"?>
<Product xmlns='http://code.google.com/p/open-zwave/'>
    
        
        < CommandClass id="32" mapping="endpoints" />
        < CommandClass id="37" mapping="endpoints" />
        
</Product>

I switched one on, then off, and got this logfile snapshot:

OZW_log.txt has:

>
grep Node021 OZW_Log.txt 
2016-08-29 19:20:58.640 Detail, Node021, AdvanceQueries queryPending=0 queryRetries=0 queryStage=CacheLoad live=1
2016-08-29 19:20:58.640 Detail, Node021, QueryStage_CacheLoad
2016-08-29 19:20:58.640 Info, Node021, Node Identity Codes: 0109:2017:1717
2016-08-29 19:20:58.640 Info, Node021, NoOperation::Set - Routing=true
2016-08-29 19:20:58.640 Detail, Node021, Queuing (NoOp) NoOperation_Set (Node=21): 0x01, 0x09, 0x00, 0x13, 0x15, 0x02, 0x00, 0x00, 0x25, 0x16, 0xc1
2016-08-29 19:20:58.640 Detail, Node021, Queuing (Query) Query Stage Complete (CacheLoad)
2016-08-29 19:20:59.032 Detail, Node021, Notification: NodeAdded
2016-08-29 19:20:59.037 Detail, Node021, Notification: NodeProtocolInfo
2016-08-29 19:20:59.043 Detail, Node021, Notification: EssentialNodeQueriesComplete
2016-08-29 19:20:59.054 Detail, Node021, Notification: ValueAdded
2016-08-29 19:20:59.056 Info, Node021, DisablePoll failed - value not on list
2016-08-29 19:20:59.058 Detail, Node021, Notification: ValueAdded
2016-08-29 19:20:59.059 Info, Node021, DisablePoll failed - value not on list
2016-08-29 19:20:59.060 Detail, Node021, Notification: ValueAdded
2016-08-29 19:20:59.063 Info, Node021, DisablePoll failed - value not on list
2016-08-29 19:20:59.068 Detail, Node021, Notification: ValueAdded
2016-08-29 19:20:59.071 Detail, Node021, Notification: Group
2016-08-29 19:20:59.072 Detail, Node021, Notification: ValueAdded
2016-08-29 19:20:59.074 Detail, Node021, Notification: ValueAdded
2016-08-29 19:20:59.075 Detail, Node021, Notification: ValueAdded
2016-08-29 19:20:59.076 Detail, Node021, Notification: NodeNaming
2016-08-29 19:21:15.750 Info, Node021, Sending (NoOp) message (Callback ID=0x16, Expected Reply=0x13) - NoOperation_Set (Node=21): 0x01, 0x09, 0x00, 0x13, 0x15, 0x02, 0x00, 0x00, 0x25, 0x16, 0xc1
2016-08-29 19:21:15.756 Detail, Node021,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2016-08-29 19:21:15.756 Detail, Node021,   ZW_SEND_DATA delivered to Z-Wave stack
2016-08-29 19:21:15.768 Detail, Node021,   Received: 0x01, 0x05, 0x00, 0x13, 0x16, 0x00, 0xff
2016-08-29 19:21:15.768 Detail, Node021,   ZW_SEND_DATA Request with callback ID 0x16 received (expected 0x16)
2016-08-29 19:21:15.768 Info, Node021, Request RTT 17 Average Request RTT 17
2016-08-29 19:21:15.768 Detail, Node021, Removing current message
2016-08-29 19:21:15.768 Detail, Node021, Notification: Notification - NoOperation
2016-08-29 19:21:25.176 Detail, Node021, Query Stage Complete (CacheLoad)
2016-08-29 19:21:25.176 Detail, Node021, AdvanceQueries queryPending=0 queryRetries=0 queryStage=Associations live=1
2016-08-29 19:21:25.176 Detail, Node021, QueryStage_Associations
2016-08-29 19:21:25.176 Info, Node021, Number of association groups reported for node 21 is 1.
2016-08-29 19:21:25.176 Info, Node021, Get Associations for group 1 of node 21
2016-08-29 19:21:25.176 Detail, Node021, Queuing (Send) AssociationCmd_Get (Node=21): 0x01, 0x0a, 0x00, 0x13, 0x15, 0x03, 0x85, 0x02, 0x01, 0x25, 0x1c, 0x4f
2016-08-29 19:21:25.176 Detail, Node021, Queuing (Query) Query Stage Complete (Associations)
2016-08-29 19:21:25.176 Info, Node021, Sending (Send) message (Callback ID=0x1c, Expected Reply=0x04) - AssociationCmd_Get (Node=21): 0x01, 0x0a, 0x00, 0x13, 0x15, 0x03, 0x85, 0x02, 0x01, 0x25, 0x1c, 0x4f
2016-08-29 19:21:25.182 Detail, Node021,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2016-08-29 19:21:25.182 Detail, Node021,   ZW_SEND_DATA delivered to Z-Wave stack
2016-08-29 19:21:25.230 Detail, Node021,   Received: 0x01, 0x05, 0x00, 0x13, 0x1c, 0x00, 0xf5
2016-08-29 19:21:25.230 Detail, Node021,   ZW_SEND_DATA Request with callback ID 0x1c received (expected 0x1c)
2016-08-29 19:21:25.230 Info, Node021, Request RTT 53 Average Request RTT 35
2016-08-29 19:21:25.237 Detail, Node021,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x15, 0x06, 0x85, 0x03, 0x01, 0x05, 0x00, 0x01, 0x67
2016-08-29 19:21:25.238 Info, Node021, Response RTT 60 Average Response RTT 60
2016-08-29 19:21:25.238 Info, Node021, Received Association report from node 21, group 1, containing 1 associations
2016-08-29 19:21:25.238 Info, Node021,   The group contains:
2016-08-29 19:21:25.238 Info, Node021,     Node 1
2016-08-29 19:21:25.238 Info, Node021, Querying associations for node 21 is complete.
2016-08-29 19:21:25.238 Detail, Node021,   Expected reply and command class was received
[... 95 lines cut ...]
2016-08-29 19:22:44.522 Info, Node021, Received MultiChannelCapabilityReport from node 21 for endpoint 1
2016-08-29 19:22:44.522 Info, Node021,     Endpoint is not dynamic, and is a Binary Power Switch
2016-08-29 19:22:44.522 Info, Node021,     Command classes supported by the endpoint are:
2016-08-29 19:22:44.522 Info, Node021,         COMMAND_CLASS_BASIC
2016-08-29 19:22:44.522 Info, Node021,         COMMAND_CLASS_SWITCH_BINARY
[... cut ...]
2016-08-29 19:22:44.605 Info, Node021, DisablePoll failed - value not on list
2016-08-29 19:22:44.606 Detail, Node021,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x15, 0x03, 0x25, 0x03, 0x00, 0xc2
2016-08-29 19:22:44.606 Info, Node021, Received SwitchBinary report from node 21: level=Off
2016-08-29 19:22:44.606 Detail, Node021, Refreshed Value: old value=false, new value=false, type=bool
2016-08-29 19:22:44.606 Detail, Node021, Changes to this value are not verified
2016-08-29 19:22:44.606 Detail, Node021, Notification: ValueChanged

ThomasADavis, 2016-08-30:
Ah, success! I found a typo, and fixed that, restarted and now it's stopped adding them.

I do not see any tomorrow, I will generate a pull request for this.

ThomasADavis, 2016-08-31:
Well, it's still adding them.

Using this:

    <?xml version="1.0" encoding="utf-8"?>
    <Product xmlns='http://code.google.com/p/open-zwave/'>
      
          
          <CommandClass id="32" mapping="endpoints" />
          <CommandClass id="37" mapping="endpoints" />
          <CommandClass id="96" mapping="endpoints" />
          
   </Product>

I get this:

        <Node id="21" name="" location="" basic="4" generic="16" specific="1" type="Binary Power Switch" listening="true" frequentListening="false" beaming="true" routing="true" max_baud_rate="40000" version="4" query_stage="Complete">
                <Manufacturer id="109" name="Vision">
                        <Product type="2017" id="1717" name="ZL7432 In-Wall Switch, Dual Relay" />
                </Manufacturer>
                <CommandClasses>
                        <CommandClass id="32" name="COMMAND_CLASS_BASIC" version="1" request_flags="5" mapping="37">
                                <Instance index="1" endpoint="1" />
                                <Instance index="2" endpoint="2" />
                                <Instance index="3" endpoint="1" />
                                <Instance index="4" endpoint="1" />
                        </CommandClass>
                        <CommandClass id="37" name="COMMAND_CLASS_SWITCH_BINARY" version="1" request_flags="5" innif="true">
                                <Instance index="1" endpoint="1" />
                                <Instance index="2" endpoint="2" />
                                <Instance index="3" endpoint="1" />
                                <Instance index="4" endpoint="1" />
                                <Value type="bool" genre="user" instance="1" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
                                <Value type="bool" genre="user" instance="2" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
                                <Value type="bool" genre="user" instance="3" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
                                <Value type="bool" genre="user" instance="4" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
                        </CommandClass>
                        <CommandClass id="39" name="COMMAND_CLASS_SWITCH_ALL" version="1" request_flags="5" innif="true">
                                <Instance index="1" />
                                <Value type="list" genre="system" instance="1" index="0" label="Switch All" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="3" size="1">
                                        <Item label="Disabled" value="0" />
                                        <Item label="Off Enabled" value="1" />
                                        <Item label="On Enabled" value="2" />
                                        <Item label="On and Off Enabled" value="255" />
                                </Value>
                        </CommandClass>
                        <CommandClass id="96" name="COMMAND_CLASS_MULTI_INSTANCE/CHANNEL" version="3" request_flags="1" innif="true" mapping="endpoints">
                                <Instance index="1" />
                        </CommandClass>
                        <CommandClass id="114" name="COMMAND_CLASS_MANUFACTURER_SPECIFIC" version="1" request_flags="5" innif="true">
                                <Instance index="1" />
                        </CommandClass>
                        <CommandClass id="133" name="COMMAND_CLASS_ASSOCIATION" version="1" request_flags="5" innif="true">
                                <Instance index="1" />
                                <Associations num_groups="1">
                                        <Group index="1" max_associations="5" label="Group 1" auto="true">
                                                <Node id="1" />
                                        </Group>
                                </Associations>
                        </CommandClass>
                        <CommandClass id="134" name="COMMAND_CLASS_VERSION" version="1" request_flags="5" innif="true">
                                <Instance index="1" />
                                <Value type="string" genre="system" instance="1" index="0" label="Library Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="6" />
                                <Value type="string" genre="system" instance="1" index="1" label="Protocol Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="3.67" />
                                <Value type="string" genre="system" instance="1" index="2" label="Application Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="13.07" />
                        </CommandClass>
                </CommandClasses>
        </Node>

and
[... 116 lines cut ...]
[... cut ...]
2016-08-30 18:31:42.755 Info, Node021, DisablePoll failed - value not on list
2016-08-30 18:31:42.760 Detail, Node021,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x15, 0x03, 0x25, 0x03, 0x00, 0xc2
2016-08-30 18:31:42.760 Info, Node021, Received SwitchBinary report from node 21: level=Off
2016-08-30 18:31:42.760 Detail, Node021, Refreshed Value: old value=true, new value=false, type=bool
2016-08-30 18:31:42.760 Detail, Node021, Changes to this value are not verified
2016-08-30 18:31:42.761 Detail, Node021, Notification: ValueChanged

Fishwaldo, 2016-08-31:
Can you provide a full log file (no filtering) after removing your zwcfg_*.xml file and restarting

ThomasADavis, 2016-09-05:
Here's a file with a few switches turned on/off, no filtering, from start to finish of the application.

[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/455672/OZW_Log.txt)

Fishwaldo, 2016-09-15:
Hi,

> 2016-09-05 15:26:05.742 Detail, Node019,   Received: 0x01, 0x0d, 0x00, 0x04, 0x04, 0x13, 0x07, 0x60, 0x0a, 0x02, 0x10, 0x01, 0x20, 0x25, 0x9a
2016-09-05 15:26:05.742 Detail, 
2016-09-05 15:26:05.742 Info, Node019, Received MultiChannelCapabilityReport from node 19 for endpoint 2
2016-09-05 15:26:05.742 Info, Node019,     Endpoint is not dynamic, and is a Binary Power Switch
2016-09-05 15:26:05.742 Info, Node019,     Command classes supported by the endpoint are:
2016-09-05 15:26:05.743 Info, Node019,         COMMAND_CLASS_BASIC
2016-09-05 15:26:05.743 Info, Node019,         COMMAND_CLASS_SWITCH_BINARY
2016-09-05 15:26:05.743 Detail, Node019, Notification: ValueAdded

This device looks buggy, and looking at the Specs, it seems like its messed up:

1) its sending a "unsolicited" MultiChannelCapabilityReport when sending a notification. Normally, this is only requested during the "interview" stage and isn't required after that. 

You can fix this up by adding
`<CommandClass id="96" mapping="endpoints" ignoreUnsolicitedMultiChnCapReport="true" />`
to the config file for your device.

But the bigger issue here is that its then sending the Value Updates after that, but not encapsulated in a MultiInstanceEncap Report, which is how we differentiate different endpoints. Without that information, we can't tell what instance has changed, and just assume instance No 1.

>2016-09-05 15:26:05.758 Detail, Node019,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x13, 0x03, 0x25, 0x03, 0x00, 0xc4
2016-09-05 15:26:05.758 Detail, 
2016-09-05 15:26:05.758 Info, Node019, Received SwitchBinary report from node 19: level=Off
2016-09-05 15:26:05.758 Detail, Node019, Refreshed Value: old value=false, new value=false, type=bool
2016-09-05 15:26:05.758 Detail, Node019, Changes to this value are not verified
2016-09-05 15:26:05.758 Detail, Node019, Notification: ValueChanged

So, so I don't think you are going to make much progress getting this device to work well with OZW. You might want to contact their support and point them to this comment, and see if they have a fix, or workaround etc, otherwise we are out of luck..

ThomasADavis, 2016-09-21:
What a freaking piece of junk.  I picked them up from Monoprice, and they no longer sell them from looks of them.  I will be replacing them with something else that works.

Fishwaldo, 2016-09-21:
haha. Trust me, i know the feeling. 

Ironically, I've been having a chat with one of the Sigma Field Engineers the last couple of days. Let me quote him:

> Multi-channel is ..... (Trust me, even customers with Dev Kits, Support access, and on-site help get this wrong over and over...)

bhasden, 2017-02-17:
For anyone else that comes across this and is using Home Assistant, I was able to get around this issue by using the customize functionality and wildcards in configuration.yaml. Basically, I used a wildcard to hide everything related to the zd2102 sensor by default and then used specific entity ids to selectively show only the sensors I want.

Basically, something like:

    customize:
      - entity_id: sensor.vision_zd2102_au_doorwindow_sensor_*
        hidden: true
      - entity_id: sensor.vision_zd2102_au_doorwindow_sensor_alarm_level_18_1
        friendly_name: 'Front Door'
        hidden: false