OpenZWave/open-zwave #985: Sigma Designs opens up more of Z-Wave

bbqkees, 2016-09-01:
Sigma Designs decided to open up more of Z-Wave to the public domain.
Will probably be useful for OpenZWave too.

Press release:
[](http://www.sigmadesigns.com/news/sigma-designs-releases-z-wave-interoperability-layer-into-the-public-domain/)

[](http://z-wave.sigmadesigns.com/design-z-wave/z-wave-public-specification/)

Hellorheaven, 2016-09-01:
http://z-wave.sigmadesigns.com/design-z-wave/z-wave-public-specification/

nechry, 2016-09-01:
I hope is not a issue for you.
Please use the mailing list for discutions. 
A thread already exist
https://groups.google.com/forum/#!topic/openzwave/3rmb5a-IDWM

Of course is a great think and opportunity for us

closed 2016-09-01 by nechry