OpenZWave/open-zwave #986: OpenZWave Mutex Lock

Ryszard-Trojnacki, 2016-09-04:
I have a problem with hardware:
https://groups.google.com/forum/#!topic/openzwave/tgtXfSude5I
and for workaround I'm reloading driver (Manager::RemoveDriver, Manager::AddDriver) but unfortunately sometimes library (OpenZWave) hangs on a mutex.

I'm doing Remove, Sleep, Add, Sleep:
```java
						synchronized(lock) {
							try {
								Manager.removeDriver(zwaveDevice);
							}catch(Exception e) {
								return;
							}
							try {
								Thread.sleep(1000);
							} catch (InterruptedException e) { return; }
							try {
								Manager.addDriver(zwaveDevice);
							}catch(Exception e) {
								return;
							}
							log.info("Driver added");
							try {
								Thread.sleep(3000);
							} catch (InterruptedException e) { return; }
						}

```
Synchronized block is for blocking calls to Manager.
I'm sleeping 3000 ms to wait for reload and after that my polling thread is calling Manager::RefreshValue and it is locked.

My Stacktrace (from Java - jstack) is:
```
----------------- 8917 -----------------
0x00007f7bf8c76b5d      __lll_lock_wait + 0x1d
0x00007f7ba6acdb9f      _ZN9OpenZWave5Mutex4LockEb + 0x27
0x00007f7ba6aa64d7      _ZN9OpenZWave9LockGuardC1EPNS_5MutexE + 0x2f
0x00007f7ba6aa0fc2      _ZN9OpenZWave7Manager12RefreshValueERKNS_7ValueIDE + 0x68
0x00007f7ba6a746cd      Java_pl_rtprog_paigo_zwave_Manager_refreshValue + 0xb4
0x00007f7be9f3ad7f      <Unknown compiled code>
```
And GDB backtrace for this thread is:
```
Thread 36 (Thread 0x7f7ba72b5700 (LWP 8917)):
#0  __lll_lock_wait () at ../sysdeps/unix/sysv/linux/x86_64/lowlevellock.S:135
#1  0x00007f7bf8c70d82 in __GI___pthread_mutex_lock (mutex=0x7f7bb8007f18) at ../nptl/pthread_mutex_lock.c:115
#2  0x00007f7ba6ad0433 in OpenZWave::MutexImpl::Lock (this=0x7f7bb8007f10, _bWait=true) at /home/jenkins/jobs/paigo-inteligence/workspace/smarthome/src/main/cpp/open-zwave/cpp/src/platform/unix/MutexImpl.cpp:82
#3  0x00007f7ba6acdb9f in OpenZWave::Mutex::Lock (this=0x7f7bb80183e0, _bWait=true) at /home/jenkins/jobs/paigo-inteligence/workspace/smarthome/src/main/cpp/open-zwave/cpp/src/platform/Mutex.cpp:73
#4  0x00007f7ba6aa64d7 in OpenZWave::LockGuard::LockGuard (this=0x7f7ba72b44e0, mutex=0x7f7bb80183e0) at /home/jenkins/jobs/paigo-inteligence/workspace/smarthome/src/main/cpp/open-zwave/cpp/src/Utils.h:83
#5  0x00007f7ba6aa0fc2 in OpenZWave::Manager::RefreshValue (this=0x7f7bc800a370, _id=...) at /home/jenkins/jobs/paigo-inteligence/workspace/smarthome/src/main/cpp/open-zwave/cpp/src/Manager.cpp:2925
#6  0x00007f7ba6a746cd in Java_pl_rtprog_paigo_zwave_Manager_refreshValue (env=0x7f7baca699f8, vid=0x7f7ba72b45e0) at manager.cpp:375
#7  0x00007f7be9f3ad7f in ?? ()
#8  0x00000000f67c9a60 in ?? ()
#9  0x0000000000000000 in ?? ()
```

I'm not sure what is blocking this mutex, because in backtrace I'm only seeing other thread that are waiting for this mutex, but I don't know what is keeping it.

Java Stacktrace and GDB backtrace for all threads is in attachment.
[backtrace.txt](https://github.com/OpenZWave/open-zwave/files/454000/backtrace.txt)

Fishwaldo, 2016-09-05:
Try uncommenting the std:cout lines in cpp/src/Defs.h around the LockGuard functions.

Or read here: https://en.wikibooks.org/wiki/Linux_Applications_Debugging_Techniques/Deadlocks

It should give some hints as to whats holding the Lock.

Fishwaldo, 2017-02-17:
Please test the latest master branch and let us know if its is resolved.

closed 2017-03-21 by Fishwaldo