OpenZWave/open-zwave #987: How to make meters report in 1 decimal instead of 3?

ado464, 2016-09-08:
Hello, what do i change in the code to make the meters report in 1 decimal instead of the current 3 decimal values?

Best regards,
Ado

nechry, 2016-09-08:
you can use the override_precision attribute

ado464, 2016-09-08:
hello nechry, how can i do this to all 22 meters at once?

best regards,
ado

nechry, 2016-09-08:
yes if you apply the policy on the CC all value will reported with only 1 decimal.

but you want change because the device report not right the values or is your wish ?

the CC policy can be apply to the config file if is a device misconfiguration. if is only a wish do the precision in your application

ado464, 2016-09-09:
Its simply me wanting less decimals, the meters are correct as is. 

How do i go about doing this? Is it in the options.xml file?

Do i put it like this in the options.xml:
Override_precision=1
?

nechry, 2016-09-09:
In this case we will not modify the device definition.

My approach is not long time right, in this case, you must do this feature in your application.

This is a issue tracker and not a support forum.

Please use our Forums for Questions regarding OpenZWave: 
https://groups.google.com/forum/#!forum/openzwave

Thanks

closed 2016-09-09 by nechry