OpenZWave/open-zwave #989: Missing BASIC and NODE_NAMING classes for Stellaz radiator thermostat

magorski, 2016-09-26:
Hi,

It seems that there are missing `COMMAND_CLASS_BASIC` (32) and `COMMAND_CLASS_NODE_NAMING` (119) classes missing in `/config/eurotronic/eur_stellaz.xml` config file. I'm afraid I'm not familiar enough with those XML-s to propose a PR.

The thing is that in order to change this the thermostat mode (i.e. to Direct Control) you need to change both `COMMAND_CLASS_BASIC` and `COMMAND_CLASS_THERMOSTAT_MODE` values.

Please see link below for details:
http://manuals.zwaveeurope.com/make.php?lang=en&type=&sku=EUR_STELLAZ

For time being `COMMAND_CLASS_BASIC` is missing from configuration therefore I'm not able to change thermostat mode using openzwave-node-shared. :( (i.e. getting ` OpenZWave valueId not found` error when trying to set 31 as value of class 32 to 254)

Can you please help?

nechry, 2016-09-26:
> Switching between the four modes is accomplished using the Z-Wave command classes 'Thermostat Mode' and 'Basic'. The following mappings apply:
> 
> Frost Protection: BASIC SET 0x01, THERMOSTAT MODE SET 0x00
> Energy Saving: BASIC SET 0x00, THERMOSTAT MODE SET 0x0b
> Comfort: BASIC SET 0xff, THERMOSTAT MODE SET 0x01
> Direct Control: BASIC SET 0xfe, THERMOSTAT MODE SET 0x1f

is a mapping if you use direct association. you don't have to set the Basic CC

magorski, 2016-09-27:
OK, so I understand that `BASIC` doesn't need to be send in order to change the mode.

The thing is that it's impossible to change thermostat to Direct Controll through `THERMOSTAT MODE`.

When I'm trying to do that I'm getting this error: `Received unknown thermostat mode: index 31`

I've pin pointed it to comming from [line 276](https://github.com/OpenZWave/open-zwave/blob/5c7ac3329a4b18c644bd34a70454c102a3fdb0ef/cpp/src/command_classes/ThermostatMode.cpp#L276) of `open-zwave/cpp/src/command_classes/ThermostatMode.cpp` file.

I think that thermostat accept's that but openzwave can't interpret the massage correctly.
`Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x03, 0x03, 0x40, 0x03, 0x1f, 0xae`

Any ideas why it doesn't accept value 31 as thermostat mode?

Fishwaldo, 2016-09-27:
Would need to see the full Log file (after removing zwcfg_*.xml) to see what modes the device reports.

magorski, 2016-09-28:
After initial scan complete, I'm waiting for first wakeup. When It happens I'm getting:
```
[34m2016-09-27 20:08:59.564 Detail, Node003,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x03, 0x02, 0x84, 0x07, 0x71
[39m[34m2016-09-27 20:08:59.565 Detail, 
[39m[39m2016-09-27 20:08:59.565 Info, Node003, Received Wakeup Notification from node 3
[39m[39m2016-09-27 20:08:59.565 Info, Node003,   Node 3 has been marked as awake
[39m[34m2016-09-27 20:08:59.565 Detail, Node003, Queuing (Query) Query Stage Complete (Probe)
[39m[34m2016-09-27 20:08:59.565 Detail, Node003, Notification: Notification - Node Awake
[39m[34m2016-09-27 20:08:59.566 Detail, Node003, Query Stage Complete (Probe)
[39m[34m2016-09-27 20:08:59.566 Detail, Node003, AdvanceQueries queryPending=0 queryRetries=0 queryStage=WakeUp live=1
[39m[34m2016-09-27 20:08:59.566 Detail, Node003, QueryStage_WakeUp
[39m[34m2016-09-27 20:08:59.570 Detail, Node003, Queuing (WakeUp) WakeUpCmd_IntervalGet (Node=3): 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x84, 0x05, 0x25, 0x4e, 0x0e
[39m[34m2016-09-27 20:08:59.571 Detail, Node003, Queuing (Query) Query Stage Complete (WakeUp)
[39m[34m2016-09-27 20:08:59.571 Detail, 
[39m[39m2016-09-27 20:08:59.571 Info, Node003, Sending (WakeUp) message (Callback ID=0x4e, Expected Reply=0x04) - WakeUpCmd_IntervalGet (Node=3): 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x84, 0x05, 0x25, 0x4e, 0x0e
[39m[34m2016-09-27 20:08:59.578 Detail, Node003,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
[39m[34m2016-09-27 20:08:59.578 Detail, Node003,   ZW_SEND_DATA delivered to Z-Wave stack
[39m[34m2016-09-27 20:08:59.683 Detail, Node003,   Received: 0x01, 0x07, 0x00, 0x13, 0x4e, 0x00, 0x00, 0x0b, 0xae
[39m[34m2016-09-27 20:08:59.683 Detail, Node003,   ZW_SEND_DATA Request with callback ID 0x4e received (expected 0x4e)
[39m[39m2016-09-27 20:08:59.683 Info, Node003, Request RTT 112 Average Request RTT 112
[39m[34m2016-09-27 20:08:59.684 Detail,   Expected callbackId was received
[39m[34m2016-09-27 20:08:59.732 Detail, Node003,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x03, 0x06, 0x84, 0x06, 0x00, 0x00, 0xf0, 0x01, 0x81
[39m[34m2016-09-27 20:08:59.732 Detail, 
[39m[39m2016-09-27 20:08:59.732 Info, Node003, Response RTT 160 Average Response RTT 160
[39m[39m2016-09-27 20:08:59.732 Info, Node003, Received Wakeup Interval report from node 3: Interval=240, Target Node=1
[39m[34m2016-09-27 20:08:59.732 Detail, Node003, Initial read of value
[39m[34m2016-09-27 20:08:59.732 Detail, Node003,   Expected reply and command class was received
[39m[34m2016-09-27 20:08:59.733 Detail, Node003,   Message transaction complete
[39m[34m2016-09-27 20:08:59.733 Detail, 
[39m[34m2016-09-27 20:08:59.733 Detail, Node003, Removing current message
[39m[34m2016-09-27 20:08:59.733 Detail, Node003, Notification: ValueChanged
[39m[34m2016-09-27 20:08:59.733 Detail, Node003, Query Stage Complete (WakeUp)
[39m[34m2016-09-27 20:08:59.733 Detail, Node003, AdvanceQueries queryPending=0 queryRetries=0 queryStage=ManufacturerSpecific1 live=1
[39m[34m2016-09-27 20:08:59.734 Detail, Node003, QueryStage_ManufacturerSpecific1
[39m[34m2016-09-27 20:08:59.734 Detail, Node003, Checking for ManufacturerSpecific CC and Requesting values if present on this node
[39m[34m2016-09-27 20:08:59.734 Detail, Node003, Queuing (Query) ManufacturerSpecificCmd_Get (Node=3): 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x72, 0x04, 0x25, 0x4f, 0xf8
[39m[34m2016-09-27 20:08:59.734 Detail, Node003, Queuing (Query) Query Stage Complete (ManufacturerSpecific1)
[39m[34m2016-09-27 20:08:59.734 Detail, 
[39m[39m2016-09-27 20:08:59.734 Info, Node003, Sending (Query) message (Callback ID=0x4f, Expected Reply=0x04) - ManufacturerSpecificCmd_Get (Node=3): 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x72, 0x04, 0x25, 0x4f, 0xf8
[39m[34m2016-09-27 20:08:59.741 Detail, Node003,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
[39m[34m2016-09-27 20:08:59.742 Detail, Node003,   ZW_SEND_DATA delivered to Z-Wave stack
[39m[34m2016-09-27 20:08:59.853 Detail, Node003,   Received: 0x01, 0x07, 0x00, 0x13, 0x4f, 0x00, 0x00, 0x0c, 0xa8
[39m[34m2016-09-27 20:08:59.853 Detail, Node003,   ZW_SEND_DATA Request with callback ID 0x4f received (expected 0x4f)
[39m[39m2016-09-27 20:08:59.853 Info, Node003, Request RTT 118 Average Request RTT 115
[39m[34m2016-09-27 20:08:59.854 Detail,   Expected callbackId was received
[39m[34m2016-09-27 20:08:59.902 Detail, Node003,   Received: 0x01, 0x0e, 0x00, 0x04, 0x00, 0x03, 0x08, 0x72, 0x05, 0x01, 0x48, 0x00, 0x01, 0x00, 0x01, 0xc0
[39m[34m2016-09-27 20:08:59.903 Detail, 
[39m[39m2016-09-27 20:08:59.903 Info, Node003, Response RTT 168 Average Response RTT 164
[39m[39m2016-09-27 20:08:59.903 Info, Node003,   Opening config param file /usr/local/etc/openzwave/eurotronic/eur_stellaz.xml
[39m[39m2016-09-27 20:08:59.903 Info, Node003, Unable to find or load Config Param file /usr/local/etc/openzwave/eurotronic/eur_stellaz.xml
[39m[39m2016-09-27 20:08:59.904 Info, Node003, Received manufacturer specific report from node 3: Manufacturer=EUROtronic, Product=EUR_STELLAZ Wall Radiator Thermostat Valve Control
[39m[39m2016-09-27 20:08:59.904 Info, Node003, Node Identity Codes: 0148:0001:0001
[39m[34m2016-09-27 20:08:59.904 Detail, Node003,   Expected reply and command class was received
[39m[34m2016-09-27 20:08:59.904 Detail, Node003,   Message transaction complete
[39m[34m2016-09-27 20:08:59.905 Detail, 
[... cut ...]
[39m[39m2016-09-27 20:09:19.626 Info, Node003, Node not responding - moving message to Wake-Up queue: NodeNamingCmd_LocationGet (Node=3): 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x77, 0x05, 0x25, 0x5b, 0xe8
[39m[39m2016-09-27 20:09:19.627 Info, Node003, Node not responding - moving message to Wake-Up queue: WakeUpCmd_IntervalGet (Node=3): 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x84, 0x05, 0x25, 0x5c, 0x1c
[39m[39m2016-09-27 20:09:19.627 Info, Node003, Node not responding - moving QueryStageComplete command to Wake-Up queue
[39m[34m2016-09-27 20:09:19.627 Detail, Node003, Notification: Notification - Node Asleep
[39m
```

magorski, 2016-09-28:
You probably can see line where it says `Received unknown thermostat mode: 0x1f`.

I think it's coming from [line 300](https://github.com/OpenZWave/open-zwave/blob/5c7ac3329a4b18c644bd34a70454c102a3fdb0ef/cpp/src/command_classes/ThermostatMode.cpp#L300) of [open-zwave/cpp/src/command_classes/ThermostatMode.cpp](https://github.com/OpenZWave/open-zwave/blob/5c7ac3329a4b18c644bd34a70454c102a3fdb0ef/cpp/src/command_classes/ThermostatMode.cpp) file.

I think that array `c_modeName[]` needs to have more (at least 32) elements i.e. :
```
static char const* c_modeName[] =
{
	"Off",
	"Heat",
	"Cool",
	"Auto",
	"Aux Heat",
	"Resume",
	"Fan Only",
	"Furnace",
	"Dry Air",
	"Moist Air",
	"Auto Changeover",
	"Heat Econ",
	"Cool Econ",
	"Away",
	"Unknown",
	"Unknown",
	"Unknown",
	"Unknown",
	"Unknown",
	"Unknown",
	"Unknown",
	"Unknown",
	"Unknown",
	"Unknown",
	"Unknown",
	"Unknown",
	"Unknown",
	"Unknown",
	"Unknown",
	"Unknown",
	"Unknown",
	"Manual Valve Control"
};
```
That should solve the issue?

Fishwaldo, 2016-09-28:
I repeat. I need to see the full OZW_Log.txt file without zwcfg_*.xml to diagnose. 

During the interview stage, we ask the device what modes it supports. Without seeing that exchange, we can't diagnose anything

magorski, 2016-09-28:
OK, here you go. The thermostat is node #3.
```
2016-09-28 16:27:07.630 Always, OpenZwave Version 1.3.0 Starting Up
2016-09-28 16:27:07.631 Info, Setting Up Provided Network Key for Secure Communications
2016-09-28 16:27:07.632 Info, mgr,     Added driver for controller /dev/ttyAMA0
2016-09-28 16:27:07.632 Info,   Opening controller /dev/ttyAMA0
2016-09-28 16:27:07.632 Info, Trying to open serial port /dev/ttyAMA0 (attempt 1)
2016-09-28 16:27:07.632 Info, Serial port /dev/ttyAMA0 opened (attempt 1)
2016-09-28 16:27:07.633 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2016-09-28 16:27:07.633 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2016-09-28 16:27:07.634 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2016-09-28 16:27:07.634 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2016-09-28 16:27:07.634 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2016-09-28 16:27:07.634 Detail, 
2016-09-28 16:27:07.635 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x15) - FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2016-09-28 16:27:07.638 Detail, contrlr,   Received: 0x01, 0x10, 0x01, 0x15, 0x5a, 0x2d, 0x57, 0x61, 0x76, 0x65, 0x20, 0x33, 0x2e, 0x39, 0x39, 0x00, 0x01, 0x95
2016-09-28 16:27:07.638 Detail, 
2016-09-28 16:27:07.638 Info, contrlr, Received reply to FUNC_ID_ZW_GET_VERSION:
2016-09-28 16:27:07.638 Info, contrlr,     Static Controller library, version Z-Wave 3.99
2016-09-28 16:27:07.638 Detail, Node045,   Expected reply was received
2016-09-28 16:27:07.639 Detail, Node045,   Message transaction complete
2016-09-28 16:27:07.639 Detail, 
2016-09-28 16:27:07.639 Detail, contrlr, Removing current message
2016-09-28 16:27:07.639 Detail, 
2016-09-28 16:27:07.639 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x20) - FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2016-09-28 16:27:07.642 Detail, contrlr,   Received: 0x01, 0x08, 0x01, 0x20, 0xe9, 0x46, 0x5f, 0x75, 0x01, 0x52
2016-09-28 16:27:07.642 Detail, 
2016-09-28 16:27:07.642 Info, contrlr, Received reply to FUNC_ID_ZW_MEMORY_GET_ID. Home ID = 0xe9465f75.  Our node ID = 1
2016-09-28 16:27:07.643 Detail, Node070,   Expected reply was received
2016-09-28 16:27:07.643 Detail, Node070,   Message transaction complete
2016-09-28 16:27:07.643 Detail, 
2016-09-28 16:27:07.643 Detail, contrlr, Removing current message
2016-09-28 16:27:07.643 Detail, 
2016-09-28 16:27:07.643 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x05) - FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2016-09-28 16:27:07.646 Detail, contrlr,   Received: 0x01, 0x04, 0x01, 0x05, 0x1c, 0xe3
2016-09-28 16:27:07.646 Detail, 
2016-09-28 16:27:07.646 Info, contrlr, Received reply to FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES:
2016-09-28 16:27:07.646 Info, contrlr,     There is a SUC ID Server (SIS) in this network.
2016-09-28 16:27:07.646 Info, contrlr,     The PC controller is an inclusion static update controller (SUC) and was the original primary before the SIS was added.
2016-09-28 16:27:07.647 Detail, Node227,   Expected reply was received
2016-09-28 16:27:07.647 Detail, Node227,   Message transaction complete
2016-09-28 16:27:07.647 Detail, 
2016-09-28 16:27:07.647 Detail, contrlr, Removing current message
2016-09-28 16:27:07.647 Detail, 
2016-09-28 16:27:07.648 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x07) - FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2016-09-28 16:27:07.653 Detail, contrlr,   Received: 0x01, 0x2b, 0x01, 0x07, 0x05, 0x00, 0x01, 0x47, 0x04, 0x00, 0x00, 0x01, 0xfe, 0x83, 0xff, 0x88, 0xcf, 0x1f, 0x00, 0x00, 0xfb, 0x9f, 0x7d, 0xa0, 0x67, 0x00, 0x80, 0x80, 0x00, 0x80, 0x86, 0x00, 0x00, 0x00, 0xe8, 0x73, 0x00, 0x00, 0x0e, 0x00, 0x00, 0x40, 0x1a, 0x00, 0x59
2016-09-28 16:27:07.653 Detail, 
2016-09-28 16:27:07.654 Info, contrlr,  Received reply to FUNC_ID_SERIAL_API_GET_CAPABILITIES
2016-09-28 16:27:07.654 Info, contrlr,     Serial API Version:   5.0
2016-09-28 16:27:07.654 Info, contrlr,     Manufacturer ID:      0x0147
2016-09-28 16:27:07.654 Info, contrlr,     Product Type:         0x0400
2016-09-28 16:27:07.654 Info, contrlr,     Product ID:           0x0001
2016-09-28 16:27:07.654 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_RANDOM: 0x01, 0x04, 0x00, 0x1c, 0x20, 0xc7
2016-09-28 16:27:07.655 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_INIT_DATA: 0x01, 0x03, 0x00, 0x02, 0xfe
2016-09-28 16:27:07.655 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_SET_TIMEOUTS: 0x01, 0x05, 0x00, 0x06, 0x64, 0x0f, 0x97
2016-09-28 16:27:07.655 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_APPL_NODE_INFORMATION: 0x01, 0x07, 0x00, 0x03, 0x01, 0x02, 0x01, 0x00, 0xf9
2016-09-28 16:27:07.655 Detail,   Expected reply was received
2016-09-28 16:27:07.655 Detail,   Message transaction complete
2016-09-28 16:27:07.655 Detail, 
2016-09-28 16:27:07.656 Detail, contrlr, Removing current message
2016-09-28 16:27:07.656 Detail, 
2016-09-28 16:27:07.656 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x56) - FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2016-09-28 16:27:07.658 Detail, contrlr,   Received: 0x01, 0x04, 0x01, 0x56, 0x01, 0xad
[... cut ...]
2016-09-28 16:29:30.853 Detail, Node003, AdvanceQueries queryPending=0 queryRetries=0 queryStage=NodeInfo live=1
2016-09-28 16:29:30.853 Detail, Node003, QueryStage_NodeInfo
2016-09-28 16:29:30.853 Detail, Node003, Queuing (Query) Request Node Info (Node=3): 0x01, 0x04, 0x00, 0x60, 0x03, 0x98
2016-09-28 16:29:30.853 Detail, Node003, Queuing (Query) Query Stage Complete (NodeInfo)
2016-09-28 16:29:30.854 Detail, 
2016-09-28 16:29:30.854 Info, Node003, Sending (Query) message (Callback ID=0x00, Expected Reply=0x49

Fishwaldo, 2016-09-28:
It advertises that it only supports Heating 1 and Furnace:

> 2016-09-28 16:29:31.867 Detail, Node003,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x03, 0x03, 0x43, 0x05, 0x82, 0x36
2016-09-28 16:29:31.868 Detail, 
2016-09-28 16:29:31.868 Info, Node003, Response RTT 165 Average Response RTT 165
2016-09-28 16:29:31.868 Info, Node003, Received supported thermostat setpoints
2016-09-28 16:29:31.868 Info, Node003,     Added setpoint: Heating 1
2016-09-28 16:29:31.868 Info, Node003,     Added setpoint: Furnace
2016-09-28 16:29:31.868 Detail, Node003,   Expected reply and command class was received
2016-09-28 16:29:31.868 Detail, Node003,   Message transaction complete

But later I see a message dropped, as the device went to sleep:

> 2016-09-28 16:29:32.821 Info, Node003, Sending (Query) message (Callback ID=0x5e, Expected Reply=0x04) - ThermostatSetpointCmd_Get (Node=3): 0x01, 0x0a, 0x00, 0x13, 0x03, 0x03, 0x43, 0x02, 0x07, 0x25, 0x5e, 0xdb
2016-09-28 16:29:32.828 Detail, Node003,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2016-09-28 16:29:32.828 Detail, Node003,   ZW_SEND_DATA delivered to Z-Wave stack
2016-09-28 16:29:32.940 Detail, Node003,   Received: 0x01, 0x07, 0x00, 0x13, 0x5e, 0x00, 0x00, 0x0c, 0xb9
2016-09-28 16:29:32.941 Detail, Node003,   ZW_SEND_DATA Request with callback ID 0x5e received (expected 0x5e)
2016-09-28 16:29:32.941 Info, Node003, Request RTT 119 Average Request RTT 112
2016-09-28 16:29:32.941 Detail,   Expected callbackId was received
2016-09-28 16:29:42.822 Error, Node003, ERROR: Dropping command, expected response not received after 1 attempt(s)
2016-09-28 16:29:42.823 Detail, Node003, Removing current message
2016-09-28 16:29:42.823 Detail, Node003, Notification: Notification - TimeOut

(this is a bug)

As a work around, try only waking up the device after restarting your application, and all other devices have reached QueryStage_Complete.

Fishwaldo, 2016-09-28:
For my own tracking:

> 2016-09-28 16:29:32.821 Info, Node003, Sending (Query) message (Callback ID=0x5e, Expected Reply=0x04) - ThermostatSetpointCmd_Get (Node=3): 0x01, 0x0a, 0x00, 0x13, 0x03, 0x03, 0x43, 0x02, 0x07, 0x25, 0x5e, 0xdb
2016-09-28 16:29:32.828 Detail, Node003,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2016-09-28 16:29:32.828 Detail, Node003,   ZW_SEND_DATA delivered to Z-Wave stack
2016-09-28 16:29:32.940 Detail, Node003,   Received: 0x01, 0x07, 0x00, 0x13, 0x5e, 0x00, 0x00, 0x0c, 0xb9
2016-09-28 16:29:32.941 Detail, Node003,   ZW_SEND_DATA Request with callback ID 0x5e received (expected 0x5e)
2016-09-28 16:29:32.941 Info, Node003, Request RTT 119 Average Request RTT 112
2016-09-28 16:29:32.941 Detail,   Expected callbackId was received
2016-09-28 16:29:42.822 Error, Node003, ERROR: Dropping command, expected response not received after 1 attempt(s)
2016-09-28 16:29:42.823 Detail, Node003, Removing current message
2016-09-28 16:29:42.823 Detail, Node003, Notification: Notification - TimeOut

In this case, the result is "TRANSMIT_COMPLETE_OK" which says that the controller thought it delivered the message to the device. 

In this case, there is no way to determine the device is asleep. This seems like a Device Bug rather than a OZW bug. Not sure we can fix this without causing other issues. 

As a workaround, please change this line:
https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/Defs.h#L211
to 
#define MAX_TRIES 2

and recompile OZW. This would add some overhead, but should make this device work correctly.

closed 2016-09-28 by Fishwaldo