OpenZWave/open-zwave #990: Information about a bug

Servuc, 2016-09-26:
Hello,

Today, I was searching hard, why our OpenZWave program won't work when node talk about setting commands value on module (Not on gateway). For example, put relay on.

In my case I get :  `ERROR: Not enough space in stream buffer`. It's must be a consequence about my problem, I've missed to put configurations files, in that directory https://github.com/OpenZWave/open-zwave/tree/master/config, next to my executable.

I think a check should be done in OpenZWave, and perhaps display a red message about it.

Have a good coding day.

Fishwaldo, 2016-09-27:
Its very unlikely that missing config files cause this error. (there is no coorelation). I assume you are using a Rasberry board?

nechry, 2016-09-27:
I got also this Error from time to time, last seen this morning. I got this error with a aeonlabs zstick gen5 On a Solidrun Pro hardware.

Servuc, 2016-09-27:
We use a BananaPi M1+.
Yes the corelation is bigger, I have find the problem : without the configuration file, some requests that go server return bad response. 
To put it in a nutshell, our requests to the server block mutex that block serial, the 'serial' pipe was overflow.

So we had add another thread to request server. But I'm surprised that 'serial' pipe life is very short. For example, we work with MySensors too, and we have rarely get that. And we are on the same board.

To conclude and correct this problem, dev (not this project I think) should find in their code which part goes slow and put it in another thread. It's like add another road after little bridge to prevent lot of cars on it. 

(Sorry for my bad english).

I think that issue can be closed. Isn't it ?

Fishwaldo, 2016-09-28:
Can you give a very simple program that demonstrates the blocking you talk about?

Servuc, 2016-09-28:
I don't think because it's very big, but in pseudo code, I've that : 

**BEFORE**

    void processNotificationsZWave(Notification n)
    {
        switch( notification->GetType() ) {
            case Notification::Type_ValueChanged:
            {
                  string * myValue;
                  Manager::Get()->GetValueAsString(notification->GetValueID(), myValue);
                  Array somePostData = [ myValue, sensorId, time];
                  sendToServerWithCurl("http://data.lol/api/values", somePostData);
                  break;
            }
    }

Now I've : 

**NOW** 

    void processNotificationsZWave(Notification n)
    {
        switch( notification->GetType() ) {
            case Notification::Type_ValueChanged:
            {
                  string * myValue;
                  Manager::Get()->GetValueAsString(notification->GetValueID(), myValue);
                  Array somePostData = [ myValue, sensorId, time];
                  mThreadPtr.addToThread(somePostData);
                  break;
            }
    }

And the thread : 

    Thread::addToThread(somePostData) {
        mutex.lock();
        waitingList.add(somePostData);
        mutex.unlock();
    }

    Thread::run() {
        while(true) {
            if( waitingList.size != 0) {

            mutex.lock();
            array myPostData = waitingList.getFirstAndRemoveIt();
            mutex.unlock();

            sendToServerWithCurl("http://data.lol/api/values", myPostData);
        }
    }

closed 2016-11-08 by Fishwaldo