OpenZWave/open-zwave #995: Aeotec Multisensor gen5. Temperatur Corrupted

fossejc, 2016-09-29:
Hi guys,

I am using openzwave with an aeotec multisensor gen5. Unfortunately I receive corrupted data when asking for a node resfresh. 
Of what I was able to understand it seems that the device sends corrupted data (see log below)
This line `0x01, 0x0c, 0x00, 0x04, 0x00, 0x03, 0x06, 0x31, 0x05, 0x01, 0xfb, 0x00, 0xc0, 0xfc` means 

```
2016-09-29 09:14:42.936 Detail, Node003,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x03, 0x06, 0x31, 0x05, 0x01, 0xfb, 0x00, 0xc0, 0xfc
2016-09-29 09:14:42.936 Detail, 
2016-09-29 09:14:42.936 Info, Node003, Response RTT 1832 Average Response RTT 959
2016-09-29 09:14:42.936 Info, Node003, Received SensorMultiLevel report from node 3, instance 1, Temperature: value=0.0049404F
2016-09-29 09:14:42.936 Detail, Node003, Refreshed Value: old value=19.1, new value=0.0049404, type=decimal
2016-09-29 09:14:42.936 Detail, Node003, Changes to this value are not verified
2016-09-29 09:14:42.936 Detail, Node003,   Expected reply and command class was received
2016-09-29 09:14:42.936 Detail, Node003,   Message transaction complete

```

Thanks all for your work.

nechry, 2016-09-29:
is true with all version of this device (gen2, gen5 and the 6in1), all manual refresh give a wrong result. My only workaround is to let the device report by itself, with right configuration on param 201 and 211.

fossejc, 2016-09-29:
The library could throw an error if the scale isn't 00 (Celsius) or 01 (Fahrenheit) in this particular case. Would it be feasible ?

Fishwaldo, 2018-10-26:
This needs to be reported to Aeotec if its still a issue.

closed 2018-10-26 by Fishwaldo