OpenZWave/open-zwave #996: Device support for Heatit thermostat updated

gijs69, 2016-09-29:
I made a number of changes to get the Heatit Thermostat to wrok properly. 

hope it helps others in getting it to work

nechry, 2016-09-29:
Can you share it?

gijs69, 2016-09-29:
It seam I cannot upload it to the bug tracker.

Maybe you know how to do that. 

Anyway, here it is.

nechry, 2016-09-29:
If you update the xml file you need to create a PR and I will validate and merge.

if you paste xml content her we will not see, use the Markdown for code

gijs69, 2016-09-29:
[https://www.dropbox.com/s/f8yvrju4rt5mx8l/heatit021.xml?dl=0](url)

is the link to the file. I never created a PR before :) Maybe it is faster if you pickup this file.

nechry, 2016-09-29:
ok for this time but please in future provide a PR. 
You test every mode? Any complains for this thermostat?

closed 2016-09-29 by nechry

gijs69, 2016-09-29:
As far as I can see everything works just fine right now.

nechry, 2016-09-29:
please also always do a
make xmltest
before submit a config file
see my little change:
https://github.com/OpenZWave/open-zwave/commit/691d1f17438e716b155c659bc6fe4ad6bcac5d10

commit cbab47c1 (2016-09-29): Device support for Heatit thermostat updated

Close #996 thank’s gijs69