OpenZWave/open-zwave #997: UserCode command class does not support deleting user codes

dicarlo2, 2016-10-04:
`SetValue` only supports settings user codes or refreshing user codes[1]. I'd be happy to try adding the functionality, but I'd need some pointers on the "right" way to do it. Ideally we'd be able to set a given index with any of the `UserCodeStatus` enum values.

[1] https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/command_classes/UserCode.cpp#L303

nechry, 2016-10-05:
I do this task directly in my gateway by setting only 0x00 in value. I'm not sure this feature must be present in the library.

Fishwaldo, 2016-10-06:
the UserCode isn't 100% compliant with the Specs. For clearing a usercode, we should be setting the status to available, and sending 4 NULL bytes. So this does need work.....

But

We know also know that usercode is ASCII based, so this should be converted from a RAW to string ValueID. Thats going to break a few applications, so I'd prefer to postpone that to after 1.6 is released. 

(if you want to start refactoring the UserCode CC with the above points, thats fine, please base on the Dev branch).

techgaun, 2017-06-08:
@dicarlo2 have you pursued this at all? I was wondering if I could jump in and try contributing on fixing this.
@nechry could you explain little bit what you mean by setting only 0x00 in value? Do you mean sending the raw packets to the serial device?
@Fishwaldo If @dicarlo2 is not pursuing this, I could look into it. I looked at the Dev branch and didn't see this issue done.

nechry, 2017-06-08:
something like this:
`
2017-06-08 20:44:49.704 Info, Node200, Value::Set - COMMAND_CLASS_USER_CODE - Code 1: - 1 - 1 - 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
2017-06-08 20:44:49.704 Detail, 
2017-06-08 20:44:49.704 Detail, Node200, Queuing (WakeUp) UserCodeCmd_Set (Node=200): 0x01, 0x15, 0x00, 0x13, 0xc8, 0x0e, 0x63, 0x01, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x25, 0x5e, 0x26
`
it's work for benext rfid keypad, the thing is the keypad have only 1,2,3 and 4 digit keys. if your device have the 0 digit ,10 zero will unlock so it's not a good way... 

it's more a workaround for the rfid keypad a reel way to clear a entry code will be more secure

techgaun, 2017-06-08:
@nechry I got your point. I tried it earlier and it does not work for the device I'm trying (like you said) as it has 0 digit. I could work on refactoring the UserCode CC to be more CC compliant

Fishwaldo, 2017-06-08:
There is a PR for sure me updates to the UserCode. I've not really looked into it yet. I did have some comments on it (or a older version) about how the new UserCode API would look. If you want to have a go at it, find those comments and have a go!

I'm traveling this week through Korea and Thailand so I've not had much chance todo anything in OZW. I do have 6 hour layover tonight so I'll try to find my notes and post them again if that helps.

Fishwaldo, 2017-06-08:
Sorry for the typos. On a iPhone in the back of a car at the moment and can't seem to edit my comments.

techgaun, 2017-06-09:
Sure thanks @Fishwaldo any notes you have would be helpful. I will look at the PR you are talking about and will start from there.

techgaun, 2017-06-15:
@Fishwaldo I looked at this little bit and it seems that I would need to either modify the signature of UserCode::SetValue (& that would potentially require some rewrite on how the SetValue of CC are called) or have the single value destructured into two parts: one containing the status of usercode and other part containing the usercode value. Any better ideas here that I'm not considering?

Fishwaldo, 2017-06-19:
@techgaun: You shouldn't alter the Signature of SetValue as its from the base class. Altering it would require changing all of the CommandClasses in OZW. 

Please have a look at the comments in #1059 

I've expanded the number of ValueID's we can have for individual CC's now (upto 65K!) so the way I would do it:
Index 0 - Number of UserCodes Supported (Read only)
Index 1 - Button to Refresh all User Codes
Index 2 - Write Only Int to Clear a UserCode (input the position of the usercode you want to clear, starting at 0)
Index 100-354 - UserCodes (Raw) Index 100 = UserCode 0 etc
Index 500-754 - UserCodes (String) Index 500 = UserCode 0 (but only expose if the UserCode is ascii)
Index 1000-1254 - UserCode Status (List - Read Only) - Index 1000 = UserCode 0 Status

This way:
1) Changing a UserCode - Just write to either the Raw/String UserCode Index you want to change
2) Clearing a UserCode - Input the UserCode Position (start at 0) 

On any update with 1 & 2 above, you would have to Syncronise the Raw and String UserCodes, and in the case of Clearing, update the UserCode Status List.

Finally, you should keep the logic in the Existing UserCode, when we are doing the Initial Discovery, we stop requesting other UserCode entries once we hit the first free UserCode entry. (otherwise startup times would suck - having to query upto 254 UserCodes could take forever) - If a User wants to check all UserCodes, the do the UserCode Refresh Button

(sorry for the late reply - Hope that helps - Feel free to open a PR when you start and I can review as you go)

techgaun, 2017-06-19:
@Fishwaldo perfect, this all makes more sense to me now. I will work on this this week and will do a PR when I push a branch to get your continued feedback. thank you.

techgaun, 2017-06-19:
@Fishwaldo unrelated question (but I didn't know where best it would be to ask): I see that there's no due date for 1.6 milestone (I looked here: https://github.com/OpenZWave/open-zwave/milestone/2) but is there any internal tentative schedule for releasing the 1.6 version? I am just curious to know and also would like to finish this issue as soon as possible so that we can merge it to Dev for 1.6 milestone.

Fishwaldo, 2017-06-19:
Given my real life commitments I can't say. I'd wager a bet you will be ready before I'm ready todo 1.6. 

I have some final testing on the last bit of core changes, and then need to finish the CentralScene and Notification CC. (Plus a bucketload of testing todo)

techgaun, 2017-06-19:
@Fishwaldo ahh cool. I understand and I value your contribution in building and maintaining ozw. If there're something simpler to do and can be offloaded to help you finish with the testing, I am more than happy to involve more and contribute more on that front. Maybe we could also use labels like `up-for-grabs` and level labels liike `level:beginner`, `level:intermediate` and `level:advanced` so that others can try to take particular issues and work on them.

greasysock, 2017-06-20:
Null bytes do not overwrite the user code on my device. I've sent 10 0x00's to my codes, and all my codes continue to work. I am only able to overwrite my access codes with new access codes.

techgaun, 2017-06-26:
Quick update: I broke my zwave stick and the new one arrived on Friday so I am going to try to finish this completely this week. I've only done some primitive works so far.

techgaun, 2017-06-29:
@Fishwaldo I've a quick question regarding https://github.com/OpenZWave/open-zwave/blob/f9c7704a53245baf4bb9e2b66aa5c616dfca25de/cpp/src/command_classes/UserCode.h#L47

Looking at the z-wave spec, I don't see anything ressembling to `UserCode_Unset`. Was there any purpose for having it (like non-standard devices support or anything coming up from basic command class?)

greasysock, 2017-07-26:
I'm messing around with UserCode.cpp, was wondering what the best way to add the feature short-term?
I've added a new raw value UserCodeRemove_index = 253 to the enum. I read through the specifications. It says that to reset a code, you need to set the status to 0x00 and to send 4 0x00 bytes to the access code. Here is my code:

	if ( (ValueID::ValueType_Raw == _value.GetID().GetType()) && (_value.GetID().GetIndex() == UserCodeIndex_Remove) )
	{
		ValueRaw const* value = static_cast<ValueRaw const*>(&_value);
		int target_index;
		target_index = (int)value;

		if( 1 > target_index || target_index > 251 )
		{
		    return false;
		}
		
		m_userCodesStatus[target_index] = UserCode_Available;
		Msg* msg = new Msg( "UserCodeCmd_Set", GetNodeId(), REQUEST, FUNC_ID_ZW_SEND_DATA, true );
		msg->SetInstance( this, _value.GetID().GetInstance() );
		msg->Append( GetNodeId() );
		msg->Append( 8 );
		msg->Append( GetCommandClassId() );
		msg->Append( UserCodeCmd_Set );
		msg->Append( target_index );
		msg->Append( UserCode_Available );
		for( uint8 i = 0; i < 4; i++ )
		{
		    msg->Append( 0x00 );
		}
		msg->Append( GetDriver()->GetTransmitOptions() );
		GetDriver()->SendMsg( msg, Driver::MsgQueue_Send );
		return true;
	}

I'm pretty new to C++, so I was wondering if someone could look over this to make sure it won't do anything unexpected, like ruin my hardware. It has compiled successfully.

greasysock, 2017-07-26:
I might instead change the way the user codes are added by making it so the first byte is the status of the user code and the last ten are the the user code itself.

greasysock, 2017-07-26:
So I got it to work. Might break some applications but it is going to be fine for my use case.
In order to remove a code, send 5 null bytes to the access code value, and in order to add a new access code, you have to send 0x01 followed by the access code.

Here's what I changed:

```cpp
//-----------------------------------------------------------------------------
// <UserCode::SetValue>
// Set a User Code value
//-----------------------------------------------------------------------------
bool UserCode::SetValue
(
	Value const& _value
)
{
	if( (ValueID::ValueType_Raw == _value.GetID().GetType()) && (_value.GetID().GetIndex() < UserCodeIndex_Refresh) )
	{
		ValueRaw const* value = static_cast<ValueRaw const*>(&_value);

		
		uint8* s = value->GetValue();
		uint8 len = value->GetLength();
		uint8 UserCodeStatus_set = s[0];
		UserCodeStatus target_status;
		switch(UserCodeStatus_set){
			case 0x00:
			{
				target_status = UserCode_Available;
				break;
			}
			case 0x01:
			{ 
				target_status = UserCode_Occupied;
				break;
			}
			case 0x02:
			{ 
				target_status = UserCode_Reserved;
				break;
			}
			case 0xfe:
			{ 
				target_status = UserCode_NotAvailable;
				break;
			}
			case 0xff:
			{ 
				target_status = UserCode_Unset;
				break;
			}
			default: 
			{	
				target_status = UserCode_Occupied;
			}
		}
		if( len > UserCodeLength )
		{
			return false;
		}
		m_userCodesStatus[value->GetID().GetIndex()] = target_status;
		Msg* msg = new Msg( "UserCodeCmd_Set", GetNodeId(), REQUEST, FUNC_ID_ZW_SEND_DATA, true );
		msg->SetInstance( this, _value.GetID().GetInstance() );
		msg->Append( GetNodeId() );
		msg->Append( 3 + len );
		msg->Append( GetCommandClassId() );
		msg->Append( UserCodeCmd_Set );
		msg->Append( value->GetID().GetIndex() );
		msg->Append( target_status );
		for( uint8 i = 1; i < len; i++ )
		{
			msg->Append( s[i] );
		}
		msg->Append( GetDriver()->GetTransmitOptions() );
		GetDriver()->SendMsg( msg, Driver::MsgQueue_Send );
		return true;
	}
	if ( (ValueID::ValueType_Button == _value.GetID().GetType()) && (_value.GetID().GetIndex() == UserCodeIndex_Refresh) )
	{
		m_refreshUserCodes = true;
		m_currentCode = 1;
		m_queryAll = true;
		RequestValue( 0, m_currentCode, _value.GetID().GetInstance(), Driver::MsgQueue_Query );
		return true;
	}
	return false;
}
```

Here's the modified UserCode.cpp file:
[UserCode.cpp](https://github.com/OpenZWave/open-zwave/files/1177969/UserCode.txt)

kevinduong, 2018-03-27:
@greasysock I know this is old, but could you submit a PR for this?

kovert, 2018-05-05:
For backwards compatibility reasons, would it be reasonable for the the first code to only be valid as a status if its set to one of the known values, otherwise assume its just a usercode?

I pulled together a patch based on @greasysock 's patch, but I'm an end user of something using something using the library, so I'm still figuring out how to pull together a test rig.

I'm happy to run with a PR for this if it makes sense.

-Todd

Modified code:
[UserCode.txt](https://github.com/OpenZWave/open-zwave/files/1977060/UserCode.txt)

osenft, 2018-08-09:
I had another look at the code. One option that will not break existing implementations yet allow users (like Home Assistant) to clear a user code would be as follows:

In UserCode::SetValue, check if all bytes in the value are 0.
- If all bytes are 0, then use UserCode_Available.
- It at least one byte is not 0, use UserCode_Occupied as before.

According to the [ZWave public spec](http://zwavepublic.com/sites/default/files/command_class_specs_2017A/SDS13781-5%20Z-Wave%20Application%20Command%20Class%20Specification.pdf), section 4.106.2, for the User Code "each byte in this field MUST be a digit encoded with ASCII representation (from 0x30 to 0x39)."

This ensures backwards compatibility with existing users of the library, yet enables the use case to clear a user code.

The obvious downside is that the caller would need to know to send exactly four 0s to be compliant with the ZWave spec. But I guess that's still better than not being able to clear a user code at all or to require every user of the library to change their code.

If we think think this is the right way to go, I'd be happy to prepare a patch.

kevinduong, 2018-08-09:
@osenft I think sending four 0's would be the best way to fix this. A simple google search will output that this is how it works for some locks currently, so end users shouldn't have a hard time figuring that out.

kmwoley, 2018-08-09:
Some (most?) locks have a user-configurable code length (i.e. my Schlage
Connect is configured for 5 digits). Would that matter in the
implementation? Maybe rather than requiring that the user code sent via the
API is four digits, just specify that no matter how many digits are sent,
if all of them are '0' clear the lock code.

Another option might be for the API to accept null characters, or a
zero-length string as the user code as the indicator to clear the lock code
on the physical device.

Excited to see progress on this and happy to test/code review.

osenft, 2018-08-09:
The spec is explicit about sending exactly four 0s alongside status 0 (UserCode_Available). 

Merging the various idea together:

In UserCode::SetValue, check if all bytes in the value are 0.
- If all bytes are 0, then use UserCode_Available and send exactly four 0s, ignoring the length that the user specified.
- If at least one byte is not 0, use UserCode_Occupied as before and use the user specified data.

I'll let this on my Schlage lock that's configured to >4 digits for the user code.

Would you guys accept a patch that implements that? I don't want to implement it if there isn't consensus.

zmcandee, 2018-08-09:
Could we overload the SetValue with an option for status?  That would allow existing implementations and also better implement the spec for future implementations.  As it stands, we can't set Available or Reserved both of which could be very useful.

Fishwaldo, 2018-08-09:
My thoughts on this. Seems different locks interpreted the specs differently. I’d like to have OZW expose a interface that doesn’t require the user to know the specifics of his lock. 

So maybe just expose a new decimal ValueID (Write only). Just sent the position you want to clear and the CC does it’s magic. 

For locks that don’t follow the specs 100% we implement a confit option to specify how to clear a user code for that lock. 

Then users and applications don’t need to know if it’s ASCII 0’s or NULL’s or whatever other combination etc. 

I’d accept a PR (to Dev branch) for something like that.

osenft, 2018-08-09:
@Fishwaldo, do you have a suggestion on how to map that to a new ValueID? Looking at https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/value_classes/ValueID.h, I can't see a clean match for your suggestion.

Logically, I still think that the user application writing either an empty value or a value just consisting of NULLs would be the right use of the existing interface. 

Even with that we should still be able to allow for config option for broken devices? Having said that, the spec is really 100% unambiguous. So I'd really like to see a device that is broken. From what I can tell from the existing implementation, we have no way of knowing at this time and can as well assume that devices implement the spec correctly.

osenft, 2018-08-09:
@zmcandee, I'm not sure how overloading SetValue would work, given the exposed interface in https://github.com/OpenZWave/python-openzwave/blob/master/src-api/openzwave/value.py#L292. I don't think we should change that abstraction.

osenft, 2018-08-10:
https://github.com/OpenZWave/open-zwave/pull/1576

Fishwaldo, 2018-08-10:
@osenft I think you miss understood. I mean in the UserCode CC, create a new ValueID (of type decimal) and then an application writes the UserCode Slot to that ValueID. In UserCode::SetValue we do the necessary for different devices that need different ways to clear the code. 

There are a few out there. I think the first few versions of the Spec were not as clear, and its been tidied up. I know for sure there is a Keypad out there (I have to dig it out) that got it wrong as far as ASCII versus NULL.

osenft, 2018-08-11:
@Fishwaldo, ok, here's what I understand:

1. I'll create an additional single static write-only value of "some" type.
2. The value will be assigned a new unique reserved index (e.g. 253)
3. This could be done in UserCode::CreateVars - similar to what's already done there.
4. The new value can then be written to from python-openzwave by filtering for that specific index.

A few questions:

1. Why should the vaue type be "decimal"? Would "byte" not make more sense?
2. How does this implementation make it easier to override the behavior for specific devices compared to what I proposed in PR #1576 ?
3. Users of the new implementation will need to also know the special index. This seems a little awkward.

What am missing?

osenft, 2018-08-15:
Anyone? It would be nice to get this fixed. I'm happy to write more code, but I want to make sure PR #1576 is not sufficient.

Fishwaldo, 2018-09-07:
@osenft your understanding is correct. to answer your questions:
1. Sorry, correct. Byte would be fine.
2. #1576 makes the assumption that the average joe knows that 0000 or '' or NULL would clear the the code. With the above approach (as most applications iterate over available ValueID's for a command class) they would expose a new Input Box or whatnot that is called "Reset Code" or what not and wouldn't be required to know what to enter to clear a code. (other than the index value of what they want to clear). 

3. Developers would need to know, yes, but the average user doesn't even know what a ValueID or Index is :)

Fishwaldo, 2018-12-04:
Ok. This is mostly done. I need someone with a device that doesn't use ASCII 0 to reset the device to add the last task... so I'll close this for now.

closed 2018-12-04 by Fishwaldo

kmwoley, 2018-12-13:
@Fishwaldo - do you know of a device that uses NULL (i.e. doesn't use ASCII 0) to reset the device? I'd be willing to contribute money/time to help test.

blakeblackshear, 2018-12-13:
Seems like this issue came about because of the Schlage Connect lock right? I have one, and I am sure there are hundreds of other willing testers that use homeassistant. I need to get up to speed on compiling and testing, but I could do it.

kmwoley, 2018-12-14:
Likewise. I have a Schlage Connect. Happy to jump through hoops to test.

ldfsilva, 2018-12-27:
I've re-tested it with home-assistant 0.84.6 and python-openzwave 0.4.16 with no luck. It still shows that the given `code_slot` is occupied and the old code doesn't get cleared as expected.

The lock I'm using is a Schlage BE469 Touchscreen Deadbolt.

Hey @osenft were you able to get it to work? 

As follows the output from `OZW_Log.txt` while running `lock.clear_usercode` in home-assistant for json `{ "node_id": 3, "code_slot": 3}`

```bash
2018-12-26 22:25:50.189 Info, Node003, Value::Set - COMMAND_CLASS_USER_CODE - Code 3: - 3 - 1 - 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
2018-12-26 22:25:50.190 Detail, Node003, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_USER_CODE
2018-12-26 22:25:50.190 Detail, Node003, Queuing (Send) UserCodeCmd_Set (Node=3): 0x01, 0x15, 0x00, 0x13, 0x03, 0x0e, 0x63, 0x01, 0x03, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x25, 0x26, 0x97
2018-12-26 22:25:50.191 Detail, Node003, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_USER_CODE
2018-12-26 22:25:50.191 Detail, Node003, Queuing (Send) UserCodeCmd_Get (Node=3): 0x01, 0x0a, 0x00, 0x13, 0x03, 0x03, 0x63, 0x02, 0x03, 0x25, 0x27, 0x86
2018-12-26 22:25:50.191 Detail, 
2018-12-26 22:25:50.191 Info, Node003, Processing (Send) Nonce Request message (Callback ID=0x26, Expected Reply=0x13)
2018-12-26 22:25:50.192 Info, Node003, Sending (Send) message (Callback ID=0x26, Expected Reply=0x13) - Nonce_Get(UserCodeCmd_Set) - 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x98, 0x40, 0x05, 0x02:
2018-12-26 22:25:50.199 Detail, Node003,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-12-26 22:25:50.199 Detail, Node003,   ZW_SEND_DATA delivered to Z-Wave stack
2018-12-26 22:25:51.447 Detail, Node003,   Received: 0x01, 0x07, 0x00, 0x13, 0x02, 0x00, 0x00, 0x7d, 0x94
2018-12-26 22:25:51.447 Detail, Node003,   ZW_SEND_DATA Request with callback ID 0x02 received (expected 0x02)
2018-12-26 22:25:51.447 Info, Node003, Request RTT 1255 Average Request RTT 1323
2018-12-26 22:25:51.475 Detail, Node003,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x03, 0x0a, 0x98, 0x80, 0xb0, 0xca, 0x65, 0x68, 0xef, 0x80, 0x8a, 0x48, 0x20
2018-12-26 22:25:51.475 Info, Node003, Received SecurityCmd_NonceReport from node 3
2018-12-26 22:25:51.475 Info, Node003, Sending (Send) message (Callback ID=0x28, Expected Reply=0x13) - UserCodeCmd_Set (Node=3): 0x01, 0x15, 0x00, 0x13, 0x03, 0x0e, 0x63, 0x01, 0x03, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x25, 0x28, 0x99
2018-12-26 22:25:51.488 Detail, Node003,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-12-26 22:25:51.488 Detail, Node003,   ZW_SEND_DATA delivered to Z-Wave stack
2018-12-26 22:25:51.511 Detail, Node003,   Received: 0x01, 0x07, 0x00, 0x13, 0x28, 0x00, 0x00, 0x03, 0xc0
2018-12-26 22:25:51.512 Detail, Node003,   ZW_SEND_DATA Request with callback ID 0x28 received (expected 0x28)
2018-12-26 22:25:51.512 Info, Node003, Request RTT 1319 Average Request RTT 1321
2018-12-26 22:25:51.512 Detail,   Expected callbackId was received
2018-12-26 22:25:51.512 Detail,   Expected reply was received
2018-12-26 22:25:51.512 Detail,   Message transaction complete
2018-12-26 22:25:51.512 Detail, 
2018-12-26 22:25:51.513 Detail, Node003, Removing current message
2018-12-26 22:25:51.513 Detail, 
2018-12-26 22:25:51.513 Info, Node003, Processing (Send) Nonce Request message (Callback ID=0x27, Expected Reply=0x04)
2018-12-26 22:25:51.513 Info, Node003, Sending (Send) message (Callback ID=0x27, Expected Reply=0x04) - Nonce_Get(UserCodeCmd_Get) - 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x98, 0x40, 0x05, 0x02:
2018-12-26 22:25:51.520 Detail, Node003,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-12-26 22:25:51.521 Detail, Node003,   ZW_SEND_DATA delivered to Z-Wave stack
2018-12-26 22:25:51.537 Detail, Node003,   Received: 0x01, 0x07, 0x00, 0x13, 0x02, 0x00, 0x00, 0x03, 0xea
2018-12-26 22:25:51.537 Detail, Node003,   ZW_SEND_DATA Request with callback ID 0x02 received (expected 0x02)
2018-12-26 22:25:51.538 Info, Node003, Request RTT 24 Average Request RTT 672
2018-12-26 22:25:51.674 Detail, Node003,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x03, 0x02, 0x98, 0x40, 0x2a
2018-12-26 22:25:51.674 Info, Node003, Received SecurityCmd_NonceGet from node 3
2018-12-26 22:25:51.674 Info, NONCES: 0x18, 0x22, 0x85, 0x14, 0x11, 0x34, 0x76, 0xd1
2018-12-26 22:25:51.674 Info, NONCES: 0xff, 0x0d, 0xde, 0x12, 0x01, 0xec, 0x98, 0x2e
2018-12-26 22:25:51.674 Info, NONCES: 0xb0, 0x19, 0x87, 0xc1, 0x4d, 0xfe, 0x93, 0xe0
2018-12-26 22:25:51.674 Info, NONCES: 0xfa, 0xbe, 0xe7, 0xfb, 0xaa, 0x7f, 0x29, 0xd4
2018-12-26 22:25:51.675 Info, NONCES: 0x5a, 0xaf, 0xf4, 0x96, 0xa8, 0xdb, 0x70, 0xec
2018-12-26 22:25:51.675 Info, NONCES: 0x26, 0xe1, 0xa4, 0x6e, 0x9e, 0x47, 0xc9, 0x4e
2018-12-26 22:25:51.675 Info, NONCES: 0x1b, 0x20, 0x7e, 0xc2, 0xfc, 0xef, 0xaf, 0x62
2018-12-26 22:25:51.675 Info, NONCES: 0x20, 0xcb, 0x2a, 0xbe, 0x13, 0xf3, 0x0d, 0x85
2018-12-26 22:25:51.675 Info, Node003, Sending (Send) message (Callback ID=0x01, Expected Reply=0x04) - Nonce_Report - 0x01, 0x11, 0x00, 0x13, 0x03, 0x0a, 0x98, 0x80, 0xfa, 0xbe, 0xe7, 0xfb, 0xaa, 0x7f, 0x29, 0xd4, 0x05, 0x01, 0x98:
2018-12-26 22:25:51.684 Detail, Node003,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-12-26 22:25:51.684 Detail, Node003,   ZW_SEND_DATA delivered to Z-Wave stack
2018-12-26 22:25:51.704 Detail, Node003,   Received: 0x01, 0x07, 0x00, 0x13, 0x01, 0x00, 0x00, 0x02, 0xe8
[... cut ...]
2018-12-26 22:25:53.272 Detail, Node003, Changes to this value are not verified
2018-12-26 22:25:53.272 Info, Node003, Received User Code Report from node 3 for User Code 3 (Occupied)
2018-12-26 22:25:53.272 Detail, Node003,   Expected reply and command class was received
2018-12-26 22:25:53.272 Detail, Node003,   Message transaction complete
2018-12-26 22:25:53.272 Detail, 
2018-12-26 22:25:53.272 Detail, Node003, Removing current message
2018-12-26 22:25:53.272 Detail, Node003, Notification: ValueChanged
```

Fishwaldo, 2018-12-27:
The changes are in the OZW Dev branch. Did you test that branch?

ldfsilva, 2018-12-28:
I've tested after re-installing **python_openzwave** through pip3, I've tried different flavors (git, ozwdev and dev), however they all seem to pint to same download file. What would be the way for pulling it from Dev branch?

Here's how I've tried adjusting the flavor to the ones mentioned above:

`pip install python_openzwave --no-cache-dir --no-deps --install-option="--flavor=ozwdev"`

```bash
(homeassistant) homeassistant@raspberrypi:/srv/homeassistant $ pip3 install python_openzwave --no-cache-dir --no-deps --install-option="--flavor=ozwdev"
/srv/homeassistant/lib/python3.5/site-packages/pip/_internal/commands/install.py:211: UserWarning: Disabling all use of wheels due to the use of --build-options / --global-options / --install-options.
  cmdoptions.check_install_build_global(options)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting python_openzwave
  Downloading https://files.pythonhosted.org/packages/bc/7e/84bd53a0585892ed992e2f2df4fe6edbb16e819938632170a0ad5f4843b0/python_openzwave-0.4.16.zip (139kB)
    100% |################################| 143kB 2.3MB/s 
```

kevinduong, 2018-12-28:
Pull the dev branch down and compile it.

Fishwaldo, 2018-12-28:
You would need to check in the Python-OpenZWave repo how to use the Dev Branch.

hiryou, 2019-02-02:
Hi guys. was this issue ever tested and resolved using the dev branch?

I have the same lock with @ldfsilva BE469 CEN. I installed hassbian. This is the step I used to change python_openzwave:
```
$ hastop
$ sudo -u homeassistant -H -s
$ source /srv/homeassistant/bin/activate
$ 
$ pip uninstall -y python_openzwave
$ pip install python_openzwave --no-cache-dir --no-deps --install-option="--flavor=git|ozwdev|ozwdev_shared" # See my elaboration below
# In HA configuration.xml, I need to update zwave config_path to: /srv/homeassistant/lib/python3.5/site-packages/python_openzwave/ozw_config
$ exit
$ sudo reboot
```

For the step of installing python_openzwave
* If I use ozwdev or ozwdev_shared, HA failed to start after reboot. When I browse to hassbian.local, all existing nodes show "entity not available", then HA crashed soon after. In home-assistant.log, I found:
```
2019-02-02 19:29:09 ERROR (Dummy-22) [libopenzwave] notif_callback exception
Traceback (most recent call last):
  File "src-lib/libopenzwave/libopenzwave.pyx", line 512, in libopenzwave.notif_callback
IndexError: list index out of range
```
* If I use the git flavor then HA works fine as whatever default version it was installed with. However clearing user code doesn't work.

Did I do something wrong here?

hiryou, 2019-02-03:
@Fishwaldo I compiled and installed python_openzwave from source (tag v0.4.18), using the latest Dev branch of openzwave. The issue is not fixed yet, at least for Schlage lock BE469CEN. Can we reopen this issue?
When calling clear_usercode, from OZW log it appears the lib does send 10 zeroes to the lock (from my fuzzy understanding this your your approach?). However the code isn't cleared.

Using hassbian web UI Developer Tools to call
```
lock.clear_usercode
{
  "node_id": "3",
  "code_slot": "3"
}
```
Corresponding OZW log:
```
2019-02-03 18:06:55.739 Info, Node003, Value::Set - COMMAND_CLASS_USER_CODE - Code 3: - 3 - 1 - 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
2019-02-03 18:06:55.739 Detail, Node003, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_USER_CODE
2019-02-03 18:06:55.739 Detail, Node003, Queuing (Send) UserCodeCmd_Set (Node=3): 0x01, 0x15, 0x00, 0x13, 0x03, 0x0e, 0x63, 0x01, 0x03, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x25, 0x43, 0xf2
2019-02-03 18:06:55.739 Detail, Node003, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_USER_CODE
2019-02-03 18:06:55.739 Detail, Node003, Queuing (Send) UserCodeCmd_Get (Node=3): 0x01, 0x0a, 0x00, 0x13, 0x03, 0x03, 0x63, 0x02, 0x03, 0x25, 0x44, 0xe5
2019-02-03 18:06:55.740 Detail, 
2019-02-03 18:06:55.740 Info, Node003, Processing (Send) Nonce Request message (Callback ID=0x43, Expected Reply=0x13)
2019-02-03 18:06:55.740 Info, Node003, Sending (Send) message (Callback ID=0x43, Expected Reply=0x13) - Nonce_Get(UserCodeCmd_Set) - 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x98, 0x40, 0x05, 0x02:
2019-02-03 18:06:55.747 Detail, Node003,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-02-03 18:06:55.747 Detail, Node003,   ZW_SEND_DATA delivered to Z-Wave stack
2019-02-03 18:06:56.995 Detail, Node003,   Received: 0x01, 0x07, 0x00, 0x13, 0x02, 0x00, 0x00, 0x7d, 0x94
2019-02-03 18:06:56.995 Detail, Node003,   ZW_SEND_DATA Request with callback ID 0x02 received (expected 0x02)
2019-02-03 18:06:56.995 Info, Node003, Request RTT 1255 Average Request RTT 1366
2019-02-03 18:06:57.023 Detail, Node003,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x03, 0x0a, 0x98, 0x80, 0x74, 0x99, 0xcd, 0x50, 0x27, 0x6b, 0x0c, 0xb0, 0x7a
2019-02-03 18:06:57.023 Info, Node003, Received SecurityCmd_NonceReport from node 3
2019-02-03 18:06:57.023 Info, Node003, Sending (Send) message (Callback ID=0x45, Expected Reply=0x13) - UserCodeCmd_Set (Node=3): 0x01, 0x15, 0x00, 0x13, 0x03, 0x0e, 0x63, 0x01, 0x03, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x25, 0x45, 0xf4
2019-02-03 18:06:57.036 Detail, Node003,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-02-03 18:06:57.036 Detail, Node003,   ZW_SEND_DATA delivered to Z-Wave stack
2019-02-03 18:06:57.059 Detail, Node003,   Received: 0x01, 0x07, 0x00, 0x13, 0x45, 0x00, 0x00, 0x03, 0xad
2019-02-03 18:06:57.060 Detail, Node003,   ZW_SEND_DATA Request with callback ID 0x45 received (expected 0x45)
2019-02-03 18:06:57.060 Info, Node003, Request RTT 1320 Average Request RTT 1343
2019-02-03 18:06:57.060 Detail,   Expected callbackId was received
2019-02-03 18:06:57.060 Detail,   Expected reply was received
2019-02-03 18:06:57.060 Detail,   Message transaction complete
2019-02-03 18:06:57.060 Detail, 
2019-02-03 18:06:57.060 Detail, Node003, Removing current message
2019-02-03 18:06:57.060 Detail, 
2019-02-03 18:06:57.060 Info, Node003, Processing (Send) Nonce Request message (Callback ID=0x44, Expected Reply=0x04)
2019-02-03 18:06:57.061 Info, Node003, Sending (Send) message (Callback ID=0x44, Expected Reply=0x04) - Nonce_Get(UserCodeCmd_Get) - 0x01, 0x09, 0x00, 0x13, 0x03, 0x02, 0x98, 0x40, 0x05, 0x02:
2019-02-03 18:06:57.068 Detail, Node003,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-02-03 18:06:57.069 Detail, Node003,   ZW_SEND_DATA delivered to Z-Wave stack
2019-02-03 18:06:57.085 Detail, Node003,   Received: 0x01, 0x07, 0x00, 0x13, 0x02, 0x00, 0x00, 0x02, 0xeb
2019-02-03 18:06:57.085 Detail, Node003,   ZW_SEND_DATA Request with callback ID 0x02 received (expected 0x02)
2019-02-03 18:06:57.085 Info, Node003, Request RTT 24 Average Request RTT 683
2019-02-03 18:06:57.223 Detail, Node003,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x03, 0x02, 0x98, 0x40, 0x2a
2019-02-03 18:06:57.223 Info, Node003, Received SecurityCmd_NonceGet from node 3
2019-02-03 18:06:57.223 Info, NONCES: 0xc4, 0xb2, 0x1f, 0xaf, 0x62, 0xc6, 0xf1, 0xea
2019-02-03 18:06:57.223 Info, NONCES: 0xd6, 0xb9, 0x2d, 0x38, 0x7f, 0x1f, 0x23, 0x5c
2019-02-03 18:06:57.223 Info, NONCES: 0xa8, 0xc3, 0x7c, 0x28, 0xe2, 0xa0, 0x84, 0x35
2019-02-03 18:06:57.223 Info, NONCES: 0xfc, 0x56, 0xe5, 0x22, 0x69, 0x01, 0xc8, 0xc6
2019-02-03 18:06:57.223 Info, NONCES: 0x66, 0xd0, 0xaf, 0xe9, 0x7b, 0x37, 0xf3, 0xeb
2019-02-03 18:06:57.224 Info, NONCES: 0x72, 0x39, 0x30, 0x46, 0x8e, 0x6a, 0x2b, 0xe8
2019-02-03 18:06:57.224 Info, NONCES: 0xc0, 0x5e, 0x4b, 0x3b, 0x95, 0x3e, 0x27, 0xbb
2019-02-03 18:06:57.224 Info, NONCES: 0x2d, 0x3d, 0xcc, 0xbb, 0xa8, 0xf7, 0xa3, 0xc2
2019-02-03 18:06:57.224 Info, Node003, Sending (Send) message (Callback ID=0x01, Expected Reply=0x04) - Nonce_Report - 0x01, 0x11, 0x00, 0x13, 0x03, 0x0a, 0x98, 0x80, 0xfc, 0x56, 0xe5, 0x22, 0x69, 0x01, 0xc8, 0xc6, 0x05, 0x01, 0xe3:
2019-02-03 18:06:57.233 Detail, Node003,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-02-03 18:06:57.233 Detail, Node003,   ZW_SEND_DATA delivered to Z-Wave stack
[... cut ...]
2019-02-03 18:06:57.542 Detail, Node003, Changes to this value are not verified
2019-02-03 18:06:57.542 Info, Node003, Received User Code Report from node 3 for User Code 3 (Occupied)
2019-02-03 18:06:57.542 Detail, Node003,   Expected reply and command class was received
2019-02-03 18:06:57.542 Detail, Node003,   Message transaction complete
2019-02-03 18:06:57.542 Detail, 
2019-02-03 18:06:57.542 Detail, Node003, Removing current message
2019-02-03 18:06:57.542 Detail, Node003, Notification: ValueChanged
```

ldfsilva, 2019-08-12:
For context in case someone else is still having trouble with deleting codes, I was able to get it to work following @srpape advise ([here](https://github.com/home-assistant/home-assistant/issues/22464#issuecomment-49461063)).

> source /srv/homeassistant/bin/activate
> pip3 install --upgrade git+https://github.com/srpape/python-openzwave.git

And here is my current config, and my lock is Schlage BE469 Touchscreen Deadbolt.

```
(homeassistant) homeassistant@raspberrypi:/srv/homeassistant $ python -V
Python 3.6.9
(homeassistant) homeassistant@raspberrypi:/srv/homeassistant $ hass --version
0.97.1
(homeassistant) homeassistant@raspberrypi:/srv/homeassistant $ pip freeze|grep zw
homeassistant-pyozw==0.1.4
python-openzwave==0.4.19
(homeassistant) homeassistant@raspberrypi:/srv/homeassistant $ 
```