OpenZWave/open-zwave #1003: Fibaro double switch 2 (FGS-223) association s2

kwibox, 2016-10-09:
Hey there!

I recently bought my first z-wave equipment; a z-wave.me USB-stick with a Fibaro double switch 2 (fgs-223). Pairing and configuration of openzwave and the end-user app (home assistant) was quite easy.

However, the state of the 2nd switch isn't updated correctly: e.g. when the light is switched on (using the wall switch or using the application), the state in the application isn't updated until the polling round. The light itself is switched on instantaneous (as it should)  and the problem does also exists the other way around (from "on" to "off"). 

The problem exists in the end-user app as well in  openzwavecp and is reproducible between configuration resets.

I started reading about the way z-wave works and I suspect there is a bug in the implementation of the association for providing the state of the 2nd switch.  

Attached below is the ozw-log where I toggle both switches a couple of times. I wasn't able to run it through the analyser.

[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/518460/OZW_Log.txt)

Fishwaldo, 2016-11-01:
Check the associations of the device. Associations are what report updates back to the controller. If the controller node isn't in the right association (or not at all present) then no updates are sent.

Fishwaldo, 2016-11-08:
ping?

davispw, 2016-11-09:
Might this be related to Home Assistant issues [#4159](https://github.com/home-assistant/home-assistant/issues/4159) and [#4224](https://github.com/home-assistant/home-assistant/pull/4224)?  Their fix is to ignore the 2nd instance which seems less than ideal.  Also seeing with an Aeon Smart Dimmer 6, which has a "night light" instance in addition to the regular plug.  Logs show it calling Get to "confirm a reported value" even though verify is false.

steve2507, 2016-11-14:
Just to confirm that this is an issue; I myself have recently acquired a FGS-223. When I run MinOZW (on a Razberry PI 3), I only see new status messages for one of the instances on the FGS-223. This means that switching one of the two buttons connected to the FGS-223 does yield results in MinOZW, but the other one does not.

Sending a command to either of the instances does work properly (to switch them on/ or off). It manages to switch both instances correctly.

4SFED, 2016-11-29:
Just weighing in on this issue. I am experiencing the exact same problem. RPi3 running HASS built by the all in one installer and the FGS-223. Happy to provide any logs or assistance this if required.

antares2001, 2016-12-01:
Same problem here

tobiasgraf, 2016-12-13:
Hi,

I also had this problem with this device but I got it working.

The thing is, the FGS 223 (and also 213) needs to be associated with the multi channel association.
So what i did (in my Application when adding these devices):
- first remove the previous association in Group 1 with 
`Manager::Get()->RemoveAssociation(homeID,nodeID, 1, 1, 0);` (group, target [openZwave], instance)
- then add it as multiChannelAssociation:
`Manager::Get()->AddAssociation(homeID,nodeID, 1, 1, 1);`

Now i get the button presses on and also the energy consumption of the S2 Switch and Q2 output.

Is there a way to tell openZwave to use multichannel instead of normal association in the xml ?

nechry, 2016-12-13:
Strange, I already try this with my device fgs223 whiteout success. I will try again, and do the patch in config file if it's work as expected.
``` <CommandClass id="142" ForceInstances="true" />```

tobiasgraf, 2016-12-13:
I just also tried with `<CommandClass id="142" ForceInstances="true" />` (removed my workaround first) and it works for me :)

nechry, 2016-12-13:
@tobiasgraf Your device is include in secure mode ?

For me it's seems work only if device is include secure. In no secure, I still have to perform the workaround. I add the Behaviour to the config files.

closed 2016-12-13 by nechry

tobiasgraf, 2016-12-13:
Yes, my device is included in secure mode.

nechry, 2016-12-13:
ok so the ForceInstances attribute seems not working for a device not included in secure mode.

drabbert, 2016-12-16:
> <CommandClass id="142" ForceInstances="true" />
I'm quite new on this, in which file do I have to make this change?
Thanks

tobiasgraf, 2016-12-16:
nechry already made the changes 3 days ago. See this commit https://github.com/OpenZWave/open-zwave/commit/d39a11ed43856437f719088eb4850d74535ee36e

drabbert, 2016-12-16:
Thanks Tobias, should these two changes do the trick? I'm not at home, but so far as I can see it's still not working. Or do I have the re-associate the switch?

tobiasgraf, 2016-12-16:
Yes, just re-add the device.

drabbert, 2016-12-17:
I've re-added the switch but I keep having problems. The sate of the switch is not always seen by HA. Sometimes I have to press several times to switch on or off. And now I'm not able to change the state of one channel at all. The switch keeps returning to its'old state after pressing.
Does anybody have the same problem?

napsio, 2017-01-04:
I have the same problem.

jinnerbichler, 2017-07-23:
I am experiencing the same problem. Has anybody found a solution?

nechry, 2017-07-23:
The controller must be set with instance 1 in lifeline.

rrozema, 2017-12-01:
What is the notational format to use when I want to specify a specific instance n of a specific node m in an association? is that m.n or n.m?

nechry, 2017-12-01:
n.i for node.instance 
The target node first and the specific instance after