OpenZWave/open-zwave #1011: No overflow checks in HandleMsg() in ManufacturerSpecific.cpp

jonathanfrenz, 2016-10-15:
The HandleMsg() function in ManufacturerSpecific.cpp doesn't have any overflow checks. If a malformed message gets passed to it, the application crashes (at least on Windows... undetermined behavior elsewhere) because the function blindly accesses _data without checking _length. 

Presumably, this works normally, but it is very terrible behavior when a bad message gets passed through (as my setup seems to be doing).

[OZW_Log2.txt](https://github.com/OpenZWave/open-zwave/files/531358/OZW_Log2.txt)

Fishwaldo, 2016-10-17:
I'm interested to know why you are getting a corrupt message... might be a sign of some other problem, but you log doesn't contain the "offending" message. are you able to provide?

jonathanfrenz, 2016-10-17:
I'm trying to use a BarrierOperator with the dev branch with no success thus-far.

I think this is the message coming in. I've followed it down the callstack into ManufacturerSpecific.cpp where it's crashing.

`2016-10-15 13:50:29.373 Info, Node005, Sending (Send) message (Callback ID=0x0d, Expected Reply=0x04) - ManufacturerSpecificCmd_Get (Node=5): 0x01, 0x09, 0x00, 0x13, 0x05, 0x02, 0x72, 0x04, 0x25, 0x0d, 0xbc
2016-10-15 13:50:29.379 Detail, Node005,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2016-10-15 13:50:29.381 Detail, Node005,   ZW_SEND_DATA delivered to Z-Wave stack
2016-10-15 13:50:29.535 Detail, Node005,   Received: 0x01, 0x05, 0x00, 0x13, 0x0d, 0x01, 0xe5
2016-10-15 13:50:29.537 Detail, Node005,   ZW_SEND_DATA Request with callback ID 0x0d received (expected 0x0d)
2016-10-15 13:50:29.538 Info, Node005, WARNING: ZW_SEND_DATA failed. No ACK received - device may be asleep.
2016-10-15 13:50:29.540 Warning, Node005, WARNING: Device is not a sleeping node.
2016-10-15 13:50:29.541 Detail, Node001,   Expected callbackId was received
2016-10-15 13:50:29.746 Detail, Node005,   Received: 0x01, 0x0e, 0x00, 0x04, 0x00, 0x05, 0x08, 0x72, 0x05, 0x01, 0x4f, 0x47, 0x44, 0x35, 0x30, 0xc7`

Fishwaldo, 2016-10-18:
There is nothing wrong with that packet. 
Manufacturer ID = 014f (Linear)
Product Type = 4744
Product ID = 3530

So the crash is not related to a malformed packet. Can you get a full backtrace to identify the actual line its crashing in?

Fishwaldo, 2016-11-01:
ping....

jonathanfrenz, 2016-11-01:
I'll try to get back to you later this week when I'm back in town.

jonathanfrenz, 2016-11-03:
I finally got the Debug mode working (there were some config path issues), and, sure enough, the issue is not the unsafe array reads. It's dying because m_Product in the node is null. It thus dies on getConfigPath():
`
		uint16 productType = (((uint16)_data[3])<<8) | (uint16)_data[4];
		uint16 productId = (((uint16)_data[5])<<8) | (uint16)_data[6];

		if( Node* node = GetNodeUnsafe() )
		{
			// Attempt to create the config parameters
			SetProductDetails(manufacturerId, productType, productId);

			**if( node->getConfigPath().size() > 0 )**
			{
				LoadConfigXML( );
			}
`
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/569443/OZW_Log.txt)

Fishwaldo, 2016-11-04:
ok. that area of work is a bit in progress with the refactoring. Should be fixed now.

closed 2016-11-04 by Fishwaldo