OpenZWave/open-zwave #1015: Security handling with sequenced frames

tobiasgraf, 2016-10-25:
The ZWave.Me Wall Controller Wall C-S sometimes sends its frames sequenced when holding down a button.
In ZWSecurity.cpp:350 there is a todo
```
		/* XXX TODO: Check the Sequence Header Frame to see if this is the first part of a
		 * message, or 2nd part, or a entire message.
		 *
		 * I havn't actually seen a Z-Wave Message thats too big to fit in a encrypted message
		 * yet, so we will look at this if such a message actually exists!
		 */
```
describing exactly this behavior.
See the attached file for some logs.

[sequenced_security_frame.txt](https://github.com/OpenZWave/open-zwave/files/551294/sequenced_security_frame.txt)

Fishwaldo, 2016-11-01:
Thanks. As the comment says, I'd never seen this before. I'll look into it soon.

aromanro, 2018-04-06:
I can confirm that there are devices that send sequenced security frames. One example is 'Senzor 4 in 1 Zipato (Philio)'.

Fishwaldo, 2018-04-26:
Can you provide a log extract of the sequenced frames?

aromanro, 2018-04-26:
I didn't use open zwave for noticing that, but code I wrote. I'm not sure I can provide a log from it, I signed an NDA and I'm not sure if that doesn't qualify as disclosure. Anyway, solving this wasn't hard at all.