OpenZWave/open-zwave #1027: Danfoss thermostatic radiator valve model 014G0013 does not react to Command Protection when set to "Protection_NOP"

alexus1211, 2016-11-04:
Hi,

the Danfoss  TRV model 014G0013  works fine under Openzwave except when it comes to the command class Protection. When I try to modify the protection level, I always get the response 00 ("Unprotected").
According to specs, this TRV supports 2 values only:
_0: Unprotected
2: No Operation allowed_

When I use Zway to control the protection class, It works flawlessly. Zway uses 2 bytes in the parameter list compared to Openzwave.
From the [Zway developer manual, page 80](http://razberry.z-wave.me/docs/zway.pdf):

_Syntax: Set(state, rfState = 0, successCallback = NULL, failureCallback = NULL)
Description: Send Protection Set
Parameter state: Local control protection state
Parameter rfState: RF control protection state_

the parameter rfState=0 is always sent. The Danfoss TRV LC13 seems to cope with this.

With OpenZwave, this parameter is not used (check line 160 in file ["Protection.cpp"](https://github.com/OpenZWave/open-zwave/blob/ad0d31e407aca6be328c82e58aa8b4ddcded42a4/cpp/src/command_classes/Protection.cpp
)):
```
//-----------------------------------------------------------------------------
// <Protection::SetValue>
// Set the device's protection state
//-----------------------------------------------------------------------------
bool Protection::SetValue
(
	Value const& _value
)
{
	if( ValueID::ValueType_List == _value.GetID().GetType() )
	{
		ValueList const* value = static_cast<ValueList const*>(&_value);
		ValueList::Item const *item = value->GetItem();
		if (item == NULL)
			return false;

		Log::Write( LogLevel_Info, GetNodeId(), "Protection::Set - Setting protection state to '%s'", item->m_label.c_str() );
		Msg* msg = new Msg( "ProtectionCmd_Set", GetNodeId(), REQUEST, FUNC_ID_ZW_SEND_DATA, true );
		msg->SetInstance( this, _value.GetID().GetInstance() );
		msg->Append( GetNodeId() );
		msg->Append( 3 );
		msg->Append( GetCommandClassId() );
		msg->Append( ProtectionCmd_Set );
		msg->Append( (uint8)item->m_value );
		msg->Append( GetDriver()->GetTransmitOptions() );
		GetDriver()->SendMsg( msg, Driver::MsgQueue_Send );
		return true;
	}

	return false;
}
```

This seems to be a bit problematic for Danfoss LC13.
Please note that other Zwave devices can cope with this wihtout any problem. A successfull example is the Room Sensor from Danfoss.

nechry, 2016-11-04:
I confirm, jeedom users with the Danfoss 014G0013, have the same issue.

alexus1211, 2016-11-04:
It implements CC Protection Version 2.
Here the link to the specs at [page 103](http://z-wave.sigmadesigns.com/wp-content/uploads/2016/08/SDS12652-13-Z-Wave-Command-Class-Specification-N-Z.pdf)

Fishwaldo, 2016-11-05:
OZW has only protection version 1 support which is a single byte. 

as per the specs, it should be backwards compatible. If its not, its a device bug. regardless, we can put CC Version 2 on the roadmap.

alexus1211, 2016-11-29:
I agree, the device seems to have a problem. Any timeline to implement Protection Class V2?

poty87, 2018-05-22:
some updates? I have the same problem

if i try {"cmd":"setValue","args":[2,117,1,0,2]} 

2: NODEID
117: number class protection
1:instance
0:index
2: value for class protection (0 no protection 2 full protection)

with postman, the Danfoss not response but if i try anyone command for example COMMAND_CLASS_THERMOSTAT_SETPOINT the Danfoss it works