OpenZWave/open-zwave #1028: Assumption of BASIC command class isn't always correct

jonathanfrenz, 2016-11-04:
Sorry to bother you again!

I have this device: http://www.pepper1.net/zwavedb/device/684

The device does not support the Basic command class (unlike pretty much every other Z-wave device I've learned). It seems like OpenZWave is assuming that it supports the class and then spends all of its time querying that command class (and getting no response).

I can make things work better if I comment out this line in Node.cpp which seems to be the one that is adding the Basic command class:
```
		// Add the mandatory command classes for this generic class type
		//AddMandatoryCommandClasses( genericDeviceClass->GetMandatoryCommandClasses() );
```

This file is the bad behavior:
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/571929/OZW_Log.txt)

And here's what happens if I comment that line out and delete the cache:
[OZW_Log success.txt](https://github.com/OpenZWave/open-zwave/files/571954/OZW_Log.success.txt)

jonathanfrenz, 2016-11-04:
Also curious to me is that it in the better run it finds a lot of the command classes:
2016-11-04 10:22:27.096 Info, Node005,   Zwave+ Device Type  (0xc06) - Access Control Sensor. Mandatory Command Classes:
2016-11-04 10:22:27.097 Info, Node005,     COMMAND_CLASS_DEVICE_RESET_LOCALLY
2016-11-04 10:22:27.098 Info, Node005,     COMMAND_CLASS_ZWAVE_PLUS_INFO
2016-11-04 10:22:27.099 Info, Node005,     0x59 (Not Supported)
2016-11-04 10:22:27.100 Info, Node005,     COMMAND_CLASS_MANUFACTURER_SPECIFIC
2016-11-04 10:22:27.101 Info, Node005,     COMMAND_CLASS_POWERLEVEL
2016-11-04 10:22:27.102 Info, Node005,     COMMAND_CLASS_ASSOCIATION
2016-11-04 10:22:27.103 Info, Node005,     COMMAND_CLASS_VERSION
2016-11-04 10:22:27.104 Info, Node005,     COMMAND_CLASS_SENSOR_BINARY

But it still doesn't find COMMAND_CLASS_BARRIER_OPERATOR which is the most important class to find.

Fishwaldo, 2016-11-05:
1) Its a broken device. As per the Z-Wave Specs, Entry control Device class, Basic class is mandatory (Section 5.6.1) - http://z-wave.sigmadesigns.com/wp-content/uploads/2016/08/SDS10242-29-Z-Wave-Device-Class-Specification.pdf

2) the master branch doesn't have BARRIER_OPERATOR in it, and its WIP in the dev branch, which is why it doesn't show up.

closed 2016-11-05 by Fishwaldo

Fishwaldo, 2016-11-07:
> If mandated by a generic or specific device class, a device MUST support the Basic Command Class...

In this case, the Generic Class, Entry Control, mandates the Basic Class. 

> 2) I am using the Dev branch. There is a BarrierOperator .cpp and .h in my solution.
because you removed that code, BarrierOperator.cpp is not added to the device.

jonathanfrenz, 2016-11-08:
I think there are three very basic points here.

1) The Z-Wave Plus specification is extremely clear that Barrier Operators do NOT need to implement the Basic type.

2) This is a Z-Wave certified product, and it was certified explicitly not supporting the Basic type.

3) This is the top-selling Z-Wave garage door opener... From a pragmatic standpoint, it would be good to support it:
https://amazon.com/s/ref=nb_sb_ss_c_1_12?url=search-alias%3Daps&field-keywords=z-wave+garage+door+controller&sprefix=z-wave+garag%2Caps%2C158&crid=3502FAE3X621Q

Fishwaldo, 2016-11-08:
All your points would be valid if it was a Barrier Z-Wave+ Device, but its not. Its a Access Control Sensor. 

And suggest you read back over the documents and understand the relationships between classic Zwave and ZWave+ to understand how they map to each other. If one device type says MUST, it doesn't matter what other device types say.